Xinyuan-LilyGO/T-SIM7600X #5: gps code sample work

tradexsrl, 2021-08-04:
Hi , i get this board but i have found very few documentation, have you got a working code for gps?
for the version 7000G i had to modify tinygsm library otherwise i get no response from gps satellite
please let me know

LilyGO, 2021-08-05:
> Hi , i get this board but i have found very few documentation, have you got a working code for gps?
> for the version 7000G i had to modify tinygsm library otherwise i get no response from gps satellite
> please let me know

Allfunctions.ino has all the tests.

tradexsrl, 2021-08-10:
ok i'm using this but i get negative latitude and negative longitude with .getps() , if i delete the minus sign everythings is ok
instead the gpsraw() get positive latitude and longitude.......

tradexsrl, 2021-09-07:
i have used that .ino file but it doesn't work correctly , you need to make a modification in the tinygsm library as written in #566 issue

osugiw, 2021-09-09:
Im using SIM7600GH and have problem with GPS, I also using AllFunctions.ino and it couldn't retrieve the coordinates like this:
[image]

Is there any solutions to this?

tradexsrl, 2021-09-09:
are you inside or outside? inside of a building these gps never works , to make it work you need to use an external antenna and place it outside the window , then you need also to modify the library as written in #566 issue.
first time you locate it takes at least 3-5 minutes

osugiw, 2021-09-09:
> are you inside or outside? inside of a building these gps never works , to make it work you need to use an external antenna and place it outside the window , then you need also to modify the library as written in #566 issue.
> first time you locate it takes at least 3-5 minutes

yeah it's first time to power on, true i'm inside building but near to window and already attach external antenna.. Finally the coordinates appears after some minutes. But why it took so long to get the coordinates any idea? I've tried using SIM7000C to retrieve coordinates, it just need some seconds to get the coordinates.

osugiw, 2021-09-09:
> are you inside or outside? inside of a building these gps never works , to make it work you need to use an external antenna and place it outside the window , then you need also to modify the library as written in #566 issue.
> first time you locate it takes at least 3-5 minutes

could you share the url for #566 issue? maybe it could help on other problems

tradexsrl, 2021-09-09:
for external antenna i mean not the one that comes with the board , i have to buy one on amazon
in my experience also the sim7000 takes some times to locate.
this is the url: https://github.com/vshymanskyy/TinyGSM/issues/566

osugiw, 2021-09-09:
I bought from the aliexpress the external antenna is included.
[image]

I think it depends on type of the external antenna, because I use this antenna on SIM 7000C and just took some seconds to get the coordinates
[image]

Thanks for the url!

tradexsrl, 2021-09-09:
the small one marked 1568R-A that comes with the board with simcom7600 never worked inside , i use exactly the same black antenna with long cable and it is better

osugiw, 2021-09-09:
> the small one marked 1568R-A that comes with the board with simcom7600 never worked inside , i use exactly the same black antenna with long cable and it is better

Got it! Thanks sir for the sharing it helps a lot!

closed 2021-10-06 by LilyGO