Xinyuan-LilyGO/T-SIM7600X #12: How to detect if card is battery powered or usb powered?

Paranoramix, 2021-12-30:
I am wondering how to detect if the card is powered by USB or if it is running on battery. In schematics, I don't see anything that would allow this. 

Can you confirm this for me?

thanks in advance

LilyGO, 2022-01-06:
Hello, are you asking whether the SIM card is battery powered or USB powered?

Paranoramix, 2022-01-06:
Hello,

Thanks for your anwser. 
After reading again schematics, I saw the ADC(35) can be used to check if the card is battery powered or USB powered.

```
if (analogRead(35) > 0) {
Serial.println("Battery Powered!");
} else {
Serial.println("USB Powered!");
}
```

Basically, I would like know the state of the battery (is charging, charged, or in use). Is it possible?

Regards

LilyGO, 2022-01-06:
You can check the charging status through the indicato
[image]
r

Paranoramix, 2022-01-06:
OK, thanks, but this is just a visual indicator.
I meant programmatically ;)

closed 2022-01-12 by Paranoramix

rayenbouzidi, 2023-05-24:
hello bro do you solve this problem ?