Xinyuan-LilyGO/T-SIM7600X #13: Power consumption of SIM7600 in sleep mode

MilkoKrastev, 2022-02-10:
I have been trying for days to reduce the power consumption of the SIM7600 module while keeping it registered to the network. It is the following scenario:

0. ESP32 provisions GPIO 32 (DTR) as an RTC output pin. 
1. ESP32 disconnects SIM7600 by issuing an ATH command.  
2. ESP32 issues AT+CSCLK=1 command to SIM7600. 
3. ESP32 sets DTR to HIGH so that SIM7600 can enter sleep mode. 
4. ESP32 enters deep sleep. 

Expected result: SIM7600 consumes ~2mA, i.e. SIM7600 is in sleep mode
Actual result: SIM7600 consumes ~20mA

The above sequence with the same source code works fine with TTGO ESP32 SIM7000G, but it doesn't work with TTGO SIM7600E. 

Any idea what could be wrong? Do I need to issue an explicit command to disable something on the SIM7600 side which consumes power? GNSS is off for sure. Not sure whether the USB interface of SIM7600 could be using some power.

LilyGO, 2022-02-11:
[reference](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7600X/blob/master/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino)  ，Finally, there is the sleep code

MilkoKrastev, 2022-02-11:
Thanks for the quick response. I had a look at the provided sample code, it overlaps with my implementation. 

Let me rephrase my question:

When using TTGO SIM7600E with ESP32 in deep sleep and SIM7600 in sleep mode (GNSS and USB off), what is the battery current draw I should get from this breakout board considering there are no other peripheral devices connected and drawing any current?

My setup works just fine with TTGO SIM7000G and draws 2-3 mA in this mode (ESP32 deep sleep, SIM7000G in sleep mode). 

Should I expect roughly the same current draw from TTGO SIM7600E or it is more power hungry in this setup. Please, note that I want to keep SIM7600E ON (in sleep mode, registered to the network). If I turn off SIM7600E, the current draw goes down to 2-3 mA. However, the next time it starts up, it has to register again to the network which is inefficient with frequent wake ups. My device is solar powered, and I need it to be efficient. It already works when the modem is SIM7000G, but due to the phase out of 2G, and the lack of LTE-M, I am forced to use regular LTE achieving the same power efficiency.

LilyGO, 2022-02-14:
There is no AT+CSCLK in the AT instruction manual for SIM7600.  
Try the following commands  
1） AT+CHSICSLEEP=1
2） AT+CLANCTRL=5
[image]
.

MilkoKrastev, 2022-04-30:
> There is no AT+CSCLK in the AT instruction manual for SIM7600.
Try the following commands
1） AT+CHSICSLEEP=1
2） AT+CLANCTRL=5

Actually, the commands you suggested above are not recognized by SIM7600. 

I think there is a much more significant hardware issue in the design of your latest series of modules. For example, to avoid the issue I am having with TTGO SIM7600, I purchased a module with A7670 instead, namely, this one:

https://github.com/Xinyuan-LilyGO/LilyGO-T-A7670X/tree/main/examples/old_version

It seems to follow the older layout of your TTGO SIM7000 boards. And its A7670 sleep mode works great. However, when I later purchased several more boards, I got the newer version, namely, this one:

https://github.com/Xinyuan-LilyGO/LilyGO-T-A7670X/tree/main/examples/new_version

It has the same issue with sleep mode as your modules with SIM7600. And it also got a USB connected to the modem. It seems to me this USB addition draws some current either through a voltage regulator or something else. It may also prevent the modem from entering sleep mode via AT+CSCLK=1 and DTR pulled up by probably violating the USB condition outlined in the hardware design guides in SIM7100_SIM7500_SIM7600 Series_Sleep Mode_Application Note_V3.00 or A76XX Series_Sleep Mode_Application Note_V1.02.

Is there any way to disconnect this USB addon via an instruction sent through UART or hardware modification? 

Thanks.

LilyGO, 2022-05-04:
T-A7670X(new_version).  The IO12 controls the power supply of module A7670 when batteries are in use.  You should pull up before sleeping  .(pinMode(BAT_EN, INPUT_PULLUP);) 。I get about 2.5mA power consumption

MilkoKrastev, 2022-05-04:
> T-A7670X(new_version). The IO12 controls the power supply of module A7670 when batteries are in use. You should pull up before sleeping .(pinMode(BAT_EN, INPUT_PULLUP);) 。I get about 2.5mA power consumption

Thanks for the feedback. Two questions regarding T-A7670X(new_version). 

1. When you say you are getting 2.5mA power consumption, what is the state of T-A7670X then. Is it in power off or in sleep mode (i.e AT+CSCLK=1 and DTR high)?

2. If the PWRKEY for T-A7670X is IO12, what is the DTR pin then? Is it IO25 or IO32, or something else?

Thanks!

LilyGO, 2022-05-04:
1. AT+CSCLK=1 ,  DTR  HIGH  
2. #define PIN_DTR      25  
    #define BAT_EN       12
    #define PWR_PIN      4
[image]

MilkoKrastev, 2022-07-19:
Hi again, did you have time to measure the current consumption of TTGO SIM7600 when ESP32 is in deep sleep and SIM7600 is in sleep mode (registered on the network)? 

Is there any resistor I could remove to achieve current consumption similar to TTGO SIM7000G/A7670? My understanding is that something is preventing SIM7600 to enter sleep mode, it is most likely the USB which remains active all the time even when on battery. But I may be wrong.

LilyGO, 2022-11-02:
Try to remove it as shown in the picture[image]
[image]

MilkoKrastev, 2023-01-02:
Thanks a lot for the suggested solution. It works! The resulting power consumption is as expected. 

Feel free to close the ticket.

hardastyle, 2023-01-24:
> 
> 0. ESP32 provisions GPIO 32 (DTR) as an RTC output pin.
> 1. ESP32 disconnects SIM7600 by issuing an ATH command.
> 2. ESP32 issues AT+CSCLK=1 command to SIM7600.
> 3. ESP32 sets DTR to HIGH so that SIM7600 can enter sleep mode.
> 4. ESP32 enters deep sleep.
>

Question... Could you share the code you are using for sleeping and waking up the modem? I'm having issues with the asleep modem not reacting to the DTR pin.

Arnold1, 2023-03-10:
@MilkoKrastev can you share the code?

lewisxhe, 2023-03-13:
1. If the modem is off and the esp32 is asleep, the current consumption is less than 1mA. I can't remember the details
2. If esp32 is sleeping, the current of modem is about 100mA, about 20mA
3. You can use the timed wake-up function of esp32 to send data when the time is up
3. If the voltage is too low, the board may not work properly. When using solar energy, you need to judge whether the battery voltage can maintain the normal operation of the modem at the time of startup. If not, you need to continue to enter the up-reading mode

Arnold1, 2023-03-13:
@lewisxhe so for 4 there is never a manual reset required?

when do you need to use the manual reset on that board?

lewisxhe, 2023-03-14:
Currently, there is no situation that requires manual reset

Eya-hm, 2023-04-20:
Hello
I have just modified a code and I am testing the consumption afterwards with a multimeter. Indeed, the idea is to program a loop of x current measurements with a deep sleep between each measurement. If we do the deep sleep normally we will fall into an infinite loop because after the deep sleep we rerun the program from the beginning. But normally I was able to solve this by calling on the preferences library of esp32 which stores each time the iteration at which we know we have stopped and that after we continue from the number of the stored iteration. In the rest of the program, once we have finished the measurement, we initialize the simcom7600 modem, we connect to mqtt, we send the data and finally we turn off the modem. The program works perfectly in the serial monitor, we can see x measurements with a deep sleep between and once it finishes the measurements it continues its program. The problem is when I measure the consumption with the multimeter we can see that it goes down to the deep sleep value which is 0.2 mA only once after the first measurement and for all the rest of the program we see a high consumption then that it is supposed to be at 0.2 mA after each measurement. the mcu goes into deep sleep for one minute. I do not understand why it works in the serial monitor and we even have the values ​​received in the platform but when we measure with the multimeter it seems that it does not enter the measurement loop with deep sleep knowing that the goal is to minimize the card consumption. Could you help me if you have a solution to this?

testing a code in order to optimize it and Iam expecting ideas or solutions for my problem

lewisxhe, 2023-04-21:
You mean the deep sleep cycle will fail after measuring with a multimeter? he won't start?

Eya-hm, 2023-04-21:
Thank you for responding. The deep sleep cycle works well in the serial monitor i can see good results. he enters the sleep mode after each measurments and then continue the rest of code but the problem is when i measure the consumption with a multimeter  i can't see good results of consumption. For example in deep sleep i reach the 97 mA , it says the same for one minute (time wakeup of the deep sleep ) and then it increases and then for the next deep sleep it decrease to 97 mA again. I don't understand why it reachs the 1mA in deep sleep while i can see well tha the program works.
When i test whith a simple code without loop measures ( one measure) and put the deep sleep at the end of the program it works well and i can reach the 1mA but when i try it with the loop the consumption is false.  knowing that I initiate the modem only after the loop and I turn it off after each sending.

Eya-hm, 2023-04-21:
> You mean the deep sleep cycle will fail after measuring with a multimeter? it won't start ?

at the beginning I had this problem but after a small modification I solved it and I had another problem. 

This: https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/13#issuecomment-1517973808

lewisxhe, 2023-04-21:
1. Did you close the modme when setting the sleep mode? Is the status of the indicator light off? Has the esp32 gone to sleep? I feel that esp32 has not entered sleep, otherwise it will not consume so much current

Eya-hm, 2023-04-21:
In the function of loop measurments i didn’t close it , but i close it after its initialisation and transmission of data ( at the end of the void loop) . And for the deep sleep in the serial monitor after each measurments it stays one minute and then redo the next iteration with a message displayed so logically the esp32 has gone to sleep for one minute.

lewisxhe, 2023-04-21:
I think esp32 didn't go to sleep. Please check your code

Eya-hm, 2023-04-21:
Okay i will check it thank you

thanhtin1859, 2023-06-26:
> Thanks a lot for the suggested solution. It works! The resulting power consumption is as expected.
> 
> Feel free to close the ticket.

Hello, Can you tell me how did you fix it? Thanks!

lfdominguez, 2023-09-24:
Only add to the `examples/sleep` that when you come back from sleep, put on the Radio again, because when the modem (in this board) come back from sleep, the FLIGH pin stuck on DOWN, son not network connection again.

neuralprocessing, 2023-09-25:
Can this be achieved on the A7670E (newer version) as well?

neuralprocessing, 2023-09-25:
> On newer hardware T54_V1.3, unsoldered here
> 
> [image]
> 
> Power consumption dropped from 20+mA to ~1mA.  Used 'examples/sleep'.
> 
> 

Can this be achieved as well on the A7670E?

babak7293, 2023-09-25:
Which component is that in the schematic? why does it reduce current consumption?

lfdominguez, 2023-09-25:
I tried to search this component on schematic but I dont found it. What are the draw backs of removing this component?