Xinyuan-LilyGO/T-SIM7600X #14: Powering the module

AirTobe91, 2022-03-10:
Hello,

I have a few questions regarding the powering of the module. I have inserted a 18650 battery. 

1.) When I connect the module to a 5V power source via USB-C cable this will charge the 18650 battery or?

2.) Is it possible to connect a 5V power source to the VBUS pin to power the module? Or is this a output pin of 5V of USB is connected? (See the red circle on the picture attached)
[image]

3.) If 2.) Is possible this will also charge the 18650 battery or?

Thanks in advance!
Best regards,
Tobias

LilyGO, 2022-03-11:
1. Usb-c cable will charge the 18650 battery when connecting the module to a 5V power supply  
2.  5V power supply connected to VBUS pins can supply power to the module
3. VBUS cannot charge the 18650

AirTobe91, 2022-03-11:
Ok, I need to power and charge the module via a 5V DC power supply. So this should work if I connect 5V to the solar panel input or? I think there shouldn't be a difference if this is solar panel DC power or 5V DC power?

LilyGO, 2022-03-12:
Yes. What can you do

closed 2022-03-15 by AirTobe91