Xinyuan-LilyGO/T-SIM7600X #16: Pin-out reference in utilities.h example

JDDV, 2022-04-11:
I'm currently looking at the provided ['AllFunctions'](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7600X/tree/master/examples/Arduino_TinyGSM/AllFunctions) example, which contains an `utilities.h` file containing the pin numbers of some pins of the modem. I can't seem to find the relation between the pins in the `utilities.h` and the pins in the schematic.

For example the `utilities.h` says:
```
#define MODEM_TX            27
#define MODEM_RX            26
#define MODEM_PWRKEY        4
#define MODEM_DTR           32
#define MODEM_RI            33
#define MODEM_FLIGHT        25
#define MODEM_STATUS        34
```
Where you can see that for example the `MODEM_PWRKEY` is set to pin 4.

yet the [schematic](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7600X/blob/master/schematic/T54_SIM7600_20201012.pdf) does not show the `MODEM_PWRKEY` set to pin 4 but to pin 3. This also goes or the RX and TX. But if I change it in the software to the pins I see in the schematic, the example doesn't work. Can someone explain me if I look wrong at the schematics, or what else the problem is?

Thanks in advance!

JDDV, 2022-04-11:
@LilyGO, any Idea what I'm doing wrong, or where I can see the correct pin numbering? Thanks in advance!

LilyGO, 2022-04-12:
PWRKEY is controlled by GPIO4  
[image]

JDDV, 2022-04-12:
Ah I see, I figured it would be pin 3. but GPIO4 pulls the pwrkey to GND I see. Thank you for clarifying it. Keep up the good work!

closed 2022-04-12 by JDDV