Xinyuan-LilyGO/T-SIM7600X #23: how to use mqtt on this device

spereg53, 2022-07-03:
can you help me to use mqtt on this device,thanks

LilyGO, 2022-07-06:
[MQTT Application](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7600X/blob/master/Manual/SIM7500_SIM7600_SIM7800%20Series_MQTT(S)_Application%20Note_V3.00.pdf)

spereg53, 2022-07-06:
please some example with mqtt for lilygo-t- sim7600g-h like in this github:
LilyGO-T-A7670X <https://github.com/Xinyuan-LilyGO/LilyGO-T-A7670X>/examples
<https://github.com/Xinyuan-LilyGO/LilyGO-T-A7670X/tree/main/examples>/
old_version/
thanks in advance
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

LilyGO, 2022-07-07:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7600X/tree/master/examples/MqttClient

spereg53, 2022-07-07:
thanks....
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***