Xinyuan-LilyGO/T-SIM7600X #24: SIM7600 did not connect to network

DAChT98, 2022-07-21:
A few days ago I bought new boards and they have not been able to connect to the network. I did not have this problem with other plates that I bought before.

The code I'm trying to test is the following, it works perfectly on the boards I bought a couple of months ago, but it doesn't on the ones I received yesterday

[EMMABOX2.0.zip](https://github.com/Xinyuan-LilyGO/T-SIM7600X/files/9160179/EMMABOX2.0.zip)
.

LilyGO, 2022-07-22:
Hello，Try to see if the [example  ](https://github.com/Xinyuan-LilyGO/T-SIM7600X/tree/master/examples/Arduino_TinyGSM/AllFunctions) works.

lewisxhe, 2023-03-13:
No activity for a long time, please reopen it if there is any problem

closed 2023-03-13 by lewisxhe