Xinyuan-LilyGO/T-SIM7600X #25: Red blinking light (without battery) error

limneos, 2022-08-24:
I just got the board and tested the AllFunctions.ino.
Everything worked perfectly.
After a few successful tests, I changed the sleep time to 10 minutes (600 seconds) and used a battery and logged the VBAT (pin 35) analog reading. I worked fine for the first 2 wake ups. (I set up a TCP test to my own server and  the GET requests were coming through via 4G/LTE).
After 2 successful wake-ups, the module stopped responding.
I removed the battery, and tried connecting via USB to flash it again.
The red battery charging LED turned on (without a battery connected).
It has been randomly changing from blinking red to constant red on every USB connection.
It's been like this ever since, it does get detected as a serial port but it doesn't accept any connections , I cannot flash it, and while I keep it connected via USB and weirdly the battery charging red LED is on, after a a minute the SIM7600 chip and the ESP32 chip get really hot (at which point I disconnect it)

I don't see anything that could have caused this problem in my doings, I just uploaded the test code which was working fine and got it looping between deep sleeps...

Is this a known fault? is there a solution? is the board faulty? missing chip pins soldering? 

Any help would be appreciated.

 

https://user-images.githubusercontent.com/239807/186529416-96639fcf-b284-4ee6-b3c0-c9b0f05bfa50.MOV

limneos, 2022-08-24:
Update: after connecting to a different USB port on my Mac Mini M1, I got the "USB Accessory disabled" warning message about the device drawing too much power from USB... it appears there's a short of some kind on the board, but I can't find it anywhere...

[image]

Arnold1, 2023-03-10:
hi did you solve that problem?

lewisxhe, 2023-03-13:
It may be damaged. Please contact the dealer for after-sales service

closed 2023-03-13 by lewisxhe