Xinyuan-LilyGO/T-SIM7600X #26: Is there a way to see how much battery I have?

NicoYaez, 2022-08-25:
Hello good, is there any way to see how much battery percentage my board has? or an estimate

limneos, 2022-08-25:
You can read the analog value of GPIO35 to get the battery voltage. It returns 0 if you're connected via USB though, so you can't really log it to the console, you should e.g. text the value via SMS or post it to a URL in order to log it. 

In my tests, it requires a mapping to get the real actual voltage:

```
int batteryLevel=analogRead(35);
int mappedBatteryLevel=map(batteryLevel,0,2430,0,4200);
```

Proxicon, 2022-10-29:
Hello, are you sure it is pin35 ? I tried it and got nothing back, pin35 is not listed on the product specification?
https://www.lilygo.cc/products/t-sim7600

limneos, 2022-10-30:
Yes, I'm sure as I am currently using it with pin 35. It's not listed because it's an internal pin, it doesn't provide an input/output pin.  Like I mentioned above, you won't get a value if you're connected with USB. The device must be running solely on battery in order to get this reading, so if you are connected with USB and try to log the value, it will be 0 . You need e.g. to send the value by SMS every 1 minute or post it to a URL in order to have proper readings while on battery.

Proxicon, 2022-11-02:
Thanks @limneos - Yes was aware of the connected to batt=0 vs not. My issue was initiation GPIO controller on pin 35 while on battery only & connected to my local Wi-Fi posting the values to my mqtt broken had 0 for values returned well.

I tried doing the same thing with Esphome & noticed that the example sim800 instance an ADC controller class instead of GPIO.

I'm using .net Nanoframework C# by the way, since I like the debugger & coding in visual studio. 

[image]

Instancing an ADC controller as opposed to GPIO controller returns data.

[image]

Now to convert to battery % :-)

closed 2023-03-13 by lewisxhe