Xinyuan-LilyGO/T-SIM7600X #27: Is there any way to get Serial logs when module is powered by battery?

milindmovasha, 2022-09-18:
Hello,  
In order to do debugging when the module is on battery we need to get the Serial logs when module is on battery. Is there any way to get the Serial logs when module is powered by battery? What we need is to see the logs printed by Serial.println() on Serial console when module is powered by battery.

When module is powered by USB port by connecting it to computer we are able to see the logs on Serial console.
We tried connecting external USB TTL UART CP2102 module to computer and connecting the pins as follows:
Module Rx: CP2102 TTL Tx
Module Tx: CP2102 TTL Rx
Module GND: CP2102 TTL GND

With this setup when the module powered by battery and is booted we see initial logs dumped by the ESP32 boot code on the Serial console but none of the logs as per Serial.println() calls in the sketch are shown on the console.

lewisxhe, 2022-09-19:
You can redirect RX/TX Pin when initializing Serial

````
Serial.begin(115200, SERIAL_8N1, rxpin, txpin);
````

milindmovasha, 2022-09-19:
Thanks for your reply. Tried with following sketch:
```
const byte rxPin = 21;
const byte txPin = 22;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200, SERIAL_8N1, rxPin, txPin);
  }

void loop() {
  Serial.println("hello world 01");
  delay(1000);
  // put your main code here, to run repeatedly:

}
```
Module GPIO21: CP2102 TTL Tx
Module GPIO22: CP2102 TTL Rx
Module GND: CP2102 TTL GND

It is working fine as expected.

closed 2022-09-19 by milindmovasha