Xinyuan-LilyGO/T-SIM7600X #28: Can't run the module from a battery

barcesat, 2022-10-27:
Hi,
I'm not sure what I'm missing but I'll ask this anyway:
When I unpug the USB cable from the module the module turns off. It does not matter if:
1) the state of the power button
2) If there's a li-ion battery plugged in the back.
Can't this be ran from the battery??

LilyGO, 2022-10-28:
Hello，
1. Is your power switch on?
2. Do you use 18650 batteries? How much voltage does the battery have?

LilyGO, 2022-10-29:
Which example are you using? Is there a video demonstration of what you did?

barcesat, 2022-10-29:
I'm using the AllFunctions example.
I did this:
1. Upload the example to the T-SIM board
2. made sure it is working while the USB cable and battery are plugged in.
3. Connect an FTDI UART to usb board to the TXRX lines and make sure that I can read the printout on the serial monitor (While battery and USB cable are plugged in).
4. Disconnect the USB cable, trying to switch ON-OFF and ON again the T-SIM and I get nothing on the serial, no LED lights, it is dead :(

Photo:
[image]

barcesat, 2022-10-31:
No print at all.
I also tried to upload blink. It works when USB is connected and does not work when USB is disconnected and power switch is set to ON.

LilyGO, 2022-11-01:
Try to use a multimeter to measure whether there is 3.3v

lewisxhe, 2023-03-13:
No activity for a long time, please reopen it if there is any problem

closed 2023-03-13 by lewisxhe