Xinyuan-LilyGO/T-SIM7600X #31: How can I access the SPK and MIC pins with this board?

contractorwolf, 2022-12-08:
I see the pins in the interface schematics but I am unsure of how I can access them.  Are they mapped or labels somewhere?

lewisxhe, 2023-03-13:
[image]
You need to make sure that your board has an audio decoding chip, otherwise it cannot be used

closed 2023-03-13 by lewisxhe

contractorwolf, 2023-03-13:
Where can I buy the one with the audio board?

lewisxhe, 2023-03-14:
When purchasing, you can explain to the sales that you need to decode the chip

contractorwolf, 2023-03-14:
Explain to who? Is there no link for that feature? Any idea on the cost or have someones email I could talk to?

lewisxhe, 2023-03-14:
I have reported the situation to Lilygo. I believe that there will be a corresponding purchase link soon. At present, please contact the staff of Quicksell and explain to them that you need a SIM7600 with a decoding chip