Xinyuan-LilyGO/T-SIM7600X #36: No GPS being received

Tissy, 2023-03-07:
I wonder if someone can assist please?  I have recently powered up the LILYGO® TTGO T-SIM7600G-H R2 T-SIM7600E-H ESP32 Module ESP32-WROVER, however can't seem to get anything working.

I have successfully ensured I have flashed the latest firmware onto the modem, however using the [AllFunctions.ino](https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino) demo, I get only the following output.

`SDCard MOUNT FAIL
[1781] Initializing modem...
[1781] ### TinyGSM Version: 0.11.5
[1781] ### TinyGSM Compiled Module:  TinyGsmClientSIM7600
AT
AT
AT
AT
[11991] Failed to restart modem, delaying 10s and retrying
[11992] Initializing modem...
[11992] ### TinyGSM Version: 0.11.5
[11992] ### TinyGSM Compiled Module:  TinyGsmClientSIM7600
AT
AT
AT
+CPIN: READY

OK
[16485] Enabling GPS/GNSS/GLONASS
AT+CGPS=1

OK
[18495] Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,,,,,,,,

OK
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,,,,,,,,
`

Any suggestions please as I am at a loss.

Many thanks.

Tissy, 2023-03-08:
This issue has now been resolved with a firmware update to the SIM7600G-H.  Finding the firmware was the hardest part !

closed 2023-03-08 by Tissy

Sell24, 2023-04-12:
@Tissy Would you be able to point me in the direction of where you got the firmware from? Thanks!

OmerFarukOruc, 2023-06-09:
I am having same issue. How to update firmware? I couldnt see any link regarding this.