Xinyuan-LilyGO/T-SIM7600X #39: the power consumption of Tsim7600G-H boards

Mariem114114, 2023-03-28:
Hello, @LilyGO @lewisxhe 

I am testing the deep sleep mode using the same code on two different but identical Tsim7600G-H boards. However, the power consumption of the two boards is different (one is 0.967mA and the other is 4.778mA) even though I have made the same connections and used the same code. Is this normal and where could this error come from?

lewisxhe, 2023-03-29:
Normally, it is below 1mA 4.7mA is a bit too much. Are you measuring the power supply from the battery?

Mariem114114, 2023-03-30:
Hello, @lewisxhe 

Thank you for your response. No, I am currently powering the card using an external power supply and measuring its consumption using an ammeter. What's strange is that I'm using the same code and wiring for both cards, but they're not consuming power in the same way. Could the problem be with the card? If not, is there a minimum value for consumption that can be reached with the T-SIM7600G-H card?

Thank you.

lewisxhe, 2023-03-31:
The average operating time of the SIM7600 is about 150mA, which is normal when used with the ESP 32 at 190mA
Turn off the power supply of the modem when it is not in use, control PWRKEY (io4), or use the AT command to shut down the modem. Because the boost controller of this board does not have separate control, after turning off the modem, DCDC conversion will also consume a portion of the current

Mariem114114, 2023-04-04:
Hello again @lewisxhe ! Thank you for your question. I'm using MQTT for transmission and it lasts for 13 seconds, which is not normal and significantly increases power consumption. Do you have a solution to this problem?