Xinyuan-LilyGO/T-SIM7600X #40: MQTT Disconnects after 24hrs

droidblastnz, 2023-04-02:
My code is working as expected, except a MQTT disconnect around 24hrs. 

Using the example MQTT code I can connect to broker.hivemq.com and blink the LED without issues but after close to 24hrs disconnects and will not reconnect. Up to the disconnect is +CIPCLOSE: 1,0,0,0,0,0,0,0,0,0 the +CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

`18:51:18.272 -> fail
18:51:18.344 -> ERROR: CRC_ERROR
18:51:18.344 -> AT+CGREG?
18:51:18.344 -> 
18:51:18.388 -> +CGREG: 0,1
18:51:18.388 -> 
18:51:18.388 -> OK
18:51:18.388 -> AT+CIPRXGET=4,0
18:51:18.432 -> 
18:51:18.432 -> +CIPRXGET: 4,0,0
18:51:18.432 -> 
18:51:18.432 -> OK
18:51:18.432 -> AT+CIPCLOSE?
18:51:18.476 -> 
18:51:18.476 -> +CIPCLOSE: 0,0,0,0,0,0,0,0,0,0
18:51:18.518 -> 
18:51:18.518 -> OK
18:51:18.518 -> fail
18:51:18.606 -> AT+CGREG?
18:51:18.650 -> 
18:51:18.650 -> +CGREG: 0,1
18:51:18.650 -> 
18:51:18.650 -> OK
18:51:18.650 -> fail
18:51:18.781 -> AT+CGREG?
18:51:18.781 -> 
18:51:18.781 -> +CGREG: 0,1
18:51:18.827 -> 
18:51:18.827 -> OK
18:51:18.827 -> fail
18:51:18.899 -> AT+CGREG?
18:51:18.919 -> 
18:51:18.919 -> +CGREG: 0,1
18:51:18.964 -> 
18:51:18.964 -> OK
18:51:18.964 -> AT+CIPRXGET=4,0
18:51:18.964 -> 
18:51:19.012 -> +CIPRXGET: 4,0,0
18:51:19.012 -> 
18:51:19.012 -> OK
18:51:19.012 -> AT+CIPCLOSE?
18:51:19.052 -> 
18:51:19.052 -> +CIPCLOSE: 0,0,0,0,0,0,0,0,0,0
18:51:19.052 -> 
18:51:19.101 -> OK
18:51:19.101 -> fail
`

lewisxhe, 2023-04-13:
There are many reasons for the disconnection. First, you need to find out the specific reason for the disconnection. The following reasons may cause the disconnection. You can refer to
1. Change of network base station
2. Unstable power supply
3. The signal becomes weak or hours

There is no good way to deal with disconnection. You can only check whether the network is connected at intervals. If it is not connected, reconnect

droidblastnz, 2023-04-13:
> There are many reasons for the disconnection. First, you need to find out the specific reason for the disconnection. The following reasons may cause the disconnection. You can refer to
>
>     1. [...]

Thanks been working on this for some time and think its a network issue. I have a consumer SIM and also a M2M sim.  At this stage I need to have it working on the consumer SIM (prepaid).

1. 
2. I have a powered hub running the SIM7600 and a USB V/A monitor which show stable power and low amp draw. I don't see this to be my issue at this stage.
3. I have a external pigtail antenna on the SIM7600 for this reason

Not sure of the actual connection 2G, 4G LTE or 4G Cat 1 need to figure out how to check this and reduce power based on the connection type. Assuming 4G LTE or 4G Cat 1 as 2G networks here is decommissioned.

Signal connection is 21 using a pigtail external antenna as well.
Added the external antenna as its sitting inside on a bench getting - 2…30 – - 109… - 53 dBm

[4G LTE Antenna Wifi NB IoT High Gain Router Antenna for Huawei B315 593S B880 B310 SMA Male IPEX Wireless Module Aerial](https://www.aliexpress.com/item/1005003390439099.html?spm=a2g0o.order_list.order_list_main.85.21ef1802ftObUH)

`int csq = modem.getSignalQuality();`

I am in the city with good signal reception.

This link also has more details but I need to understand how to debug this issue as I also see a lot of other users with this exact issue. So any guidance would be great.

[https://community.home-assistant.io/t/mqtt-noise-when-starting-and-issues-sending-commands-over-3-4g/554708/3](url)

lewisxhe, 2023-04-15:
1. You mentioned that you are using the MQTT library, have you tried the MQTT built into the Modem?
2. Random characters appear, and I don't know what causes it. The best way is to try to use the built-in MQTT service,
Note. I have tested most of SIMCom's built-in MQTT, none of them have the function of reconnection, and must be reconnected by the user after disconnection

The built-in MQTT of SIM7600 is very convenient to use, it is not difficult to use like SIM7000.

droidblastnz, 2023-04-17:
To me this part of the code is looping, the netlight is still 200ms flashing so that means its connected to the network but the MQTT function is looping and cant get out of the loop to check if its connected see the results below. Suspect its a combo of MQTT (free version), Network that is where the issue lies.

```
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
Network State is: AT+NETOPEN?

+NETOPEN: 0

OK
0
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
MQTT NOT CONNECTED! 
AT+CIPSEND=0,2

+CIPERROR: 2

ERROR
AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
Network State is: AT+NETOPEN?

+NETOPEN: 0
```

Revised the code and testing to this...

```
if (!mqtt.connected()) {
    SerialMon.println("MQTT NOT CONNECTED! ");
    mqtt.disconnect(); // Disconnect from MQTT  //added 08/04/2023
    SerialMon.print("Network State is: ");
    SerialMon.println(modem.isGprsConnected()); //added 08/04/2023

    // Reconnect every 10 seconds
    uint32_t t = millis();
    if (t - lastReconnectAttempt > 10000L) {
        lastReconnectAttempt = t;
        if (!modem.isGprsConnected()) {
            // Reconnect to GPRS network if not connected
            SerialMon.println("GPRS not connected, reconnecting...");
            if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
                SerialMon.println("GPRS reconnect failed");
                delay(10000);
                return;
            }
        }

        if (mqttConnect()) {
            lastReconnectAttempt = 0;
        }
    }
    delay(100);
    return;
}
```

Checking if the network is disconnected with modem.isNetworkConnected(), it attempts to reconnect to the GPRS network using modem.gprsConnect() if it's not connected. Then, it proceeds with the original logic of checking if the MQTT client is connected and attempting to reconnect to the MQTT broker/server every 10 seconds, but with the added check for GPRS connectivity before attempting MQTT reconnection. This ensures that the GPRS network is reconnected before attempting MQTT reconnection, if necessary.

droidblastnz, 2023-05-01:
After a few tests and waiting 24hrs at least after a change to the code the following code removes the MQTT from a loop and reconnects if disconnected.  As I am using a consumer SIM rather than a M2M SIM suspect the network operator forces the disconnect 18-24hrs after connection.

Original Tiny_GSM code block

```
  if (!mqtt.connected()) {
    SerialMon.println("=== MQTT NOT CONNECTED ===");
    // Reconnect every 10 seconds
    uint32_t t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
      if (mqttConnect()) { lastReconnectAttempt = 0; }
    }
    delay(100);
    return;
  }

  mqtt.loop();
}
```

Modified code block to reconnect.

```
if (!mqtt.connected()) {
    SerialMon.println("MQTT NOT CONNECTED! ");
    SerialMon.print("Disconnecting from: ");
    SerialMon.println(broker);
    mqtt.disconnect(); // Disconnect from MQTT  //added 08/04/2023
    delay(500);
    SerialMon.print("Network State is: ");
    SerialMon.println(modem.isGprsConnected()); //added 08/04/2023

    // Reconnect every 10 seconds
    uint32_t t = millis();
    if (t - lastReconnectAttempt > 10000L) {
        lastReconnectAttempt = t;
        if (!modem.isGprsConnected()) {
            // Reconnect to GPRS network if not connected
            **SerialMon.println("GPRS not connected, reconnecting...");
            if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
                SerialMon.println("GPRS reconnect failed");**
                delay(10000);
                return;
            }
        }

        if (mqttConnect()) {
            lastReconnectAttempt = 0;
        }
    }
    delay(100);
    return;
}
```

closed 2023-05-01 by droidblastnz