Xinyuan-LilyGO/T-SIM7600X #42: GPS not working for SIM7600NA

Sell24, 2023-04-12:
I have the NA version of the SIM7600 and it doesn't seem to return the GPS coordinates. I have tried with the AllFunctions.io (AF) program as well as just issuing the following commands:

**"AT+CGPS=1"**
**"AT+CGPSINFO"**

In the case of the AF code, it gets stuck in the GPS request for loop. And in the case above, the module just returns:

**+CGPSINFO: ,,,,,,,,**

I have an extender cable attached to the GPS module which is located by my window. This setup worked for the SIM7000G so I can't imagine it not working in this case. However with those modules, you had to toggle a GPIO pin in order to power the GPS. 

I have tried to use the command **"AT+CGPSAUTO"** to have the GPS begin on startup but no luck either.

Is there anything special I have to do in order to get this GPS to work for this module?

droidblastnz, 2023-04-12:
Similar? https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/36

Sell24, 2023-04-12:
I did see that but the board specified there is different. The firmware located here is over 3 years old. Not sure where to find the updated firmware for this specific board

lewisxhe, 2023-04-13:
If the NMEA sentence is output, it can prove that the GPS is running, then you need to check the antenna pointed by the arrow in the picture, whether there is a voltage of about 3V

[image]

Sell24, 2023-04-13:
There is approximately 2.5V there

lewisxhe, 2023-04-13:
That should be normal.
AT+CGNSSMODE=15,1
The sentence returns invalid ?

lewisxhe, 2023-04-13:
Please try to output the NMEA sentence for 10 minutes, and then save it in txt, and then I will check if it is normal, remember that the antenna needs to be outdoors.

lewisxhe, 2023-04-13:
oh ! Yes, you should set the mode and NMEA sentence configuration first, and then turn on the GPS. You are right, I made a mistake in the order. Please try to leave it for a while to see if it works.

Sell24, 2023-04-13:
Ok thank you. I will put it outside and let it run for a while and see if it picks it up. 

Also, just so you know, the reason I mentioned firmware is because in issue #36 , @tissy mentioned having to update his firmware in order to get his GPS to work.

I will get back to you to see if moving it outside with the configuration given solves the issue.

Sell24, 2023-04-13:
Hi, 

Ok that worked perfectly. I really appreciate you taking the time and effort to help me!

OmerFarukOruc, 2023-06-09:
@Sell24 What exact move did fix your problem? Because i am having same error and couldn't still find any solution regarding this.

Sell24, 2023-06-09:
The issue for me was a connection issue. I had an extender cable on the GPS connector and it was not making a proper connection. The firmware itself was working properly so I did not need to update it.

I would suggest that you follow the same steps that @lewisxhe suggested to make sure that the hardware is functioning properly. If it is then check your connection and make sure you are setting up your GPS correctly using the AT commands.

flokstra, 2024-01-11:
Apologies for re-opening this issue, but I am experiencing the same thing.
That is with two major differences, the first is that the board worked flawlessly for the first few hours and then just spontaneously stopped. The second is that none of the fixes above helped. 

I have a SIM7600G-H which I first tested with the Allfunctions code. Worked perfectly.  I then altered the code so that it would periodically (every 10 seconds) relay its position over TCP to a MySQL API. Worked like a charm. I then let it do this for a few hours just to populate my database with test data. Starting at 15:16 every 11-14 seconds a GPS fix was acquired and written to the database. Not one was skipped. And this was with the antenna indoors. I didn't even touch it during this time and then at 19:29 it just stopped. The board functions as normal, I just haven't gotten a GPS fix since then anymore. I re-installed the Allfunctions.ino which now only responds with +CGPSINFO: ,,,,,,,, in the GPS test.

After reading the things above and trying the fixes (tried with two different antennas inside and outside in two different cities) am I right to label this as hardware failure?

flokstra, 2024-01-13:
I tried what you asked, no luck!
But I really do think my board is faulty. As I was reading the suggestions above again, and found out I missed something. Never measured the antenna voltage... It's only 0.03V. And since all other modem capabilities are working perfectly, in my opinion, this must be a hardware failure.
Do you agree @lewisxhe ?

Sell24, 2024-01-17:
Well normally I would agree with you if it didn't work at all. But faulty hardware does not account for why it works for a couple of hours then stops working. To me there is something else going on here.

I was also thinking this could be temperature related so I had changed my code to report the mod temperature to see if there was any correlation. It seems to me that powering down the module and waiting (in my case overnight) seems to get it to work again if only briefly. I am still waiting for the faulty hardware to test it.

The problem now is that I was hoping a full restart would fix it so even if it is temperature related knowing that really won't make a difference unless we put in some sort of cooling like a heatsink.

Not sure what I am going to do here. If multiple boards have this issue, then how would I know if any boards I buy in the future will not have this issue?

flokstra, 2024-01-17:
I totally agree with that! I am reluctant to buy another one now too.

Just for the record, mine doesn't start working again after a period of power loss, it's completely without GPS capability. I'm gonna send it back.

lewisxhe, 2024-01-19:
> Well normally I would agree with you if it didn't work at all. But faulty hardware does not account for why it works for a couple of hours then stops working. To me there is something else going on [...]

I wonder if it's a power supply issue? Is it completely dead? Have you checked the supply voltage?

flokstra, 2024-01-19:
Thanks for your reaction! The 0.3V turned out to be a faulty multimeter. I re-measured everything as per your instructions and some more.

Supply voltage: 5.01V
Pin 44: 2.83V
GPS Antenna middle pin: 0.00V
Esp 3V3: 3.29V
Continuity pin 44 -> GPS antenna inner pin: NO
Continuity GND -> GPS antenna outer ring: YES

If you say that the GPS antenna is directly powered from pin 44 of the modem, then the conclusion must be a faulty connection on the board. I do not measure continuity between the GPS middle pin to anything else on the board. 

It is absolutely not dead, everything works fine, it just will not get a GPS fix anymore. Worked perfectly well for the first few hours.

flokstra, 2024-01-19:
The red dots all measure 2.83V, the blue dots measure nothing. Having a look at the close-up i also noticed the component marked with the arrow isn't soldered on straight.
[image]

lewisxhe, 2024-01-20:
It seems that the inductor pointed by the color-changing arrow has been soldered or disconnected. You can measure whether it is connected. If it is soldered, it needs to be soldered again.

Oborona, 2024-01-29:
Hello. I have the same issue. 

After activation of GPS module according to this instruction:

> AT+CGPS=0
> AT+CGNSSMODE=15,1
> AT+CGPSNMEA=200191
> AT+CGPSNMEARATE=1
> AT+CGPS=1
> // NMEA Output to AT port
> AT+CGPSINFOCFG=1,31 (511)

I'm starting to recieve messages like

> b'$GPGSV,7,1,25,01,,,27,02,,,29,03,,,29,04,,,32,1*65\r\n$GPGSV,7,2,25,05,,,27,06,,,29,07,,,28,08,,,27,1*6B\r\n$GPGSV,7,3,25,09,,,35,10,,,36,11,,,36,12,,,27,1*6F\r\n$GPGSV,7,4,25,13,,,29,14,,,28,15,,,33,16,,,34,1*62\r\n$GPGSV,7,5,25,33,,,36,39,,,35,40,,,35,41,,,34,1*68\r\n$GPGSV,7,6,25,42,,,34,46,,,36,48,,,34,49,,,36,1*67\r\n$GPGGA,,,,,,0,,,,,,,,*66\r\n$GPVTG,,T,,M,,N,,K,N*2,*00\r\n$GPGSV,7,7,25,51,,,34,1*60\r\n$GLGSV,2,2,05,,,,28,1*77\r\n'

so, many sattelites are detected, but no fix-position data is available. Voltage on 44 pin is 2.83V, and i tried to attach an external GPS antenna (03D-3S-00-A) with no effect. 
How can i fix this problem?

Oborona, 2024-02-01:
I've tried many variations, and it's not working so far. The antenna is 100% working. What other reasons could there be? Could there be some kind of settings conflict, for example?

lewisxhe, 2024-02-01:
Like @flokstra , have you tracked the voltages at these locations? The most important thing is the voltage on IPEX. If it is normal, then the module must be working.
https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/42#issuecomment-1900782786

Oborona, 2024-02-01:
[image]
Got 2.83V on every point, marked by green.

lewisxhe, 2024-02-01:
[image]
What about this part?

Oborona, 2024-02-01:
[image]

lewisxhe, 2024-02-01:
Then I think there is no problem with the hardware, but I still suspect that there is a problem with the antenna. Have you changed the connector? IPEX TO SMA

Oborona, 2024-02-01:
I've tried several standard antennas and one 03D-3S-00-A with ipex to sma connector. Also i tried on two 7600 boards, so i think, that it's not hardware, but some sort of settings problem. It looks strange, that the satellites are visible, and there are many of them, but the coordinates are missing.

Sell24, 2024-02-01:
> I wonder if it's a power supply issue? Is it completely dead? Have you checked the supply voltage?

Sorry I didn't see this message. No the board is not dead, it just will stop getting GPS coordinates after a couple of hours. As I mentioned, the board recovers after being shutdown overnight.

It could be a power supply issue but I would think that this would happen all the time, not just after a couple of hours. Also there seems to be enough power to supply the modem to send information using TCPIP which I would think would be the bigger draw. 

However I cannot discount the fact that something happens after a while which could indeed be power related. It does not happen to my other board but since these are on separate vehicles I can't rule it out.

Sell24, 2024-02-01:
@Oborona Just so I understand, are you saying you never get GPS coordinates or are they intermittent like mine? I have multiple 7600NA boards and so far it is only this one that has this issue so far, which is why I never mentioned anything until @flokstra mentioned he had an issue, which turned out to be different than mine.

@flokstra Were you able to solve your issue?

Oborona, 2024-02-01:
I always recieve something like "GPGGA,,,,,,0,,,,,,,,66", I don't know if the reason is the same, but according to the symptoms, it seems that there are probably some conditions which both of us violated, but in my case, they are always violated, and in yours only in some cases.

lewisxhe, 2024-02-02:
@Oborona  
I recorded a debugging [video](https://youtu.be/0-49-1qXW4A) and obtained commands that may affect GPS positioning

Please check if the returns of these two commands are the same as mine

```
AT+CGPSPMD?
+CGPSPMD: 65407

AT+CGPSNMEA?
+CGPSNMEA: 198143
```

Oborona, 2024-02-06:
Yes, return values are the same.

Oborona, 2024-05-28:
I've tried to run this from far outside of city and got correct position information. But this requirement makes the device quite useless. I don't think it should work that way. Is there any settings, that i can try to change to allow the device to work in a city? Maybe some precision settings?