Xinyuan-LilyGO/T-SIM7600X #43: Incoming Call and DTMF tones

droidblastnz, 2023-04-12:
If no decoding chip onboard can you still accept a incoming call and DTMF tones?

https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/31

I can ring the SIM and see "incoming call" but doesnt answer? My code example below.

`// Check for incoming call
if (SerialAT.available()) {
  String response = SerialAT.readString();
  if (response.indexOf("RING") != -1) {
    // Incoming call detected
    String phoneNumber = response.substring(response.indexOf("MISSED_CALL: ") + 14);  // Retrieve the phone number of the incoming call
    phoneNumber.trim();  // Trim leading/trailing spaces
    SerialMon.print("Incoming call from: ");
    SerialMon.println(phoneNumber);

    // Wait for a timeout period for DTMF tones
    unsigned long startTime = millis();
    String dtmfTones;
    while (millis() - startTime < 5000) {  // Adjust timeout period as needed
      if (SerialAT.available()) {
        String response = SerialAT.readString();
        if (response.indexOf("DTMF:") != -1) {
          dtmfTones += response.substring(response.indexOf("DTMF:") + 6);
          dtmfTones.trim();
        }
      }
    }
    // Call the appropriate function based on the received DTMF tones
    if (dtmfTones.equals("123")) {  // Update comparison to use equals() for exact string match
      RestartIncomingCall();
    } else if (dtmfTones.equals("456")) {  // Update comparison to use equals() for exact string match
      DeepSleepIncomingCall();
    } else if (dtmfTones.equals("ABC")) {  // Update comparison to use equals() for exact string match
      PIRONIncomingCall();
    } else if (dtmfTones.equals("DEF")) {  // Update comparison to use equals() for exact string match
      PIROFFIncomingCall();
    }
    // Hang up the call
    SerialAT.println("AT+CHUP");  // Hang up the call
  }
}`

Sell24, 2023-04-13:
You need to issue a "ATA" after the ring is detected unless you have set the modem to automatically answer with ATS0=x where x is the number of rings.

droidblastnz, 2023-04-13:
> You need to issue a "ATA" after the ring is detected unless you have set the modem to automatically answer with ATS0=x where x is the number of rings.
> 
> As for your other question it doesn't appear that the 7600x has DTMF detection (AT+DDET) unlike the say SIM800 did.

Thanks 
```

if (SerialAT.available()) {
  //Check for incoming call
  //Enable Caller Line Identification Presentation (CLIP)
  modem.sendAT("+CLIP=1");  //Calling line identification presentation
  modem.waitResponse(1000L);
  String response = SerialAT.readString();
  if (response.indexOf("RING") != -1) {
    // Incoming call detected
    String phoneNumber = response.substring(response.indexOf("MISSED_CALL: ") + 14);  // Retrieve the phone number of the incoming call
    phoneNumber.trim();  // Trim leading/trailing spaces
    SerialMon.print("Incoming call from: ");
    SerialMon.println(phoneNumber);

    // Answer the call
    //SerialAT.println("ATA");     //Answer the call
    SerialAT.println("ATS0=005");  //Answer the call after 5 rings
    
    // Wait for a timeout period for DTMF tones
    unsigned long startTime = millis();
    String dtmfTones;
    while (millis() - startTime < 5000) {  // Adjust timeout period as needed
      if (SerialAT.available()) {
        String response = SerialAT.readString();
        if (response.indexOf("DTMF:") != -1) {
          dtmfTones += response.substring(response.indexOf("DTMF:") + 6);
          dtmfTones.trim();
        }
      }
    }
    // Call the appropriate function based on the received DTMF tones
    if (dtmfTones.equals("123")) {  // Update comparison to use equals() for exact string match
      RestartIncomingCall();
    } else if (dtmfTones.equals("456")) {  // Update comparison to use equals() for exact string match
      DeepSleepIncomingCall();
    } else if (dtmfTones.equals("ABC")) {  // Update comparison to use equals() for exact string match
      PIRONIncomingCall();
    } else if (dtmfTones.equals("DEF")) {  // Update comparison to use equals() for exact string match
      PIROFFIncomingCall();
    }
    // Hang up the call
    SerialAT.println("AT+CHUP");  // Hang up the call
  }
}
```

Answering the call and shows the call but no DTMF tones?

droidblastnz, 2023-04-13:
https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/31 I have no audio chip on the SIM7600 do I then assume no DTMF tones will be generated?

Sell24, 2023-04-13:
Just an FYI, the ATS0 setting will not be saved. So on next reboot you would need to set it up again. It is probably best to send the command during your setup. You really need a whole routine that send commands and then parses the response. Or you could just use TinyGSM library. 

As for the DTMF, the 7600 can generate DTMF but unlike the 800 doesn't appear to have the capability to decipher DTMF. I could be wrong that you could add the hardware (as in #31 ) but I dont see how you get that information from the SIM since the 7600 AT commands dont seem to have the ability access it. 

My suggestion is to download the 7600 AT command PDF from here if you don't already have it.

droidblastnz, 2023-04-13:
Thanks assume you are correct -doesn't appear to have the capability to decipher DTMF? If so this can be closed.

closed 2023-04-28 by droidblastnz