Xinyuan-LilyGO/T-SIM7600X #44: Response to incoming call from a explicte number

droidblastnz, 2023-04-13:
Trying to response to a call from a number xxxxx struggling to return the incoming call number...

```
if (SerialAT.available()) {
  String response = SerialAT.readString();
  if (response.indexOf("RING") != -1) {
    // Incoming call detected
    String phoneNumber = response.substring(response.indexOf("MISSED_CALL: ") + 14);  // Retrieve the phone number of the incoming call
    phoneNumber.trim();  // Trim leading/trailing spaces
    // Extract the phone number from the response
    int colonIndex = phoneNumber.indexOf(":");
    if (colonIndex != -1) {
      phoneNumber = phoneNumber.substring(colonIndex + 1);
    }
    SerialMon.print("Incoming call from: ");
    SerialMon.println(phoneNumber);

    // Call the appropriate function based on the received DTMF tones
    RestartIncomingCall();
    //ESP.restart();

    // Hang up the call
    SerialAT.println("AT+CHUP");  // Hang up the call
  }
}
```

OK
Incoming call from: 

```
if (SerialAT.available()) {
  String response = SerialAT.readString();
  if (response.indexOf("RING") != -1) {
    // Incoming call detected
    String phoneNumber = response.substring(response.indexOf("MISSED_CALL: ") + 14);  // Retrieve the phone number of the incoming call
    phoneNumber.trim();  // Trim leading/trailing spaces
    // Extract the phone number from the response
    int colonIndex = phoneNumber.indexOf(":");
    if (colonIndex != -1) {
      phoneNumber = phoneNumber.substring(colonIndex + 1);
    }
    SerialMon.print("Incoming call from: ");
    SerialMon.println(phoneNumber);

    // Check if the incoming call is from the specified phone number
    if (phoneNumber.equals("027xxxxxxx")) {
      // Call the appropriate function based on the received DTMF tones
      RestartIncomingCall();
      //ESP.restart();
    } else {
      // Hang up the call for other phone numbers
      SerialAT.println("AT+CHUP");  // Hang up the call
    }
  }
}
```

lewisxhe, 2023-04-13:
Did you buy the version with the codec? If there is no decoder version, it is impossible to make and receive calls.

droidblastnz, 2023-04-13:
I can make a call to the SIM and call a ESP.Restart so assume it does? Just need to understand how to see the calling number ID (AT+CLIP=1)?

Also how to sync Cell network time with modem and ESP? Seem to have different zones/times.

```
  //Send AT command to retrieve current time
  //modem.sendAT("+CCLK=\"23/04/12,22:06:00+12\""); //e.g., "23/04/12,17:00:32+48" YY/MM/DD,HH,SS,TZ
  //modem.waitResponse(1000);
  modem.sendAT("+CCLK?");    
  modem.waitResponse(1000);
```

Result:

Incoming call from: 1,0  <- not correctly finding the incoming number e.g., 0274123456
RING
Restaring Call recieved!
MISSED_CALL: 06:11AM 0274123456

```
if (SerialAT.available()) {
  String response = SerialAT.readString();
  if (response.indexOf("RING") != -1) {
    // Incoming call detected
    String phoneNumber = response.substring(response.indexOf("MISSED_CALL: ") + 14);  // Retrieve the phone number of the incoming call
    phoneNumber.trim();  // Trim leading/trailing spaces
    // Extract the phone number from the response
    int colonIndex = phoneNumber.indexOf(":");
    if (colonIndex != -1) {
      phoneNumber = phoneNumber.substring(colonIndex + 1);
    }
    SerialMon.print("Incoming call from: ");
    SerialMon.println(phoneNumber);

    // Call the appropriate function based on the received DTMF tones
    RestartIncomingCall(); //Display calling image
    //ESP.restart(); 

    // Hang up the call
    SerialAT.println("AT+CHUP");  // Hang up the call
  }
}
```

lewisxhe, 2023-04-15:
After AT+CLIP=1 is enabled, the dialed-in number will be passed in, and the returned number will be provided by the operator

[image]

droidblastnz, 2023-04-15:
Do you have some example code please?

lewisxhe, 2023-04-15:
I just burned ATDebug and then used Serial for AT interaction

After waiting for network registration, send AT+CLIP=1
Then use another mobile phone to make a call to the board

Also, the returned data is completely issued by the operator. If it is not correct, you can try to replace a SIM card

droidblastnz, 2023-04-16:
Thanks enabled in the setup(); section.

Q. How to sync Cell network time with modem and ESP? Seem to have different zones/times.

lewisxhe, 2023-04-21:
You can get the time from the modem, and then write the time to the esp32
, the specific writing method can be viewed here

https://github.com/lewisxhe/SensorsLib/blob/745462a134ee326d1e1fec55acd9724fe5472f23/src/SensorPCF8563.hpp#L449

droidblastnz, 2023-04-28:
Following code works for two incoming numbers.

```
if (SerialAT.available()) {
  String response = SerialAT.readString();
  SerialAT.println("AT+CLIP=1");  // Calling line identification presentation
  if (response.indexOf("RING") != -1) {
    // Incoming call detected
    String phoneNumber = response.substring(response.indexOf("+CLIP: ") + 7, response.indexOf("+CLIP: ") + 18);  // Retrieve the phone number of the incoming call
    phoneNumber.trim();  // Trim leading/trailing spaces
    SerialMon.print("Incoming call from: ");
    SerialMon.println(phoneNumber.substring(1, phoneNumber.length()));
    phoneNumber = (phoneNumber.substring(1, phoneNumber.length()));
    
    if (phoneNumber == "01233445667") {
      //If caller is 0274371247 do something
      mqtt.publish(topicIncomingCallStatus, "Incoming");// Publish Incoming Call
      IncomingCall();
    //ESP.restart(); //Call the SIM7600 to reboot

      //Hang up the call
      SerialAT.println("AT+CHUP");  // Hang up the call
      delay(1000);
      mqtt.publish(topicIncomingCallStatus, "No Calls");// Publish Incoming Call
    }
    if (phoneNumber == "0123456789") {
      //If caller is 0212971015 do something
      mqtt.publish(topicIncomingCallStatus, "Incoming");// Publish Incoming Call
      IncomingCall();
    //ESP.restart(); //Call the SIM7600 to reboot

      //Hang up the call
      SerialAT.println("AT+CHUP");  // Hang up the call
      delay(1000);
      mqtt.publish(topicIncomingCallStatus, "No Calls");// Publish No Incoming Call
    }
  }
}
```

closed 2023-04-28 by droidblastnz