Xinyuan-LilyGO/T-SIM7600X #46: external battery

Mariem114114, 2023-04-21:
Hello @LilyGO  @lewisxhe  ,

Im currently looking into the T-SIM7600G-H R2 board and have a few questions:
-  Can I use a different battery than a 18650? I would like to connect this one instead: https://www.adafruit.com/product/353 - this one has more mAh than a standard 18650 battery... - if not what needs to change to make it work? il Try to connect the + to Vbat and the  to Gnd but it does not work .

- Does that board need a manual reset if the battery is empty but the solar panel fully recharges the battery? Assume that board operates remotely - i cannot go there and press a reset button...

- I plan to connect the following solar cell - would that be ok with the battery above? https://www.adafruit.com/product/3809 - any problem you see with this one - too slow charing?

- Is there a waterproof antenna i can use? I think the one which is included with that board is not waterproof?
Im also looking for a case/box to protect the board from water and sun/heat - any recommendation?

thank you .

lewisxhe, 2023-04-21:
1. Okay, you said it doesn't work after connecting? Have you confirmed that the switch has been turned to the ON side?

2. If the battery is exhausted, the board will fall into an infinite loop, that is, it will continue to restart with low power, so you must detect the battery voltage on the software, and leave a margin. When starting up, you need to judge whether the battery voltage is sufficient. Provide power consumed by work

3. Yes, the board has solar charging, fully supported

4. Sorry for this, I don't have a suitable case, AliExpress should have many suitable ones

Mariem114114, 2023-04-22:
Hello @lewisxhe @LilyGO ,

I changed the battery and now I use 15000mAH of 3.6 V and  yes the  the switch has been turned to the ON side,  the board lights up but it does not execute the code and I also noticed that the led of the modm barely lights up (I use mqtt in my code), how should I proceed? Is there a circuit to add between the external battery and the Vbat (knowing that I directly connect the + to the Vbat and - to the gnd), the problem can come from the code? is there a configuration of the pin to do or not?

lewisxhe, 2023-04-22:
Please connect the battery, and then directly measure whether there is 3.3V output on the pins on both sides. I suspect that the battery protection circuit has been damaged

Mariem114114, 2023-04-22:
@lewisxhe 
yes there is 3;3V  and I tried to change the battery with a 3500mah and it now works perfectly, so I think the problem comes from the 150000 mAH battery despite it being charged and I can't use the 3500mah battery because I need more make the card work for a long time can I put several batteries in a support? or will it not work? what do you offer me as a solution knowing that with my code and the 3500mAH battery it only lasts 5 days and I need to operate the card for 6 months, despite the fact that I have optimized my code .do you have an idea ? and thank you for your time

lewisxhe, 2023-04-22:
You mean the 3500mA battery works, but the 15000mA battery doesn't? ?

Mariem114114, 2023-04-22:
@lewisxhe  yes exactly

lewisxhe, 2023-04-23:
I think this is unlikely, as long as the voltage is consistent, the battery capacity will not affect the normal operation of the board, do you have a voltmeter? You connect the batteries to the board separately, and then measure the voltage of the two batteries after connecting to see if they are the same

Mariem114114, 2023-04-25:
Hello @LilyGO @lewisxhe 

 Ido that and i have 4.14V for the battery of 3500mAh and 3.5V for the battery of 15000mAh .

Mariem114114, 2023-04-26:
@lewisxhe what's aboute the current of decharge i have 17A in the 3500mAh and 100mA in the 15000mAh it can be the problem ? and what's the minimum of current that the card need , thank you

lewisxhe, 2023-04-28:
The battery you are using may not be suitable for use here. The working voltage of the board should be above 3.8v. You said that the battery connection voltage varies between 3.3 and 2.6v, indicating that there may be a problem with the battery

Sakura-TA, 2024-02-20:
> 2\. If the battery is exhausted, the board will fall into an infinite loop, that is, it will continue to restart with low power, so you must detect the battery voltage on the software, and leave a [...]

what if the board keeps restarting with low power and triggerred the over discharge protection, can it boot up without a manual reset when the battery voltage is sufficient? If not, does it mean that when the weather comes a long time rainy that exhausts the battery, a manual reset is still necessary?

lewisxhe, 2024-03-05:
No, it can't be done. You must go to sleep before the battery is low, and check whether the battery voltage can maintain operation every time you wake up.

For example, the over-discharge protection has been reached. Even if the sun is sufficient the next day and the battery is charged, if there is no software management and the board is allowed to consume power, it will enter a low-voltage reset until the battery voltage is exhausted, creating an infinite loop. .

Sakura-TA, 2024-03-08:
> No, it can't be done. You must go to sleep before the battery is low, and check whether the battery voltage can maintain operation every time you wake up.
>
> For example, the over-discharge protection [...]

Actually, I tried to recharge the battery to 4.2V and the board just stays quiet, without any uart output.

victorandueza, 2025-02-03:
Hello,

I am working on a project where I need at least 6 days of autonomy with a deep sleep of about 15 minutes between executions and an execution time of no more than 1 minute. I am currently experiencing the over-discharge issue you mentioned, and I would like to know how to implement the pre-deep sleep check you referred to in previous messages. I have several questions:

What should be the battery voltage limit to ensure that the device does not get stuck in an infinite loop?
If I manage to detect this minimum charge and turn off the device, will the charging system work correctly during the shutdown, allowing the battery to charge while the device is off?
Once the battery has enough charge, how will I know when to turn the device back on? Do I need to wake up and sleep at certain intervals to check the charge level?
Thanks for your help.

lewisxhe, 2025-02-06:
Only enough power can be left to keep the ESP working. This voltage point needs to be tested. The working voltage of esp32 is 3.3V. If it is lower than or equal to 3.0V, it will reset.

The charging system also works when it is turned off.

If it falls into an infinite loop, it may need to be manually restored, because below 3.0V or around 2.8V, esp does not limit the startup, but keeps trying to start.

It is recommended to check the battery voltage every time you use it. The reasonable battery working range point needs to be tested by yourself according to the battery

lewisxhe, 2025-02-07:
My opinion is that before each sleep and after startup, first check the battery voltage to detect whether the voltage is sufficient for operation. If not, then immediately enter sleep after startup, and then try to prolong the sleep time, waiting for the next day's sunlight to charge the battery, rather than waking up at intervals, because you already know that the power is not enough for work. This is the only way I can think of at present, or add an additional watchdog module externally, when ESP32 resets the voltage, force ESP RST to be pulled low, so that ESP cannot start, and then resume when the voltage is normal.