Xinyuan-LilyGO/T-SIM7600X #49: MQTTS 8883

droidblastnz, 2023-05-04:
Please provide some example code using the Tiny_GSM library to connect to a cloud MQTT broker on 8883.

Example in the library will not connect to a private xxxxx.s1.eu.hivemq.cloud 8883 but connects to broker.hivemq.com 1883 without issue. 

```
// MQTT details
const char* broker = "broker.hivemq.com";
```

```
boolean mqttConnect() {
  SerialMon.print("Connecting to ");
  SerialMon.print(broker);

  // Connect to MQTT Broker
  boolean status = mqtt.connect("GsmClientTest");  //works 1883

  // Or, if you want to authenticate MQTT:
  // boolean status = mqtt.connect("GsmClientName", "mqtt_user", "mqtt_pass"); //doesnt worj 8883 with user/password added

  if (status == false) {
    SerialMon.println(" fail");
    return false;
  }
  SerialMon.println(" success");
  mqtt.publish(topicInit, "GsmClientTest started");
  mqtt.subscribe(topicLed);
  return mqtt.connected();
}
```

droidblastnz, 2023-05-05:
Revised code yet to test keen for advice on this below....?

Added SPIFFS
`#include "SPIFFS.h`

Downloaded isrgrootx1.pem
You can download the root certificate [here 1.2k](https://letsencrypt.org/certs/isrgrootx1.pem).
This will create a file called “isrgrootx1.pem”, which you can use as “Server Certificate”.

Created a data dir and copied the isrgrootx1.pem  which is C:\Users\ddd\Documents\Arduino\SIM7600\5.5502\data\isrgrootx1.pem

Used ESP32 Sketch Data Upload to add the isrgrootx1.pem to the ESP32 

`boolean mqttConnect() //boolean true or false
{
  SerialMon.print("Connecting to ");
  SerialMon.println(broker);
  SerialMon.println("");  //add CR

  // Connect to MQTT Broker
  
  String serverAddress = "xxxxxxxx.s1.eu.hivemq.cloud";
  int serverPort = 8883;
  int keepAliveSeconds = 60;
  int cleanSession = 1;
  String username = "xxxxxx";
  String password = "xxxxxx";
  
  // Set SSL version and authentication mode
  modem.sendAT("+CSSLCFG=\"sslversion\",0,4"); //Set the SSL version of the first SSL context
  if (!modem.waitResponse(1000)) {
    Serial.println("CSSLCFG command failed");
    return false;
  }
  
  modem.sendAT("+CSSLCFG=\"authmode\",0,1"); //Set the authentication mode(verify server) of the first SSL context
  if (!modem.waitResponse(1000)) {
    Serial.println("CSSLCFG command failed");
    return false;
  }
  
  // Set server root CA
  //modem.sendAT("+CSSLCFG=\"cacert\",0,\"isrgrootx1.pem\""); //Set the server root CA of the first SSL context
  modem.sendAT("+CSSLCFG=\"cacert\",0,\"/data/isrgrootx1.pem\"");
  if (!modem.waitResponse(1000)) {
    Serial.println("CSSLCFG command failed");
    return false;
  }
  
  // Start MQTT service and activate PDP context
  modem.sendAT("+CMQTTSTART");
  if (!modem.waitResponse(1000)) {
    Serial.println("MQTT START command failed");
    return false;
  }
  
  // Acquire a client
  modem.sendAT("+CMQTTACCQ=0,\"Client1\",1,4"); //1 MQTT server with SSL/TLS 4 MQTT version 3.1.1
  if (!modem.waitResponse(1000)) {
    Serial.println("MQTT ACCQ command failed");
    return false;
  }
  
  // Set the first SSL context to be used in the SSL connection
  modem.sendAT("+CMQTTSSLCFG=0,0");
  if (!modem.waitResponse(1000)) {
    Serial.println("MQTT SSLCFG command failed");
    return false;
  }
  
  // Connect to MQTT server
  String mqttConnectCommand = "AT+CMQTTCONNECT=0,\"ssl://" + serverAddress + ":" + String(serverPort) + "\"," + String(keepAliveSeconds) + "," + String(cleanSession) + ",\"" + username + "\",\"" + password + "\"";
  boolean status = SerialAT.print(mqttConnectCommand);
  
  if (status == false) {
    SerialMon.println("MQTT Connect fail");
    return false;
  }
  SerialMon.println("MQTT Connect success");

  //If GRPS Connected
  if ((status == true) && modem.isGprsConnected()) { 
  }

  mqtt.subscribe(topicLed);                                                     
  return mqtt.connected();
}`

droidblastnz, 2023-05-05:
What cert do I need?  last one?

`openssl s_client -showcerts -verify 5 -connect xxxxx.s1.eu.hivemq.cloud:8883`

See here [https://community.hivemq.com/t/arduino-example-with-server-signed-certificate/502/5](url)

result gives me 

```
verify depth is 5
CONNECTED(00000003)
depth=2 C = US, O = Internet Security Research Group, CN = ISRG Root X1
verify return:1
depth=1 C = US, O = Let's Encrypt, CN = R3
verify return:1
depth=0 CN = *.s1.eu.hivemq.cloud
verify return:1
---
Certificate chain
 0 s:CN = *.s1.eu.hivemq.cloud
   i:C = US, O = Let's Encrypt, CN = R3
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: May  4 07:57:45 2023 GMT; NotAfter: Aug  2 07:57:44 2023 GMT
-----BEGIN CERTIFICATE-----

-----END CERTIFICATE-----
 1 s:C = US, O = Let's Encrypt, CN = R3
   i:C = US, O = Internet Security Research Group, CN = ISRG Root X1
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Sep  4 00:00:00 2020 GMT; NotAfter: Sep 15 16:00:00 2025 GMT
-----BEGIN CERTIFICATE-----

-----END CERTIFICATE-----
 2 s:C = US, O = Internet Security Research Group, CN = ISRG Root X1
   i:O = Digital Signature Trust Co., CN = DST Root CA X3
   a:PKEY: rsaEncryption, 4096 (bit); sigalg: RSA-SHA256
   v:NotBefore: Jan 20 19:14:03 2021 GMT; NotAfter: Sep 30 18:14:03 2024 GMT
-----BEGIN CERTIFICATE-----

-----END CERTIFICATE-----
---
Server certificate
subject=CN = *.s1.eu.hivemq.cloud
issuer=C = US, O = Let's Encrypt, CN = R3
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA-PSS
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 4523 bytes and written 446 bytes
Verification: OK
---
New, TLSv1.2, Cipher is ECDHE-RSA-AES128-GCM-SHA256
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES128-GCM-SHA256
    Session-ID: xxxx
    Session-ID-ctx:
    Master-Key: xxxx
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    Start Time: 1683271214
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: yes
```

droidblastnz, 2023-05-06:
Update using tera term to write isrgrootx1.pem cert directly then list which now I can see the cert file on the module.

```
AT+CCERTDOWN="isrgrootx1.pem", 1939
AT+CCERTLIST
```

Added due to 09:00:28.533 -> +CMQTTSTART: 23

`  // Start MQTT service release the client first
  modem.sendAT("+CMQTTREL=0");
  if (!modem.waitResponse(1000)) {
    Serial.println("CMQTTREL RELEASE command failed");
    return false;
  }

  // Stop MQTT service
  modem.sendAT("+CMQTTSTOP");
  if (!modem.waitResponse(1000)) {
    Serial.println("CMQTTSTOP STOP command failed");
    return false;
  }

  // Start MQTT service and activate PDP context
  modem.sendAT("+CMQTTSTART");
  if (!modem.waitResponse(1000)) {
    Serial.println("CMQTTSTART START command failed");
    return false;
  }`

which is returning these logs

```
10:39:37.199 -> AT+IPADDR
10:39:37.199 -> 
10:39:37.199 -> +IPADDR: 100.67.101.191
10:39:37.245 -> 
10:39:37.245 -> OK
10:39:37.245 -> 1
10:39:37.245 -> AT+NETOPEN?
10:39:37.292 -> 
10:39:37.292 -> +NETOPEN: 1
10:39:37.292 -> 
10:39:37.292 -> OK
10:39:37.339 -> AT+IPADDR
10:39:37.339 -> 
10:39:37.339 -> +IPADDR: xx.xx.xx.xx
10:39:37.385 -> 
10:39:37.385 -> OK
10:39:37.385 -> Connecting to xxx.s1.eu.hivemq.cloud
10:39:37.431 -> 
10:39:37.431 -> AT+CNSMOD
10:39:37.431 -> 
10:39:37.431 -> ERROR
10:39:37.431 -> AT+CSSLCFG="sslversion",0,4
10:39:37.431 -> 
10:39:37.431 -> OK
10:39:37.479 -> AT+CSSLCFG="authmode",0,1
10:39:37.479 -> 
10:39:37.479 -> OK
10:39:37.479 -> AT+CCERTLIST
10:39:37.479 -> 
10:39:37.526 -> +CCERTLIST: "isrgrootx1.pem"
10:39:37.573 -> 
10:39:37.573 -> OK
10:39:37.573 -> AT+CSSLCFG="cacert",0,"isrgrootx1.pem"
10:39:37.573 -> 
10:39:37.573 -> OK
10:39:37.619 -> AT+CMQTTREL=0
10:39:37.619 -> 
10:39:37.619 -> OK
10:39:37.619 -> AT+CMQTTSTOP
10:39:37.666 -> 
10:39:37.666 -> +CMQTTSTOP: 0
10:39:37.666 -> 
10:39:37.666 -> OK
10:39:37.712 -> AT+CMQTTSTART
10:39:37.712 -> 
10:39:37.712 -> +CMQTTSTART: 0
10:39:37.760 -> 
10:39:37.760 -> OK
10:39:37.760 -> AT+CMQTTACCQ=0,"Client1",1,4
10:39:37.760 -> 
10:39:37.760 -> OK
10:39:37.808 -> AT+CMQTTSSLCFG=0,0
10:39:37.808 -> 
10:39:37.808 -> OK
10:39:37.808 -> AT+CMQTTCONNECT=0,"tcp://xxx.s1.eu.hivemq.cloud",8883,60,1,"xxx","xxx"
10:39:38.789 -> MQTT Connect success
10:39:38.789 -> AT+NETOPEN?
10:39:38.836 -> 
10:39:38.836 -> +CMQTTCONNECT: 0,12
10:39:38.883 -> 
10:39:38.883 -> ERROR
10:39:38.883 -> AT+CIPRXGET=4,0
10:39:38.929 -> 
10:39:38.929 -> +CIPRXGET: 4,0,0
10:39:38.929 -> 
10:39:38.929 -> OK
10:39:38.976 -> AT+CIPCLOSE?
10:39:38.976 -> 
10:39:38.976 -> +CIPCLOSE: 0,0,0,0,0,0,0,0,0,0
10:39:39.023 -> 
10:39:39.023 -> OK
10:39:39.116 -> AT+CGREG?
10:39:39.162 -> 
10:39:39.162 -> +CGREG: 0,1
10:39:39.162 -> 
10:39:39.162 -> OK
10:39:39.162 -> MQTT NOT CONNECTED! 
10:39:39.162 -> Disconnecting from: xxx.s1.eu.hivemq.cloud
10:39:39.208 -> AT+CIPSEND=0,2
```

Error is +CMQTTCONNECT: 0,12  which is invalid parameter??

droidblastnz, 2023-05-07:
Revised code

Result error is +CMQTTCONNECT: 0,12 which ‘invalid parameter’

```
10:39:37.199 -> AT+IPADDR
10:39:37.199 -> 
10:39:37.199 -> +IPADDR: xx.xx.xx.xx
10:39:37.245 -> 
10:39:37.245 -> OK
10:39:37.245 -> 1
10:39:37.245 -> AT+NETOPEN?
10:39:37.292 -> 
10:39:37.292 -> +NETOPEN: 1
10:39:37.292 -> 
10:39:37.292 -> OK
10:39:37.339 -> AT+IPADDR
10:39:37.339 -> 
10:39:37.339 -> +IPADDR: xx.xx.xx.xx
10:39:37.385 -> 
10:39:37.385 -> OK
10:39:37.385 -> Connecting to xxx.s1.eu.hivemq.cloud
10:39:37.431 -> 
10:39:37.431 -> AT+CNSMOD
10:39:37.431 -> 
10:39:37.431 -> ERROR
10:39:37.431 -> AT+CSSLCFG="sslversion",0,4
10:39:37.431 -> 
10:39:37.431 -> OK
10:39:37.479 -> AT+CSSLCFG="authmode",0,1
10:39:37.479 -> 
10:39:37.479 -> OK
10:39:37.479 -> AT+CCERTLIST
10:39:37.479 -> 
10:39:37.526 -> +CCERTLIST: "isrgrootx1.pem"
10:39:37.573 -> 
10:39:37.573 -> OK
10:39:37.573 -> AT+CSSLCFG="cacert",0,"isrgrootx1.pem"
10:39:37.573 -> 
10:39:37.573 -> OK
10:39:37.619 -> AT+CMQTTREL=0
10:39:37.619 -> 
10:39:37.619 -> OK
10:39:37.619 -> AT+CMQTTSTOP
10:39:37.666 -> 
10:39:37.666 -> +CMQTTSTOP: 0
10:39:37.666 -> 
10:39:37.666 -> OK
10:39:37.712 -> AT+CMQTTSTART
10:39:37.712 -> 
10:39:37.712 -> +CMQTTSTART: 0
10:39:37.760 -> 
10:39:37.760 -> OK
10:39:37.760 -> AT+CMQTTACCQ=0,"Client1",1,4
10:39:37.760 -> 
10:39:37.760 -> OK
10:39:37.808 -> AT+CMQTTSSLCFG=0,0
10:39:37.808 -> 
10:39:37.808 -> OK
10:39:37.808 -> AT+CMQTTCONNECT=0,"tcp://xxx.s1.eu.hivemq.cloud",8883,60,1,"xxx","xxx"
10:39:38.789 -> MQTT Connect success
10:39:38.789 -> AT+NETOPEN?
10:39:38.836 -> 
10:39:38.836 -> +CMQTTCONNECT: 0,12
10:39:38.883 -> 
10:39:38.883 -> ERROR
10:39:38.883 -> AT+CIPRXGET=4,0
10:39:38.929 -> 
10:39:38.929 -> +CIPRXGET: 4,0,0
10:39:38.929 -> 
10:39:38.929 -> OK
10:39:38.976 -> AT+CIPCLOSE?
10:39:38.976 -> 
10:39:38.976 -> +CIPCLOSE: 0,0,0,0,0,0,0,0,0,0
10:39:39.023 -> 
10:39:39.023 -> OK
10:39:39.116 -> AT+CGREG?
10:39:39.162 -> 
10:39:39.162 -> +CGREG: 0,1
10:39:39.162 -> 
10:39:39.162 -> OK
10:39:39.162 -> MQTT NOT CONNECTED! 
10:39:39.162 -> Disconnecting from: xxx.s1.eu.hivemq.cloud
10:39:39.208 -> AT+CIPSEND=0,2
```

Q. Do I need to send a AT+CMQTTWILLTOPIC and then AT+CMQTTWILLMSG prior to the AT+CMQTTCONNECT?

Added

 ```
 // Set the first SSL context to be used in the SSL connection
  modem.sendAT("+CMQTTSSLCFG=0,0");
  if (!modem.waitResponse(1000)) {
    Serial.println("CMQTTSTART SSLCFG command failed");
    return false;
  }

  // Send AT+CMQTTWILLTOPIC=0,20 test/temperature
  modem.sendAT("+CMQTTWILLTOPIC=0,28,\"test/temperature\"");
  if (!modem.waitResponse(1000)) {
    Serial.println("CMQTTWILLTOPIC command failed");
    return false;
  }

  // Send AT+CMQTTWILLMSG=0,17,1  
  modem.sendAT("+CMQTTWILLMSG=0,2,1");
  if (!modem.waitResponse(1000)) {
    Serial.println("CMQTTWILLMSG command failed");
    return false;
  }
```

Result

```
  // Send AT+CMQTTWILLTOPIC=0,20 test/temperature
  modem.sendAT("+CMQTTWILLTOPIC=0,28,\"test/temperature\"");
  if (!modem.waitResponse(1000)) {
    Serial.println("CMQTTWILLTOPIC command failed");
    return false;
  }

  // Send AT+CMQTTWILLMSG=0,17,1  
  modem.sendAT("+CMQTTWILLMSG=0,2,1");
  if (!modem.waitResponse(1000)) {
    Serial.println("CMQTTWILLMSG command failed");
    return false;
  }
```

11:11:38.110 -> +NETOPEN: 1
11:11:38.110 -> 
11:11:38.158 -> OK
11:11:38.158 -> AT+IPADDR
11:11:38.158 -> 
11:11:38.158 -> +IPADDR: xx.xx.xx.xx
11:11:38.206 -> 
11:11:38.206 -> OK
11:11:38.206 -> Connecting to xxx.s1.eu.hivemq.cloud
11:11:38.252 -> 
11:11:38.252 -> AT+CNSMOD?
11:11:38.252 -> 
11:11:38.252 -> +CNSMOD: 0,8
11:11:38.252 -> 
11:11:38.252 -> OK
11:11:38.299 -> AT+CSSLCFG="sslversion",0,4
11:11:38.299 -> 
11:11:38.299 -> OK
11:11:38.299 -> AT+CSSLCFG="authmode",0,1
11:11:38.299 -> 
11:11:38.346 -> OK
11:11:38.346 -> AT+CCERTLIST
11:11:38.346 -> 
11:11:38.346 -> +CCERTLIST: "isrgrootx1.pem"
11:11:38.395 -> 
11:11:38.395 -> OK
11:11:38.440 -> AT+CSSLCFG="cacert",0,"isrgrootx1.pem"
11:11:38.440 -> 
11:11:38.440 -> OK
11:11:38.440 -> AT+CMQTTREL=0
11:11:38.440 -> 
11:11:38.487 -> OK
11:11:38.487 -> AT+CMQTTSTOP
11:11:38.487 -> 
11:11:38.487 -> +CMQTTSTOP: 0
11:11:38.533 -> 
11:11:38.533 -> OK
11:11:38.533 -> AT+CMQTTSTART
11:11:38.580 -> 
11:11:38.580 -> +CMQTTSTART: 0
11:11:38.580 -> 
11:11:38.580 -> OK
11:11:38.628 -> AT+CMQTTACCQ=0,"Client1",1,4
11:11:38.628 -> 
11:11:38.628 -> OK
11:11:38.628 -> AT+CMQTTSSLCFG=0,0
11:11:38.628 -> 
11:11:38.674 -> OK
11:11:38.674 -> AT+CMQTTWILLTOPIC=0,28,"test/temperature"
11:11:38.674 -> 
11:11:38.674 -> +CMQTTWILLTOPIC: 0,12
11:11:38.722 -> 
11:11:38.722 -> ERROR
11:11:38.770 -> AT+CMQTTWILLMSG=0,2,1
11:11:38.770 -> 
11:11:38.770 -> >CMQTTWILLMSG command failed
11:11:39.844 -> AT+CGREG?
11:11:39.844 -> 
11:11:39.844 -> OK
11:11:40.874 -> Network disconnected
11:11:40.874 -> AT+CSQ
11:11:40.874 -> 
11:11:40.874 -> +CSQ: 24,99
11:11:40.874 -> 
11:11:40.874 -> OK
11:11:40.922 -> AT+CGREG?
11:11:40.922 -> 
11:11:40.922 -> +CGREG: 0,1
11:11:40.922 -> 
11:11:40.968 -> OK
11:11:40.968 -> AT+CGREG?
11:11:40.968 -> 
11:11:40.968 -> +CGREG: 0,1
11:11:41.014 -> 
11:11:41.014 -> OK
11:11:41.014 -> Network re-connected
11:11:41.014 -> AT+NETOPEN?

droidblastnz, 2023-05-10:
Update to this issue

Revised and minimized the code

```
  // Show current network system mode
  modem.sendAT("+CNSMOD?");
  if (!modem.waitResponse(1000)) {
    Serial.println("CNSMOD network system mode command failed");
    return false;
  }
  
  // Set SSL version and authentication mode
  modem.sendAT("+CSSLCFG=\"sslversion\",0,4"); //ALL Set the SSL version of the first SSL context
  if (!modem.waitResponse(1000)) {
    Serial.println("CSSLCFG SSLVersion command failed");
    return false;
  }

  modem.sendAT("+CSSLCFG=\"authmode\",0,1"); //Server authentication. It needs the root CA of the serverSet the authentication mode(verify server) of the first SSL context
  if (!modem.waitResponse(1000)) {
    Serial.println("CSSLCFG Authmode command failed");
    return false;
  }

  // Set server root CA
  modem.sendAT("+CSSLCFG=\"cacert\",0,\"isrgrootx1.pem\""); //Set the server root CA of the first SSL context
  if (!modem.waitResponse(1000)) {
    Serial.println("CSSLCFG CACert command failed");
    return false;
  }
 

  modem.sendAT("+CMQTTSSLCFG?");
  modem.waitResponse(1000);

   
  // Connect to MQTT server
  modem.sendAT("+CMQTTCONNECT=0,\"tcp://xxx.s1.eu.hivemq.cloud:8883\",60,1,\"xxx\",\"xxx\"");
  if (!modem.waitResponse(120000)) {
    Serial.println("CMQTTCONNECT command failed");
    return false;
  }

  modem.sendAT("+CMQTTCONNECT?");
  modem.waitResponse(1000);

  delay(5000); //Allow sufficient time for the SSL handshake to finish
  Serial.println("done!");
```

Logs

```
18:03:34.127 -> AT+NETOPEN?
18:03:34.174 -> 
18:03:34.174 -> +NETOPEN: 1
18:03:34.174 -> 
18:03:34.174 -> OK
18:03:34.221 -> AT+IPADDR
18:03:34.221 -> 
18:03:34.221 -> +IPADDR: xx.xx.xx.xx
18:03:34.268 -> 
18:03:34.268 -> OK
18:03:34.268 -> Connecting to xxx.s1.eu.hivemq.cloud
18:03:34.268 -> 
18:03:34.268 -> AT+CNSMOD?
18:03:34.316 -> 
18:03:34.316 -> +CNSMOD: 0,8
18:03:34.316 -> 
18:03:34.316 -> OK
18:03:34.361 -> AT+CSSLCFG="sslversion",0,4
18:03:34.361 -> 
18:03:34.361 -> OK
18:03:34.361 -> AT+CSSLCFG="authmode",0,1
18:03:34.361 -> 
18:03:34.361 -> OK
18:03:34.408 -> AT+CSSLCFG="cacert",0,"isrgrootx1.pem"
18:03:34.408 -> 
18:03:34.408 -> OK
18:03:34.408 -> AT+CMQTTSSLCFG=0,0
18:03:34.408 -> 
18:03:34.455 -> OK
18:03:34.455 -> AT+CMQTTSSLCFG?
18:03:34.455 -> 
18:03:34.455 -> +CMQTTSSLCFG: 0,0
18:03:34.502 -> +CMQTTSSLCFG: 1,
18:03:34.548 -> 
18:03:34.548 -> 
18:03:34.548 -> OK
18:03:34.548 -> AT+CMQTTCONNECT=0,"tcp://xxx.s1.eu.hivemq.cloud:8883",60,1,"xxx","xxx"
18:03:34.594 -> 
18:03:34.594 -> OK
18:03:39.600 -> done!
18:03:39.600 -> AT+CIPRXGET=4,0
18:03:39.600 -> 
18:03:39.600 -> +CIPRXGET: 4,0,0
18:03:39.649 -> 
18:03:39.649 -> OK
18:03:39.649 -> AT+CIPCLOSE?
18:03:39.649 -> 
18:03:39.694 -> +CIPCLOSE: 0,0,0,0,0,0,0,0,0,0
18:03:39.694 -> 
18:03:39.694 -> OK
18:03:39.838 -> AT+CGREG?
18:03:39.838 -> 
18:03:39.838 -> +CGREG: 0,1
18:03:39.887 -> 
18:03:39.887 -> OK
18:03:39.887 -> MQTT NOT CONNECTED! 
18:03:39.887 -> Disconnecting from: xxx.s1.eu.hivemq.cloud
18:03:39.887 -> AT+CIPSEND=0,2
```

Command `AT+CMQTTCONNECT` returns ok but not +CMQTTCONNECT: 0,0 as per the manual?

```
AT+CMQTTCONNECT=0,"tcp://hooleeping.com:8883",60,1
OK
+CMQTTCONNECT: 0,0
```

droidblastnz, 2023-05-12:
Using this code
```

#define uS_TO_S_FACTOR      1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP       30          /* Time ESP32 will go to sleep (in seconds) */

#define UART_BAUD           115200

#define MODEM_TX            27
#define MODEM_RX            26
#define MODEM_PWRKEY        4
#define MODEM_DTR           32
#define MODEM_RI            33
#define MODEM_FLIGHT        25
#define MODEM_STATUS        34

#define SD_MISO             2
#define SD_MOSI             15
#define SD_SCLK             14
#define SD_CS               13

#define LED_PIN             12

#define TINY_GSM_MODEM_SIM7600
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb
#define SerialAT Serial1

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials, if any
const char apn[]  = "xxxx";     //SET TO YOUR APN
const char gprsUser[] = "";
const char gprsPass[] = "";

#include <TinyGsmClient.h>
#include <SPI.h>
#include <SD.h>
#include <Ticker.h>

#ifdef DUMP_AT_COMMANDS  // if enabled it requires the streamDebugger lib
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, Serial);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

int counter, lastIndex, numberOfPieces = 24;
String pieces[24], input;

bool reply = false;

void modem_on() {
    /*
    The indicator light of the board can be controlled
    */
    pinMode(LED_PIN, OUTPUT);
    digitalWrite(LED_PIN, HIGH);

    /*
    MODEM_PWRKEY IO:4 The power-on signal of the modulator must be given to it,
    otherwise the modulator will not reply when the command is sent
    */
    pinMode(MODEM_PWRKEY, OUTPUT);
    digitalWrite(MODEM_PWRKEY, HIGH);
    delay(300); //Need delay
    digitalWrite(MODEM_PWRKEY, LOW);

    /*
    MODEM_FLIGHT IO:25 Modulator flight mode control,
    need to enable modulator, this pin must be set to high
    */
    delay(1000); //Need delay
    pinMode(MODEM_FLIGHT, OUTPUT);
    digitalWrite(MODEM_FLIGHT, HIGH);

    
  int i = 10;
  Serial.println("\nTesting Modem Response...\n");
  Serial.println("****");
  while (i) {
    SerialAT.println("AT");
    delay(500);
    if (SerialAT.available()) {
      String r = SerialAT.readString();
      Serial.println(r);
      if ( r.indexOf("OK") >= 0 ) {
        reply = true;
        break;;
      }
    }
    delay(500);
    i--;
  }
  Serial.println("****\n");
}

void setup() {
  Serial.begin(115200); // Set console baud rate
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
  delay(100);

  modem_on();
  if (reply) {
    Serial.println(F("***********************************************************"));
    Serial.println(F(" You can now send AT commands"));
    Serial.println(F(" Enter \"AT\" (without quotes), and you should see \"OK\""));
    Serial.println(F(" If it doesn't work, select \"Both NL & CR\" in Serial Monitor"));
    Serial.println(F(" DISCLAIMER: Entering AT commands without knowing what they do"));
    Serial.println(F(" can have undesired consiquinces..."));
    Serial.println(F("***********************************************************\n"));
  } else {
    Serial.println(F("***********************************************************"));
    Serial.println(F(" Failed to connect to the modem! Check the baud and try again."));
    Serial.println(F("***********************************************************\n"));
  }
}

void loop() {
  while (true) {
    if (SerialAT.available()) {
      Serial.write(SerialAT.read());
    }
    if (Serial.available()) {
      SerialAT.write(Serial.read());
    }
    delay(1);
  }

}
```

I can give AT commands direct to the Modem.

```
PDN
AT+CPIN?
AT+CSQ
AT+CGREG?
AT+COPS?
AT+NETOPEN
AT+NETOPEN?
AT+IPADDR
```
Connected with IP and network

```
AT+CSSLCFG="sslversion",0,4
AT+CSSLCFG="authmode",0,1
AT+CSSLCFG="ignorelocaltime",0,1
AT+CSSLCFG="cacert",0,"isrgrootx1.pem"
AT+CSSLCFG="enableSNI",0,1
AT+CMQTTSTART
AT+CMQTTACCQ=0,"Client1",1,4
AT+CMQTTSSLCFG=0,0
```
Here is the issues
1 If I use TeraTerm to do a copy and paste of the cert - error is` +CMQTTCONNECT: 0,34` open SSL session failed
2 Copy the copy and paste into Teraterm via Paster CR  `+CMQTTCONNECT: 0,32` handshake fail

Both can list the Cert
`AT+CCERTLIST` result is `+CCERTLIST: "isrgrootx1.pem"`

So I delete the cert with `AT+CCERTDELE="isrgrootx1.pem"` of 1,939 (1.89 KB (1,939 bytes))

Right click the cert file in windows to get the size 

`AT+CCERTDOWN="isrgrootx1.pem",1939`

Upload the cert
1 If I use TeraTerm to do a copy and paste - I dont get a OK 
2 Copy the copy and paste into Teraterm via Paster CR and paste the cert with only the cert info it comes back with OK

What is the correct method to upload the Cert PEM file to the modem?

This link is method 1 that doesnt finish with ok and hangs
`https://www.hackster.io/victorffs/connecting-sim7600x-h-to-aws-using-mqtt-and-at-commands-2a693c`

droidblastnz, 2023-05-13:
So the question is how do you upload a cert in this case .PEM to the modem? Does anyone have the steps please?

droidblastnz, 2023-05-16:
Tried FTP to get the cert

```
AT+CPIN?
AT+CSQ
AT+CGREG?
AT+COPS?
AT+NETOPEN
AT+NETOPEN?
AT+IPADDR
```

then

```
AT+CFTPSSTART
AT+CFTPSLOGIN="xxx.com",21,"xxx","xxx",0
AT+CFTPSLIST="/"
AT+CFTPSCWD="xxx.com"
AT+CFTPSPWD
AT+CFTPSGETFILE="xx.pem",4
AT+CFTPSGETFILE="xxx.der",4
AT+CFTPSLOGOUT
```
then 

```
AT+CSSLCFG="enableSNI",0,1
AT+CMQTTSTART
AT+CMQTTACCQ=0,"xxx",1,4
AT+CMQTTSSLCFG=0,0
```
then

`AT+CMQTTCONNECT=0,"tcp://xxx.s1.eu.hivemq.cloud:8883",60,1,"xxx","xxx"`

result Error is 32 handshake fail

tried to convert pem to der
`openssl x509 -in isrgrootx1.pem -out isrgrootx1.der -outform DER`

same result

```
20:53:33.709 -> Connecting to xxx.s1.eu.hivemq.cloud
20:53:33.709 -> 
20:53:33.709 -> AT+CNSMOD?
20:53:33.709 -> 
20:53:33.709 -> +CNSMOD: 0,8
20:53:33.758 -> 
20:53:33.758 -> OK
20:53:33.758 -> AT+CSSLCFG="sslversion",0,4
20:53:33.758 -> 
20:53:33.758 -> OK
20:53:33.758 -> AT+CSSLCFG="authmode",0,1
20:53:33.805 -> 
20:53:33.805 -> OK
20:53:33.805 -> AT+CSSLCFG="ignorelocaltime",0,1
20:53:33.805 -> 
20:53:33.805 -> OK
20:53:33.805 -> AT+CCERTLIST
20:53:33.853 -> 
20:53:33.853 -> +CCERTLIST: "isrgrootx1.pem"
20:53:33.900 -> 
20:53:33.900 -> OK
20:53:33.900 -> AT+CSSLCFG="cacert",0,"isrgrootx1.pem"
20:53:33.948 -> 
20:53:33.948 -> OK
20:53:33.948 -> AT+CCHSET=1,1
20:53:33.948 -> 
20:53:33.948 -> OK
20:53:33.948 -> AT+CSSLCFG?
20:53:33.995 -> 
20:53:33.995 -> +CSSLCFG: 0,4,1,1,300,"isrgrootx1.pem","","",1,"",0x0
20:53:34.121 -> +CSSLCFG: 1,4,0,1,300,"","","",0,"",0x0
20:53:34.166 -> +CSSLCFG: 2,4,0,1,300,"","","",0,"",0x0
20:53:34.260 -> +CSSLCFG: 3,4,0,1,300,"","","",0,"",0x0
20:53:34.356 -> +CSSLCFG: 4,4,0,1,300,"","","",0,"",0x0
20:53:34.447 -> +CSSLCFG: 5,4,0,1,300,"","","",0,"",0x0
20:53:34.494 -> +CS+CSSAT+CSSLCFG="enableSNI",0,1
```

Error is 32 handshake fail

lewisxhe, 2023-05-24:
Hi, please take the label of the module that needs to be upgraded. I will send a request to SIMCom for the latest firmware

droidblastnz, 2023-05-24:
> Hi, please take the label of the module that needs to be upgraded. I will send a request to SIMCom for the latest firmware

[image]

Been working with Simcom this is the B04 firmware but I cant upload?
https://www.dropbox.com/s/6lvooa2t5hyrdey/LE20B04V03SIM7600M22.rar?dl=0

lewisxhe, 2023-05-26:
The picture cannot scan the QR code on the modem. I need to confirm the information in the QR code. Can you scan the QR code and copy the information to me?

droidblastnz, 2023-05-26:
> 
Trying to QR scan it but not working

[image]

lewisxhe, 2023-05-26:
This is the latest firmware that SIMCOM provides for this version
https://1drv.ms/u/s!AmbpOqVezk5dsiYweoOa6OkU_p8I?e=6MEeJ1

droidblastnz, 2023-05-26:
> This is the latest firmware that SIMCOM provides for this version https://1drv.ms/u/s!AmbpOqVezk5dsiYweoOa6OkU_p8I?e=6MEeJ1

Thanks but doesn't this needs to be B04 not B03?  
How do you update the SIM?

lewisxhe, 2023-05-26:
They didn't say anything, just let me try flashing the latest firmware to try it out

droidblastnz, 2023-05-26:
> They didn't say anything, just let me try flashing the latest firmware to try it out
> 
>     1. You need to toggle the DIP switch at the bottom of the board to USB TO MODEM
> 
>     2. Install the complete driver, the tool can be obtained here https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/tree/master/update_simxxxx_firmware
> 
>     3. There are steps here https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/blob/master/doc/sim7600SA_update_firmware.md

Thanks if possible can you also create a hivemq instance and see that the firmware allows you to connect via MQTT TLS 8883 please?

lewisxhe, 2023-05-27:
I don't have time for testing at the moment, I need some time

droidblastnz, 2023-05-27:
> Thanks if possible can you also create a hivemq instance and see that the firmware allows you to connect via MQTT TLS 8883 please?

Thanks [lewisxhe](https://github.com/lewisxhe) will wait for your firmware test.

droidblastnz, 2023-06-01:
[lewisxhe](https://github.com/lewisxhe) any update on the firmware test?

lewisxhe, 2023-06-01:
I just found a modem with the same version as yours, I will test it after upgrading, please give me some time.

[image]

lewisxhe, 2023-06-01:
Today I first updated the [upgrade guide ](https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/docs/UpdateGuide/UpdateGuide.md)

lewisxhe, 2023-06-01:
```
AT+CMQTTSTART

+CGREG: 0,1

OK

SMS DONE

OK

+CMQTTSTART: 0
AT+CCERTDOWN="ca_cert.pem",1338

>-----BEGIN CERTIFICATE-----
MIIDrzCCApegAwIBAgIQCDvgVpBCRrGhdWrJWZHHSjANBgkqhkiG9w0BAQUFADBh
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBD
QTAeFw0wNjExMTAwMDAwMDBaFw0zMTExMTAwMDAwMDBaMGExCzAJBgNVBAYTAlVT
MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxGTAXBgNVBAsTEHd3dy5kaWdpY2VydC5j
b20xIDAeBgNVBAMTF0RpZ2lDZXJ0IEdsb2JhbCBSb290IENBMIIBIjANBgkqhkiG
9w0BAQEFAAOCAQ8AMIIBCgKCAQEA4jvhEXLeqKTTo1eqUKKPC3eQyaKl7hLOllsB
CSDMAZOnTjC3U/dDxGkAV53ijSLdhwZAAIEJzs4bg7/fzTtxRuLWZscFs3YnFo97
nh6Vfe63SKMI2tavegw5BmV/Sl0fvBf4q77uKNd0f3p4mVmFaG5cIzJLv07A6Fpt
43C/dxC//AH2hdmoRBBYMql1GNXRor5H4idq9Joz+EkIYIvUX7Q6hL+hqkpMfT7P
T19sdl6gSzeRntwi5m3OFBqOasv+zbMUZBfHWymeMr/y7vrTC0LUq7dBMtoM1O/4
gdW7jVg/tRvoSSiicNoxBN33shbyTApOB6jtSj1etX+jkMOvJwIDAQABo2MwYTAO
BgNVHQ8BAf8EBAMCAYYwDwYDVR0TAQH/BAUwAwEB/zAdBgNVHQ4EFgQUA95QNVbR
TLtm8KPiGxvDl7I90VUwHwYDVR0jBBgwFoAUA95QNVbRTLtm8KPiGxvDl7I90VUw
DQYJKoZIhvcNAQEFBQADggEBAMucN6pIExIK+t1EnE9SsPTfrgT1eXkIoyQY/Esr
hMAtudXH/vTBH1jLuG2cenTnmCmrEbXjcKChzUyImZOMkXDiqw8cvpOp/2PV5Adg
06O/nVsJ8dWO41P0jmP6P6fbtGbfYmbW0W5BjfIttep3Sp+dWOIrWcBAI+0tKIJF
PnlUkiaY4IBIqDfv8NZ5YBberOgOzW6sRBc4L0na4UU+Krk2U886UAb3LujEV0ls
YSEY1QSteDwsOoBrp+uvFRTp2InBuThs4pFsiv9kuXclVzDAGySj4dzp30d8tbQk
CAUw7C29C79Fv1C5qfPrmAESrciIxpg0X40KPMbp1ZWVbd4=
-----END CERTIFICATE-----

OK
AT+CSSLCFG="sslversion",0,4

OK
AT+CSSLCFG="authmode",0,1

OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"

OK
AT+CSSLCFG="clientcert",0,"cert.pem"

ERROR
AT+CSSLCFG="clientkey",0,"key_cert.pem"

ERROR
AT+CMQTTSSLCFG=0,0

OK
AT+CMQTTREL=0

OK
AT+CMQTTACCQ=0,"Sim7600_fa7d2288",1

OK
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1

OK

+CMQTTCONNECT: 0,0

```

Everything seems to be working fine, of course I'm connecting to a public server.

droidblastnz, 2023-06-01:
> ```
> AT+CMQTTSTART
>
> +CGREG: 0,1
>
> OK
>
> SMS DONE
>
> OK
>
> +CMQTTSTART: 0
> AT+CCERTDOWN="ca_cert.pem",1338
>
> MIIDrzCCApegAwIBAgIQCDvgVpBCRrGhdWrJWZHHSjANBgkqhkiG9w0BAQUFADBh [...]

Thanks thats great. Can you please confirm which firmware it was just so I am clear? e.g., [](https://1drv.ms/u/s!AmbpOqVezk5dsiYweoOa6OkU_p8I?e=6MEeJ1)

And could I ask you to test to a private hiveMQ MQTT on 8883?
Free to create a private instance here [https://auth.hivemq.cloud/login?state=hKFo2SBpUzdIcXFGc29Kam5uTmRHVjVVUlZzajBCTUlMWjZGeaFupWxvZ2luo3RpZNkgWmJjQWIxNzBMamd4d2xpYzRSLWZ5OUdEUXU2ejUtMEajY2lk2SBJYWpvNGUzMmp4d1VzOEFkRnhneFFuMlZQM1l3SVpUSw&client=Iajo4e32jxwUs8AdFxgxQn2VP3YwIZTK&protocol=oauth2&audience=hivemq-cloud-api&redirect_uri=https%3A%2F%2Fconsole.hivemq.cloud&scope=openid%20profile%20email&response_type=code&response_mode=query&nonce=V1JtcWV0QzRPWjlaQm80cVI1TVdHd3VFbG45MzFhNWJyeEZsUWZLVTFMaQ%3D%3D&code_challenge=Z2jSfGDkHRMFaZ8d_BazDO3fSNnRvmukoKFYo4ukC1o&code_challenge_method=S256&auth0Client=eyJuYW1lIjoiYXV0aDAtc3BhLWpzIiwidmVyc2lvbiI6IjEuMTMuNiJ9](url) it only needs the server pem/Cert.

lewisxhe, 2023-06-02:
I am currently using version B04 of hivemq and I will test it again when I have time.
```
Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G-H
Revision: LE20B04SIM7600G22
QCN: 
IMEI: 862205052xxxxxxxx
MEID: 
+GCAP: +CGSM
DeviceInfo: 173,170
```

droidblastnz, 2023-06-02:
Thanks so thats the B04 I supplied correct just want to make sure please.

lewisxhe, 2023-06-02:
The screenshot of the upgrade guide is the version I upgraded https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/docs/UpdateGuide/UpdateGuide.md

[image]

lewisxhe, 2023-06-02:
> 这是SIMCOM为此版本提供的最新固件 https://1drv.ms/u/s!AmbpOqVezk5dsiYweoOa6OkU_p8I?e=6MEeJ1

@droidblastnz

droidblastnz, 2023-06-02:
> > 这是SIMCOM为此版本提供的最新固件 https://1drv.ms/u/s!AmbpOqVezk5dsiYweoOa6OkU_p8I?e=6MEeJ1
> 
> @droidblastnz

Sorry still confused with which firmware.  This is LE20B03SIM7600M22.rar
https://1drv.ms/u/s!AmbpOqVezk5dsiYweoOa6OkU_p8I?e=6MEeJ1 

but you mentioned 
The screenshot of the upgrade guide is the version I upgraded `https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/docs/UpdateGuide/UpdateGuide.md` LE20B04sIM7600G22 after you said you had found the same SIM `https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/49#issuecomment-1571229547`

So am I upgrading to B03 or B04?

Thanks running through the instructions this is missing -ATdebug.ino

`https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/docs/examples/ATdebug/ATdebug.ino`

lewisxhe, 2023-06-03:
Sorry, this is for B04.

- [LE20B04SIM7600G22.rar](https://1drv.ms/u/s!AmbpOqVezk5dsiUfnlitTsJBGQ3v?e=o388ew)
- [Windows USB Drivers](https://1drv.ms/u/s!AmbpOqVezk5drSuQMtGsXvJIYrhX?e=EMGq5o)
- [ATdebug](https://github.com/Xinyuan-LilyGO/T-SIM7600X/tree/master/examples/ATdebug)

The one where the USB is near the modem, and the UpdateGuide has been updated
I am using B04, B03 I have not tested, because you have tested I have not tested.

droidblastnz, 2023-06-03:
> Sorry, this is for B04.
>
>     * [LE20B04SIM7600G22.rar](https://1drv.ms/u/s!AmbpOqVezk5dsiUfnlitTsJBGQ3v?e=o388ew)
>
>     * [Windows USB [...]

Thanks again, please confirm the process for uploading the cert to the SIM/Modem?

Oswaldott, 2023-06-03:
Hello @lewisxhe 
I have same problem: I get +CMQTTCONNECT: 0,34. It’s necessary create other Pull request? or can you help me in this PR?
```
Sending command: -- AT+COPS?
AT+COPS?
+COPS: 0,0,"TELCEL TELCEL",7

OK
I (44731) SIM7600: 
********Information !!!
I (44731) SIM7600: ********Loading certificate:

Sending command: -- AT+CCERTDOWN="clientcert.pem",1224
AT+CCERTDOWN="clientcert.pem",1224
>
I (45871) SIM7600: **CERT ACCEPTED!

-----BEGIN CERTIFICATE-----
MIIDWjCCAkKgAwIBAgIVAJZhhS1zVRVWPXfHvssGjJQcaA51MA0GCSqGSIb3DQEB
CwUAME0xSzBJBgNVBAsMQkFtYXpvbiBXZWIgU2VydmljZXMgTz1BbWF6b24uY29t
IEluYy4gTD1TZWF0dGxlIFNUPVdhc2hpbmd0b24gQz1VUzAeFw0yMzA0MzAxOTQz
MzBaFw00OTEyMzEyMzU5NTlaMB4xHDAaBgNVBAMME0FXUyBJb1QgQ2VydGlmaWNh
dGUwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDkoUizNgKzQm9eBr+B
rt/SjNTxa6HxbL8uE9I/wYDCHNvRqp5ZGW6A0zweLCir803WTzL1f1TQc4d4qkAs
+BBGUAzNcAuMAPlGIjJ658qAmY8QOFNEQVdvz3Y5Z5KOZezFurm6iTtCMVkSLnkZ
dQeBSgH14nt3m5wCj1Tgp2ZIBnJimlO7Kxoe5kUfTv+/lR2Rxts/H/x9tyby+Cdg
4VFoG/Lf4bDngC/F4Ww0k2EYLJsi0cUMjekb5OfPimCsuNTzH+A6qkoqVLc1vCmw
4jwrkIxwD8qVkwcAiA7OmXZl2ATv96nHrdmjy+ZA0q4dAHYXa3bgqNhCpkX9xGtr
5ABRAgMBAAGjYDBeMB8GA1UdIwQYMBaAFJeIGPyz9mHJsLAkMPj6q7fmeSW4MB0G
A1UdDgQWBBS4wW/6vFxQlna9Ow1T/vO0BAAsUzAMBgNVHRMBAf8EAjAAMA4GA1Ud
DwEB/wQEAwIHgDANBgkqhkiG9w0BAQsFAAOCAQEAnqqySToJZqEmBYkEEn4/T9/V
ZUE2VHer//QVznaP5+P2WF7DH97sFGQl4XqK2tUARwDVWbO7Ut+DCkZkcSlXrsBw
1SCfopuiYxsgdEk+hpfcGH/RTEI8uSL9yn5Ov7MP1QMLQ19J6afzF7VK8WGYXmSg
Pkd/0rLYzra+F1XhVXBo7cCMXe0aiOr6PAAaW4apae0CAfeOT2GrS9817WSyYsfW
xS9Pfxxj3kkrCFDSSFH0JMOx/p6Pe9/DqqtuomfFOFKVV+mzLAcJ7dzSBwUjpK8f
egKFxZJKIAKXajK/tCXruTRrnSEqeq5s3PiULDZcjw1g2HFhm8aa55GgZypVvg==
E (54971) SIM7600: 
max_ATTEMPTS 20

I (55971) SIM7600: ********Loading certificate:

Sending command: -- AT+CCERTDOWN="clientkey.pem",1679
-----END CERTIFICATE----
OK
AT+CCERTDOWN="clientkey.pem",1679
>
I (58011) SIM7600: **CERT ACCEPTED!

-----BEGIN RSA PRIVATE KEY-----
MIIEpQIBAAKCAQEA5KFIszYCs0JvXga/ga7f0ozU8Wuh8Wy/LhPSP8GAwhzb0aqe
WRlugNM8Hiwoq/NN1k8y9X9U0HOHeKpALPgQRlAMzXALjAD5RiIyeufKgJmPEDhT
REFXb892OWeSjmXsxbq5uok7QjFZEi55GXUHgUoB9eJ7d5ucAo9U4KdmSAZyYppT
uysaHuZFH07/v5UdkcbbPx/8fbcm8vgnYOFRaBvy3+Gw54AvxeFsNJNhGCybItHF
DI3pG+Tnz4pgrLjU8x/gOqpKKlS3NbwpsOI8K5CMcA/KlZMHAIgOzpl2ZdgE7/ep
x63Zo8vmQNKuHQB2F2t24KjYQqZF/cRra+QAUQIDAQABAoIBAQDMLb1fTrzamloa
/qvIhQQeZOdHM3hMw3F5FPqPNaBnup4VgOopaCznn73wtD8eUQsSeWhpFm4IOySg
/mqRvnfKU3MbhRTK05PKrn+mg5q1fm3lbmfubOxJTl2mpnib8yqg60Ey9ZuWYbvK
IPbFZV5ueEmH7duV1cJeBuMmPll5t5eIPEsRtnpsHiDYgTgN5fFYfLZIRM6Y+JBh
ZRwahNRjLTR9rV92xSLI1ZspKmD3hWbY1RQrSa2rzXv9CiQ7KhrG8ZBet8BA0LEg
zxNqEHWnp6yXWhT7JWfoKP6KA1b30SXA2daJiJP9hNz9Z4FIgCXR+Y1lhP+U+zrp
gbq+dHaBAoGBAP5O45cpSsvXdJHoUs1jRZkfDRMZ6SuzqOob8yNmSrKtZKtGin/n
muqjtRSQ1Gg5qbOVLnFm5dQoVMk7uYafpupH+ExQyh8IFw7DKvWEfmXDEGDfPGW/
n3NeIRI/2LBq6CWqh+9Gie+N5sETrEYKXxUTv4wPxUZsvTbJXlt/5ml/AoGBAOYm
qaKdWMOzQF/5q6SlzVnnrUKYSeQdTgWUTRwOCirxBCh6iibnYZbQL4tK+4ibcG+b
7rZShoVASUMQFHNKu5nVOxI7lLsNc5taocV9RipsSp05LVa/BXqpL89nsacRk4Ei
ckkdxHe0pi8ap5oca/0jcmS7QOfNQU8mIv/TJl4vAoGBAJCfIILYcrbVwm8vpdcz
u2LG69KPC7FUrJzmZDEffR97yCI3LZBrkAgcLWCStJjJIh8EmbLYB5/Nh9og4vMS
E (67151) SIM7600: 
max_ATTEMPTS 20

[... 145 lines cut ...]
>

Sending command: -- SIMCom Connected!
SIMCom Connected!
OK
I (140231) SIM7600: 
********input sent !!!!********

Sending command: -- AT+CMQTTCONNECT=0,"tcp://a23fja3lj6qx28-ats.iot.us-west-1.amazonaws.com:8883",60,1
AT+CMQTTCONNECT=0,"tcp://a23fja3lj6qx28-ats.iot.us-west-1.amazonaws.com:8883",60,1
OK
I (141831) SIM7600: 
********Connect to MQTT server

Sending command: -- AT+CMQTTTOPIC=0,31

AT+CMQTTTOPIC=0,31
+CMQTTTOPIC: 0,11

ERROR

+CMQTTCONNECT: 0,34
E (151891) SIM7600: 
max_ATTEMPTS 20
```

Oswaldott, 2023-06-03:
Is firmware update via ota or is it necessary to have a computer with windows? I am currently working on mac
My version of firmware is the same that @droidblastnz-> LE20B03SIM7600M22, please can you help me with this

droidblastnz, 2023-06-03:
> Thanks again, please confirm the process for uploading the cert to the SIM/Modem and what location does the cert go to?

Thanks again, please confirm the process for uploading the cert to the SIM/Modem and what location does the cert go to?

@Oswaldott  Is firmware update via ota how are you doing this?

Oswaldott, 2023-06-03:
@droidblastnz I followed the firmware update steps but i got this https://github.com/botletics/SIM7000-LTE-Shield/issues/41#issue-372693777 so i did this https://github.com/botletics/SIM7000-LTE-Shield/issues/41#issuecomment-642713893, but now i got this:
[image]

Any suggestions?
I thought @lewisxhe updated your firmware via ota, so the only way to update the firmware is through a computer with windows? and obviously follow the [steps](https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/docs/UpdateGuide/UpdateGuide.md)

droidblastnz, 2023-06-04:
> @droidblastnz I followed the firmware update steps but i got this [botletics/SIM7000-LTE-Shield#41 (comment)](https://github.com/botletics/SIM7000-LTE-Shield/issues/41#issue-372693777) so i did this [botletics/SIM7000-LTE-Shield#41 (comment)](https://github.com/botletics/SIM7000-LTE-Shield/issues/41#issuecomment-642713893), but now i got this: [image]
> 
> Any suggestions?

Same issue? No SIM inserted rechecked 
When trying to reflash drops to COM14 diag

[image]

[image]

lewisxhe, 2023-06-06:
The upgrade tool version I shared should be 1.6.7. Please use the upgrade tool in the update guide
https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/docs/UpdateGuide/UpdateGuide.md

droidblastnz, 2023-06-06:
> The upgrade tool version I shared should be 1.6.7. Please use the upgrade tool in the update guide https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/docs/UpdateGuide/UpdateGuide.md

Followed the guide and still cannot upgrade.
Rebooted tried again no new devices appears, refreshed device manager.
Tried a different Win 10 laptop same issue

Note your screen shot has two additional devices which are bluetooth?

`Unable to download Flash Programmer using Sahara Protocol, Please repower No1 DUT`

[image]

Start the tool 
[image]

lewisxhe, 2023-06-07:
Here is the video I recorded writing
https://youtu.be/7cJzjfcrFWY

lewisxhe, 2023-06-07:
@droidblastnz This is the feedback you got from using AT+SICOMATI? If so, then this version of firmware is not suitable for you, please check the firmware name, they are M22 and G22 respectively, please confirm the current hardware version you need to update the firmware

[image]

droidblastnz, 2023-06-07:
> AT+SICOMATI

Thanks  wrote the ATdebug.ino and then ran AT+SIMCOMATI with this response

```
20:55:46.773 -> 
20:55:46.773 -> +CPIN: READY
20:55:46.773 -> 
20:55:46.773 -> SMS DONE
20:55:46.773 -> 
20:55:46.773 -> PB DONE
20:55:49.991 -> 
20:55:49.991 -> OK
20:56:03.281 -> 
20:56:03.281 -> Manufacturer: SIMCOM INCORPORATED
20:56:03.281 -> Model: SIMCOM_SIM7600G-H
20:56:03.281 -> Revision: LE20B03SIM7600M22
20:56:03.281 -> SIM7600M22_B03V02_200828
20:56:03.281 -> QCN: 
20:56:03.281 -> IMEI: 868822041632330
20:56:03.281 -> MEID: 
20:56:03.281 -> +GCAP: +CGSM
20:56:03.281 -> DeviceInfo: 173,170
20:56:03.281 -> 
20:56:03.281 -> OK

```

Watched the video and started from scratch again.
All drivers are installed

Run the sim7080_sim7500_sim7600_sim7900_sim8200 qdl v1.67 only for update.exe and select the correct SIM7500&7600&7800Series then browse to the correct firmware location.

```
2023-06-07 14:13:52: -------------SIM7080_SIM7500_SIM7600_SIM7900_SIM8200 QDL V1.67 Only for Update(Build:Oct 14 2021 16:17:32)-----------
2023-06-07 14:13:52: Mobile Version report: PASS
2023-06-07 14:13:52: Compile Date: Aug 28 2020
2023-06-07 14:13:53: Compile Time: 16:31:24
2023-06-07 14:13:53: Release Date: Jul 17 2020
2023-06-07 14:13:53: Release Time: 11:00:00
2023-06-07 14:13:53: Version Dir: 9607.gen
2023-06-07 14:13:53: Station Class Mark: 42
2023-06-07 14:13:53: CAI Rev: 9
2023-06-07 14:13:53: Mobile Model: 255
2023-06-07 14:13:53: Firmware Rev: 100
2023-06-07 14:13:53: Slot Cycle Index: 2
2023-06-07 14:13:53: Vocoder major version: 0
2023-06-07 14:13:53: Vocoder minor version: 74
2023-06-07 14:13:53: Ext MSM HW Version: 0x4A0
2023-06-07 14:13:53: Ext Mobile Model#: 0x100C
2023-06-07 14:13:53: Ext Mobile Mobile SW Rev: MPSS.JO.2.0.2.c1.7-00052-9607_GENNS_PACK-1
2023-06-07 14:13:53: Ext Model ID: 
2023-06-07 14:13:53: Phone Operating Mode: ONLINE
2023-06-07 14:13:53: Phone Call State: CM_CALL_STATE_IDLE
2023-06-07 14:13:53: Phone System Mode State: SYS_SYS_MODE_WCDMA
2023-06-07 14:13:53: FTM Mode: OFF
2023-06-07 14:13:55: Waiting switch To Download Mode...
2023-06-07 14:13:59: Connecting Sahara Server COM15 1 times....
2023-06-07 14:14:04: saharaHandle=0x07c4, Version:0
2023-06-07 14:14:07: Connecting Sahara Server COM15 2 times....
2023-06-07 14:14:12: saharaHandle=0x07f8, Version:0
2023-06-07 14:14:14: Connecting Sahara Server COM15 3 times....
2023-06-07 14:14:19: saharaHandle=0x07f8, Version:0
2023-06-07 14:14:22: Connecting Sahara Server COM15 4 times....
2023-06-07 14:14:27: saharaHandle=0x07d4, Version:0
2023-06-07 14:14:29: Connecting Sahara Server COM15 5 times....
2023-06-07 14:14:34: saharaHandle=0x0780, Version:0
2023-06-07 14:14:37: Version: 0, MSMId: 0,SN: 0.
2023-06-07 14:14:37: Waiting Sahara_FlashProgrammer...
2023-06-07 14:14:37: Unable to download Flash Programmer using Sahara Protocol, Please repower No1 DUT.
```

Waiting switch To Download Mode...  both red LEDs go off then one comes back on but not as bright....
Driver 

[image]

QHSUSB__BULK
[image]

droidblastnz, 2023-06-07:
> @droidblastnz This is the feedback you got from using AT+SICOMATI? If so, then this version of firmware is not suitable for you, please check the firmware name, they are M22 and G22 respectively, please confirm the current hardware version you need to update the firmware
> 
> [image]

Mine is LE20B03SIM7600M22 and your video shows LE20B03SIM7600G22

lewisxhe, 2023-06-07:
ohh ? Then I was wrong, if you are M22, then B03 will be the latest version, I will consult SIMCOM to see if there is a version of B04

droidblastnz, 2023-06-07:
> ohh ? Then I was wrong, if you are M22, then B03 will be the latest version, I will consult SIMCOM to see if there is a version of B04

This was the firmware supplied LE20B04V03SIM7600M22 but again cannot get the SIM to upload as per your instructions.
[(https://www.dropbox.com/s/6lvooa2t5hyrdey/LE20B04V03SIM7600M22.rar?dl=0)]

lewisxhe, 2023-06-08:
I have arrived at a SIM7600 with the M22 version, I have a similar problem on the same side, and I have found this problem only for SIMCOM.

droidblastnz, 2023-06-08:
> 
Thanks so do you mean you have uploaded the B04 Firmware and same MQTT 8883 issue? I cant even get the firmware to load? Can you explain better please?

lewisxhe, 2023-06-08:
I also can't upgrade the firmware.

droidblastnz, 2023-06-08:
> 

Thats not good then. What is the next step? Cant upgrade the firmware, cannot connect to MQTT 8883 only 1883 which is a public MQTT instance and not secure? 

When you say can 'I also can't upgrade the firmware' you mean it fails at the ....
```

2023-06-07 14:13:55: Waiting switch To Download Mode...
2023-06-07 14:13:59: Connecting Sahara Server COM15 1 times....
2023-06-07 14:14:04: saharaHandle=0x07c4, Version:0
2023-06-07 14:14:07: Connecting Sahara Server COM15 2 times....
2023-06-07 14:14:12: saharaHandle=0x07f8, Version:0
2023-06-07 14:14:14: Connecting Sahara Server COM15 3 times....
2023-06-07 14:14:19: saharaHandle=0x07f8, Version:0
2023-06-07 14:14:22: Connecting Sahara Server COM15 4 times....
2023-06-07 14:14:27: saharaHandle=0x07d4, Version:0
2023-06-07 14:14:29: Connecting Sahara Server COM15 5 times....
2023-06-07 14:14:34: saharaHandle=0x0780, Version:0
```

lewisxhe, 2023-06-08:
There is no way to upgrade, and SIMCOM didn't reply me how to deal with it, I have no choice, I try to use B03 to connect to MQTTS port 8883, it is normal.
```
AT+CMQTTSTART

+CGREG: 0,1

OK

SMS DONE

OK

+CMQTTSTART: 0
AT+CCERTDOWN="ca_cert.pem",1338

>-----BEGIN CERTIFICATE-----
MIIDrzCCApegAwIBAgIQCDvgVpBCRrGhdWrJWZHHSjANBgkqhkiG9w0BAQUFADBh
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBD
QTAeFw0wNjExMTAwMDAwMDBaFw0zMTExMTAwMDAwMDBaMGExCzAJBgNVBAYTAlVT
MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxGTAXBgNVBAsTEHd3dy5kaWdpY2VydC5j
b20xIDAeBgNVBAMTF0RpZ2lDZXJ0IEdsb2JhbCBSb290IENBMIIBIjANBgkqhkiG
9w0BAQEFAAOCAQ8AMIIBCgKCAQEA4jvhEXLeqKTTo1eqUKKPC3eQyaKl7hLOllsB
CSDMAZOnTjC3U/dDxGkAV53ijSLdhwZAAIEJzs4bg7/fzTtxRuLWZscFs3YnFo97
nh6Vfe63SKMI2tavegw5BmV/Sl0fvBf4q77uKNd0f3p4mVmFaG5cIzJLv07A6Fpt
43C/dxC//AH2hdmoRBBYMql1GNXRor5H4idq9Joz+EkIYIvUX7Q6hL+hqkpMfT7P
T19sdl6gSzeRntwi5m3OFBqOasv+zbMUZBfHWymeMr/y7vrTC0LUq7dBMtoM1O/4
gdW7jVg/tRvoSSiicNoxBN33shbyTApOB6jtSj1etX+jkMOvJwIDAQABo2MwYTAO
BgNVHQ8BAf8EBAMCAYYwDwYDVR0TAQH/BAUwAwEB/zAdBgNVHQ4EFgQUA95QNVbR
TLtm8KPiGxvDl7I90VUwHwYDVR0jBBgwFoAUA95QNVbRTLtm8KPiGxvDl7I90VUw
DQYJKoZIhvcNAQEFBQADggEBAMucN6pIExIK+t1EnE9SsPTfrgT1eXkIoyQY/Esr
hMAtudXH/vTBH1jLuG2cenTnmCmrEbXjcKChzUyImZOMkXDiqw8cvpOp/2PV5Adg
06O/nVsJ8dWO41P0jmP6P6fbtGbfYmbW0W5BjfIttep3Sp+dWOIrWcBAI+0tKIJF
PnlUkiaY4IBIqDfv8NZ5YBberOgOzW6sRBc4L0na4UU+Krk2U886UAb3LujEV0ls
YSEY1QSteDwsOoBrp+uvFRTp2InBuThs4pFsiv9kuXclVzDAGySj4dzp30d8tbQk
CAUw7C29C79Fv1C5qfPrmAESrciIxpg0X40KPMbp1ZWVbd4=
-----END CERTIFICATE-----

OK
AT+CSSLCFG="sslversion",0,4

OK
AT+CSSLCFG="authmode",0,1

OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"

OK
AT+CSSLCFG="clientcert",0,"cert.pem"

ERROR
AT+CSSLCFG="clientkey",0,"key_cert.pem"

ERROR
AT+CMQTTSSLCFG=0,0

OK
AT+CMQTTREL=0

OK
AT+CMQTTACCQ=0,"Sim7600_fa7d2288",1

OK
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1

OK

+CMQTTCONNECT: 0,0

PB DONE

Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G-H
Revision: LE20B03SIM7600M22
SIM7600M22_B03V03_210525
QCN: 
SVN: 01
IMEI: 86882204249xxxxx
MEID: 
+GCAP: +CGSM
DeviceInfo: 173,170

OK

```

droidblastnz, 2023-06-08:
> 

Ok please can you try a private MQTT 8883 to HiveMQ not public but the free private connection and confirm?

HiveMQTT and emqx.io have a different certificate.

lewisxhe, 2023-06-08:
Whether I use B03 or B04 is the same problem, both return not authorized

JVeen, 2023-06-12:
i'm facing the same problems. i have a SIMCOM A7670E
 AT+SIMCOMATI gives me this:

```
AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B03A7670M7_F
A7670M7_B03V02_211123
QCN:
IMEI: 862205056091731
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK
```

I read somewhere that you need to enable SNI (server name indication) on hivemq

```
AT+CSSLCFG="enableSNI",0,1
OK
```
I can't see you're using this setting, maybe it helps!
I get a handshake failure.

```
AT+CMQTTCONNECT=0,"tcp://myinstance.s2.eu.hivemq.cloud:8883",60,1,my_username,my_password
+CMQTTCONNECT: 0,32

ERROR
```
I don't know where to look for a firmware update. Can i use the firmware you suggested?

droidblastnz, 2023-06-13:
> i'm facing the same problems. i have a SIMCOM A7670E AT+SIMCOMATI gives me this:
>
> ```
> AT+SIMCOMATI
> Manufacturer: INCORPORATED
> Model: A7670E-FASE
> Revision: A011B03A7670M7_F
> A7670M7_B03V02_211123
> QCN: [...]

Appears no firmware for my board I would try https://cloud-intl.emqx.com as it has a different cert than HiveMQ.

droidblastnz, 2023-06-14:
> i'm facing the same problems. i have a SIMCOM A7670E AT+SIMCOMATI gives me this:
>
> ```
> AT+SIMCOMATI
> Manufacturer: INCORPORATED
> Model: A7670E-FASE
> Revision: A011B03A7670M7_F
> A7670M7_B03V02_211123
> QCN: [...]

Suspect the HiveMQ cert is not support hence try cloud-intl.emqx.com as it has a different cert.
HiveMQ only needs the CA server pem cert and SNI but it does not work. Raised this with LilyGO and Simcom and the new firmware 1 I cannot load to the Simcom module and 2 has the same issue. 

So cut to the chase and try cloud-intl.emqx.com as [lewisxhe](https://github.com/lewisxhe) showed this to work. I have tested yet but will this week.

lewisxhe, 2023-06-15:
I can't do anything at the moment, I'm waiting for simcom's reply, I will test it as soon as I have news

droidblastnz, 2023-06-23:
> I can't do anything at the moment, I'm waiting for simcom's reply, I will test it as soon as I have news

Any update please?

lewisxhe, 2023-06-27:
Hi everyone,
I have not heard back from SIMCOM yet, I will follow up.

I will consult SIMCOM about the difference between LE11 and B20, thank you for @MisterRoy response

droidblastnz, 2023-06-28:
@lewisxhe @MisterRoy  can you please confirm how you are getting the cert to the module what steps/command you are using? Trying with emqx.com and they have a crt not pem.

bilabo438, 2023-06-28:
Hello i'm also trying updating SIM7600G-H to LE20B04V03SIM7600M22.
Actual:
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G-H
Revision: LE20B03SIM7600M22
SIM7600M22_B03V02_200828
QCN: 
IMEI: 868822040986638
MEID: 
+GCAP: +CGSM
DeviceInfo: 173,170

When I'am starting the upgrade tool, i will get following error:
1# Failed read imei failed. Please repower DUT.

Can anybody help me. Thanks
Best regards Daniel

droidblastnz, 2023-06-28:
> 

> There is no way to upgrade, and SIMCOM didn't reply me how to deal with it, I have no choice, I try to use B03 to connect to MQTTS port 8883, it is normal.
>
> ```
> AT+CMQTTSTART
>
> +CGREG: 0,1
>
> OK
>
> SMS [...]

Bit lost to how you are downloading the cert to the module.  Can you confirm what tool you are using, what the process is to get the file to module and actually where the pem is located? 

I have tried Ardinuo IDE, Teraterm and SSCOM V5.13.1 tool. Please detailed the steps and tool you are using as this is where my issue lies.

bilabo438, 2023-06-28:
I've got it.
First I have disconnected the Port from Arduino. 
Then I got the following error:
Connect Sahara Server failed.

The error is described here:
https://www.edaboard.com/threads/sim7600-lte-firmware-update.404060/

After Updating Firmware:
SIM_DATA: AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G-H
Revision: LE20B04SIM7600M22
SIM7600M22_B04V03_210527
QCN:
IMEI: 868822040986638
MEID:
+GCAP: +CGSM
DeviceInfo: 173,170

Now i also can connect to Azure IOT HUB.

Thanks to all

[41 more replies]