Xinyuan-LilyGO/T-SIM7600X #50: Modem wakes from sleep when ESP32 enters deep sleep

tadenham, 2023-05-09:
I have been able to get the modem to sleep by setting AT+CSCLK=1 and pulling DTR high. The modem will sleep as long as the ESP32 is awake or in light sleep. This was confirmed by the NETLIGHT shutting off and the modem failing to respond to AT commands until DTR has been pulled low again.

However, the modem will wake from sleep as soon as the ESP32 is put in deep sleep. I have verified that DTR maintains its high state during deep sleep, so something else is causing the modem to wake up. This happens when the board is powered by either USB or battery.

My application requires both the modem and ESP32 to enter sleep mode, otherwise the power consumption is too high. It's also not practical to shut the modem off completely as it takes too long to restart.

What could be the problem here?

lfdominguez, 2023-05-23:
You must set GPIO 4 HIGH before sleep

tadenham, 2023-05-24:
All that seems to do is shut down the modem completely instead of putting it to sleep.

lfdominguez, 2023-05-24:
But you must consider that when the esp32 deep sleep the GPIO can change the states.

lewisxhe, 2023-05-24:
I added a sleep test https://github.com/Xinyuan-LilyGO/T-SIM7600X/tree/master/examples/sleep

droidblastnz, 2023-07-04:
> I added a sleep test https://github.com/Xinyuan-LilyGO/T-SIM7600X/tree/master/examples/sleep

When ESP and Modem are sleeping what LEDs are still lite?