Xinyuan-LilyGO/T-SIM7600X #52: Problem starting the sim7600 module and the gps

manuelcm15, 2023-05-26:
Greetings, I recently bought a lilygo T-SIM7600G-H 16MB card and I have been experimenting with it, but on many occasions a problem happens to me when I initialize the sim7600 and if I go through that stage, the start of the GPS gives me an initial problem

I am using the tinyGSM library and copying the code of interest from the AllFunctions.ino example

    void setupModem(){
    // Set GSM module baud rate
      sim7000G.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);
      setupPin();
      delay(300);
      modemPowerOn();
      delay(500);
      connectToNet();
    }

    void setupPin(){
      pinMode(PWR_PIN, OUTPUT);
      pinMode(LED_PIN, OUTPUT);
    }

    void modemPowerOn(){
      digitalWrite(PWR_PIN, HIGH);
      delay(300); //Need delay
      digitalWrite(PWR_PIN, LOW);
    
      delay(300);
    
      if (!modem.init()) {
        Serial.println("Failed to init modem, delaying 10s and retrying");
        delay(10000);
        if(!modem.restart()){
           Serial.println("Failed to restart modem, delaying 10s and retrying");
           resetDevice();
          return;
        }
      }
    }

    void setupGPS(){
      if(!modem.enableGPS()){
        log("[ERROR] GPS no iniciado");
        w++;
      }
      else{
         log("[INFO] GPS Encendido");
         isGPSInit = true;
      }
      light_sleep(2);
    }

When I get false in modem.init() I proceed to print in the terminal and then execute modem.restart() but both methods give me false.

If luckily the modem starts correctly when it's time to start the GPS I get false and I don't know exactly the problem.

This problem happens very frequently when I use the USB cable and when I use a 3.7V 18650 battery.

lewisxhe, 2023-06-01:
please start here

https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/examples/AllFunctions/AllFunctions.ino

manuelcm15, 2023-06-14:
UPDATE: I have reviewed the example that you mention but I still have the same problem but I have noticed the following:

When connecting the card for the first time everything works fine but if I touch the reset button and start again the sim7600 does not respond it is as if it were stuck with some bad configuration of the restart, if I disconnected the power source and connected it again the sim7600 responds

The problem is when I restart and it generates this strange error, it happens the same when I restart the esp32 with the following code: 

           void resetDevice(){
            log("[ERROR] No se pudo conectar a la red telefonica, el dispositivo se va a reiniciar");
            Serial.flush(); 
            ESP.restart(); 
          }
          
If there is a problem and I need to reboot the device, how should I reboot the sim7600 modem?

antonio-emilio, 2023-11-20:
I'm having the same problem. Did you solved?

dimadolgopolovn, 2024-03-06:
For me it was a slightly different problem where the buffer wasn't emptied after crashing.
Running AT+CRESET helped in my case.