Xinyuan-LilyGO/T-SIM7600X #53: Missing link of the firmware LE20B04SIM7600G22.

OmerFarukOruc, 2023-06-09:
I am trying to follow the steps in the following youtube link: "https://www.youtube.com/watch?v=7cJzjfcrFWY".
I made LTE to work but cant get any signals from GPS. So saw #36 and #42 and followed the instructions but there is no download link for the recent firmware except i found this:  [LE20B03SIM7600M22.rar] (https://onedrive.live.com/?authkey=%21ADB6g5ro6RT-nwg&id=5D4ECE5EA53AE966%216438&cid=5D4ECE5EA53AE966&parId=root&parQt=sharedby&parCid=41E7A3AEE0C87943&o=OneUp).

Sell24, 2023-06-12:
Since you asked me for help previously, I thought I would see if I could help you out. 

Have you tried just trying to read the info? Here is what I do with my settings that makes it work. Just to be clear I am using the 7600NA version so it may be different from your board

```
AT+CGPSHOR=10  //Sets the GPS accuracy
AT+CGPS=1 //Turns on GPS (will error if it is already on so I do a AT+GPS? before but this is just an FYI)
AT+CGPSINFO //Returns a response as seen in the manual 
```
That is it. Maybe try and reset the modem to factory settings and just do these commands to see if it works? Do this all without using TinyGSM which I think you are doing but not 100% sure. Note it will take a bit before the GPS coordinates start to be valid but at least you will see if the GPS itself works

lewisxhe, 2023-06-13:
You can do it later, please check it out I've updated it
https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/42#issuecomment-1506991130

lewisxhe, 2023-06-13:
https://github.com/Xinyuan-LilyGO/T-SIM7600X/tree/master/examples/GPSDebug

dazemc, 2025-01-29:
I did find this firmware file, https://onedrive.live.com/?authkey=%21AB%2DeWK1OwkEZDe8&id=5D4ECE5EA53AE966%216437&cid=5D4ECE5EA53AE966&parId=root&parQt=sharedby&o=OneUp

I also have it uploaded here, https://github.com/dazemc/pi7600-lib/tree/master/firmware_tools/LE20B04SIM7600G22