Xinyuan-LilyGO/T-SIM7600X #54: Error occurs in FTP Upload example

tareqAmen, 2023-06-11:
When i was tried to upload file over FTP using this line:
`modem.sendAT("+CFTPSPUTFILE=\"", remotePath, filename, "\",3");`
The module response was ERROR without code of error  this error command mean the error maybe in parameter
To check the cause of the error i was tried to send AT+CFTPSPWD command to get the current directory on my ftp server and the response was ERROR too, so my problem now is how can i get the current directory in my server without any error.

Is there any one can help me to solve this problem, please.?

lewisxhe, 2023-06-13:
FTPUpload Good transport in our FTP service.

Manual import order See return results.
[image]

droidblastnz, 2023-06-14:
> 

My working example....

[  //FTP upload photo to FTP host
  modem.sendAT("+CFTPSPUTFILE=\"", f_name, "\",", 3);
  if (modem.waitResponse(30000) == 1) {
    if (modem.waitResponse(120000, "+CFTPSPUTFILE: ")) {
      int res = modem.stream.read();
      if (res == '0') {
        pDBGln("");
        pDBGln("Uploaded the image to EFS storage success");
      } else {
        pDBGln("");
        pDBGln("Uploading the image to EFS storage failed! errcode =");
        pDBGln(char(res));
        return ;
      }
    }
  } else {
    pDBGln("");
    pDBGln("Upload png to efs failed."); return ;
  }](url)

lewisxhe, 2023-06-15:
I can't help you anything, you can only rely on the error return code to find the cause, because I use my FTP server normally

tareqAmen, 2023-07-09:
> > 
> 
> My working example....
> 
> ```
> 
>   modem.sendAT("+CFTRANRX=\"e:/", f_name, "\",", ImgMetaData.imSize);        //works
>   modem.waitResponse(3000L);
>   while (theFile.available()) {
>     byte imgByte = theFile.read();
>     SerialAT.write(imgByte);
>   }
>   Serial.println("Time Send: " + String((float)(millis() - counter) / 1000));
>   theFile.close();
>   modem.waitResponse();
>   pDBGln("Finished transfering to EFS memory...");
> 
>  //FTP upload photo to FTP host
>   modem.sendAT("+CFTPSPUTFILE=\"", f_name, "\",", 3);
>   if (modem.waitResponse(30000) == 1) {
>     if (modem.waitResponse(120000, "+CFTPSPUTFILE: ")) {
>       int res = modem.stream.read();
>       if (res == '0') {
>         pDBGln("");
>         pDBGln("Uploaded the image to EFS storage success");
>       } else {
>         pDBGln("");
>         pDBGln("Uploading the image to EFS storage failed! errcode =");
>         pDBGln(char(res));
>         return ;
>       }
>     }
>   } else {
>     pDBGln("");
>     pDBGln("Upload png to efs failed."); return ;
>   }
> 
>   //FTP Logout
>   modem.sendAT("+CFTPSLOGOUT");
>   if (modem.waitResponse(3000) != 1) {
>     // Nothing needs to be done, just send this command,
>     // assuming it has been enabled
>   }
>   // Need to add some delay appropriately
>   delay(500);
> 
>   //FTP Stop
>   modem.sendAT("+CFTPSSTOP");
>   if (modem.waitResponse(3000) != 1) {
>     // Nothing needs to be done, just send this command,
>     // assuming it has been enabled
>   }
>   // Need to add some delay appropriately
>   delay(500);
> }
> ```
> 
> Note only port 21 works

Can you send me completely code for your example?

tareqAmen, 2023-07-09:
Using your example.. is the file uploaded to the server?

droidblastnz, 2023-07-10:
> Using your example.. is the file uploaded to the server?

Yes thats correct I worked it out via the AT commands. If you have issues run the command in the Arduino IDE and install a local copy of filezilla and watch the logs.

Mine device T-SIM7600 has uploads a image over FTP/4G when PIR is triggered.

tareqAmen, 2023-07-10:
> > Using your example.. is the file uploaded to the server?
> 
> Yes thats correct I worked it out via the AT commands. If you have issues run the command in the Arduino IDE and install a local copy of filezilla and watch the logs.
> 
> Mine device T-SIM7600 has uploads a image over FTP/4G when PIR is triggered.
> 
> I have a serially attached ESP32CAM that serially transfer the image to the master ESP which then copies to SDCard then EFS, once in EFS FTPs to the FTPS server.

what is the max upload data rate you had reached?

tareqAmen, 2023-07-10:
what is the max file size can i transfer to EFS?

droidblastnz, 2023-07-10:
> what is the max file size can i transfer to EFS?

No sure no documentation. Pay to try until it fails but what size are you looking at?

tareqAmen, 2023-07-13:
What is the maximum speed to upload a file using FTP?
and How can i increase the uplink speed?