Xinyuan-LilyGO/T-SIM7600X #56: update firmware

Mariem114114, 2023-06-16:
Hello @lewisxhe @LilyGO,
I am trying to install the firmware update for SIM7600, but at this step, I am not getting any output. I have tried following various YouTube videos, but I still don't see anything at this stage. What can I do?
[image]

lewisxhe, 2023-06-16:
Please do not use it in a virtual machine.

Mariem114114, 2023-06-16:
@lewisxhe 
I d'din't see what you mean, i use my pc with the os windows.

lewisxhe, 2023-06-16:
Are you using a virtual machine? Such as Vmware?

Mariem114114, 2023-06-18:
Non, non je n'utilise pas Vmware

lewisxhe, 2023-06-19:
This problem is that I'm currently using Vmware and I haven't met with my other win device.

LorenzoSiena, 2024-05-27:
Same problem "target Version" string (with strange chars) remain the same but menu is now selectable if you install vcredist_x86.exe (Microsoft Visual c++ 2010 x86 Redistributable,from the same folder or from Microsoft Site)