Xinyuan-LilyGO/T-SIM7600X #57: Power change of sim7600 in sleep mode!

thanhtin1859, 2023-06-26:
Hi everybody!
 Currently, I have a project related to using the sim7600E module to connect to the network.  I'm trying to save power by putting the sim module into sleep mode as follows:
 1. Send the sleep script: AT+CSCLK=1
 2. Set the DTR pin high to put the module to sleep.
 But it seems that the sim module does not sleep completely and the voltage is constantly jumping, anyone who has done it please tell me what are the effects that make it unable to sleep completely and how to fix it.  Thank you everyone for paying attention to my post!

[image]

https://github.com/Xinyuan-LilyGO/T-SIM7600X/assets/82450843/a03de152-f931-4452-98f2-a95487375100

thanhtin1859, 2023-07-03:
> 12V? How do you use it?

I design a power change circuit 12V to 4V use ic MP9486A! What problems do you see from it?

neuralprocessing, 2023-09-25:
Hi, you might are interested in this [thread](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/13)

thanhtin1859, 2023-09-28:
> Hi, you might be interested in this [thread](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/13)

Oh thank you! I've read it, but maybe this error is due to module sim of factory.