Xinyuan-LilyGO/T-SIM7600X #64: Solar Panel Only Charges Battery When Device is Off

anders624, 2023-08-14:
I'm working on a remote soil sensor that needs to be as able to run long term. I've run two field test so far and in both tests, the device died after 10 hours. I connected the battery terminals (without the battery in) to a voltmeter to read the Vbat while 6v was being fed to the solar connector. upon doing this, I found that the battery receives 4.2v when the device switch is in the off position. Then when the power switch is in the on position the voltage read just over 1v. 

Is this normal? If so, is there a way to change this to make the battery charge when the device is on? There isn't much info in the docs to explain what to do. This honestly seems like a pretty big design flaw for a board that's advertised as being for remote projects.

droidblastnz, 2023-08-16:
Assume you have read 

```
pin VIN is a USB input, if a battery is being used to power the device there will be no voltage output from VIN meaning 3.3v is the only power output.

When using the built-in battery pack the on/off switch will function as normal, supplying/cutting off power to the board, However, if an external battery pack is used and attached to the VBAT pin the on/off switch will be bypassed meaning the only way to shut off will be to disconnect the batteries.

On/off switch is for battery use only, it will not function when plugged into USB.

Battery voltage can not be read when plugged into USB using the onboard BAT ADC(35)

Recommended solar panels are 4.4v to 6v DO NOT EXCEED OR BOARD MAY BE DAMAGED
```

anders624, 2023-08-16:
[image]

Looking at the schematic, There's a mosfet with VIN connected to the gate which would explain why I'm only getting 1V to the battery when just the solar panel is connected.

In theory, joining the SOLARIN to VIN would power the gate and allow me to charge the battery, but I need to test it to be sure. Seems like a weird design choice if you ask me.

I'll report back after testing.

anders624, 2023-08-16:
After testing, I can report that connecting the VIN to the SOLARIN works. I will say that the design choice baffles me. Maybe the mosfet protects the the circuit somehow?

droidblastnz, 2023-08-16:
> [image]
>
> Looking at the schematic, There's a mosfet with VIN connected to the gate which would explain why I'm only getting 1V to the battery when just the solar panel is connected.
>
> In theory, [...]

@lewisxhe any idea?

lewisxhe, 2023-12-11:
I think this design means that the CN3065 can be charged using both USBC and the solar panel. If the USBC connection is detected, the solar input will be cut off.

lewisxhe, 2023-12-11:
If the board is not set to sleep for a long time, it is impossible to use solar energy for power supply. The current provided by the solar panel is limited. If you want to make reasonable use of solar energy, you must set the board to sleep mode to maximize the use of solar panel charging. Because just running esp32 even without WiFi and Bluetooth, the current consumption is almost 40~70mA.

agteigarnet, 2024-01-05:
I can add to this if it is helpful.
I have been running T-SIM7600E for 2 years on solar panel unattended remotely and it works great.
And if USB is connected then charging is from the USB.

DomolinBolivia, 2024-01-08:
> Después de las pruebas, puedo informar que conectar el VIN al SOLARIN funciona. Diré que la elección del diseño me desconcierta. ¿Quizás el Mosfet protege el circuito de alguna manera?
> 
> Editar: Después de intentar implementar esto en mi placa principal, descubrí que la placa no funciona independientemente del estado de los interruptores de alimentación, solo se carga. Esto me dice que la placa tiene dos modos cuando está encendida, cargando y funcionando. No pueden ser ambas cosas.
> 
> A menos que los desarrolladores tengan alguna solución mágica para resolver este problema, debo concluir que la carga solar no funciona para operaciones remotas ya que la placa debe apagarse manualmente para cargar la batería.

I have the same problem, the battery does not charge.

rlimaboldmss, 2024-07-24:
> I have the same problem, the battery does not charge.

Hi I am having similar issue, with deepsleep seems solar charging is not working.
Do you solve this?

agteigarnet, 2024-07-24:
Deep sleep solar charging works fine on my devices.