Xinyuan-LilyGO/T-SIM7600X #65: T-SIM7600G Call + SMS

NicoYaez, 2023-08-25:
Hello, good day. I would like to know if someone implemented calls on the 7600. What happens to me is that I would like to end the call after a few seconds so that the subsequent actions can execute correctly, such as sending a message. However, since the call is not disconnected, the message remains in a queue and gets sent some time after the call, causing a delay in the entire 'for' cycle. If anyone can help me with this, I would appreciate it. Here's the code

    for (int i = 0; i < sizeof(phoneNumbers) / sizeof(phoneNumbers[0]); i++) {
 
    DBG("Llamando:", phoneNumbers[i]);
    SerialAT.println("ATD" + String(phoneNumbers[i]) + ";");
    modem.waitResponse();
    light_sleep(20);
    SerialAT.println("ATH");
    modem.waitResponse();
    delay(5000);
  
    bool success = modem.sendSMS(phoneNumbers[i], message); // send SMS
    Serial.println(success ? "SMS sent successfully" : "SMS sending failed");
    // Wait for the SMS to be sent
    delay(10000);
    }

droidblastnz, 2023-08-28:
Check the manual 

```
AT+CVHU=0
OK
ATH
VOICE CALL: END: 000017
OK
```

NicoYaez, 2023-08-29:
Thank you very much, everything worked correctly for me, you went too far, you are a crack, thank you.

closed 2023-08-29 by NicoYaez