Xinyuan-LilyGO/T-SIM7600X #66: Unable to send HTTP POST payload of size greater than 1449 characters with SIM7600+ESP32

Eya-hm, 2023-08-29:
Attached is the code, it works with a maximum of 40 strings of 36 characters each, which I retrieve from flash memory, put in a JSON array and serialise afterwards.  

//Start GSM:
 SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
 SerialMon.println("Initializing modem...");

JsonArray data = JSONbuffer.createNestedArray("data");

for (int j =0; j <72; j++)
{    

    String key = "my-key-" + String(j);

    // Récupérer la chaîne à partir de la clé
    String storedValues = preferences.getString(key.c_str(), "");
    

    Serial.println(storedValues);
    data.add(storedValues);   
}
serializeJson(JSONbuffer,data_msg);
modem.init();
  SerialMon.print(F("Connecting to "));
   SerialMon.print(apn);
   if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
      SerialMon.println(" fail");
      delay(10000);
      return;
    }
   SerialMon.println(" success");

   if (modem.isGprsConnected()) {
      SerialMon.println("GPRS connected");
  }

  SerialMon.print("Connecting to ");
  SerialMon.println(server);
  if (!client.connect(server, port)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" success");

String url = "";
String jsonPayload = data_msg;
  int status = client.connect(server, 80);
  if (status == 1) {
    client.println("POST " + url + " HTTP/1.1");
    client.println("Host:"+ String(server));
    client.println("Content-Type: application/json");
    client.print("Content-Length: ");
    client.println(jsonPayload.length());
    client.println();
    client.println(jsonPayload);
    delay(1000);
}
  while (client.available()) {
    String line = client.readStringUntil('\r');
    Serial.println(line);
  }
  client.stop();

Anyone have any ideads? thank you

droidblastnz, 2023-08-29:
Maybe try using chunked transfer encoding for larger payloads?

```
// Start GSM:
SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
SerialMon.println("Initializing modem...");

JsonArray data = JSONbuffer.createNestedArray("data");

for (int j = 0; j < 72; j++) {
    String key = "my-key-" + String(j);
    // Récupérer la chaîne à partir de la clé
    String storedValues = preferences.getString(key.c_str(), "");
    Serial.println(storedValues);
    data.add(storedValues);   
}

serializeJson(JSONbuffer, data_msg);
modem.init();
SerialMon.print(F("Connecting to "));
SerialMon.print(apn);
if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
}
SerialMon.println(" success");

if (modem.isGprsConnected()) {
    SerialMon.println("GPRS connected");
}

String url = "";
String jsonPayload = data_msg;
int status = client.connect(server, port);

if (status == 1) {
    // Construct the HTTP POST request
    client.print("POST " + url + " HTTP/1.1\r\n");
    client.print("Host: " + String(server) + "\r\n");
    client.print("Content-Type: application/json\r\n");
    
    // Using chunked transfer encoding for larger payloads
```

Eya-hm, 2023-08-30:
it worked well thank you

closed 2023-08-30 by Eya-hm

davidUKYFU, 2024-03-27:
> it worked well thank you

Hello, I am trying something similar to you but the sever responds with a 400 Bad Request. My code is the next one:

```
client.print("POST /telemetria/post.php HTTP/1.1\r\n");
    client.print("Host: " + String(serverName) + "\r\n");
    client.print("Transfer-Encoding: chunked\r\n");
    client.print("Content-Type: application/x-www-form-urlencoded\r\n");
    client.print("\r\n");
    client.print(pendienteRed);
    client.print("\r\n");
```
(pendienterRed is the data that I need to send)

Could you share how did you get it? Thank you

droidblastnz, 2024-03-27:
@davidUKYFU  hope this helps

```
client.print("POST /telemetria/post.php HTTP/1.1\r\n");
client.print("Host: ");
client.print(serverName);
client.print("\r\n");
client.print("Content-Type: application/x-www-form-urlencoded\r\n");
client.print("Content-Length: ");
client.print(pendienteRed.length()); // Assuming pendienteRed is a String
client.print("\r\n");
client.print("\r\n");
client.print(pendienteRed);
```
Make sure `pendienteRed` contains the data you want to send in the POST request. Also, ensure that `serverName` is the correct hostname or IP address of your server?

davidUKYFU, 2024-04-04:
> @davidUKYFU hope this helps
>
> ```
> client.print("POST /telemetria/post.php HTTP/1.1\r\n");
> client.print("Host: ");
> client.print(serverName);
> client.print("\r\n");
> client.print("Content-Type: [...]

My server was returning 406 Error with your code😬. Also I was trying to get to send packages larger than 1000 characters. I finally got all with a similar code. I let it here so that if helps someone:

```
SerialMon.print("Connecting to ");
SerialMon.println(serverName);
if (!client.connect(serverName, 80)) {
  SerialMon.println(" fail");
  if (!modem.isNetworkConnected()) { conexion(); }  //"conexion" is the function that establishes data connection

} else {
  SerialMon.println(" success");
  client.print("POST " endpoint " HTTP/1.1\r\n");
  client.print("Host: ");
  client.print(serverName);
  client.print("\r\n");
  client.print("User-Agent: Arduino/2.2.0\r\n");
  client.print("Connection: close\r\n");
  client.print("Content-Type: application/x-www-form-urlencoded\r\n");
  client.print("Content-Length: ");
  packetLength = data.length();
  client.print(data.length());
  client.print("\r\n");
  client.print("Connection: keep-alive\r\n");
  client.print("\r\n");

  //Sending data in small packets equal or sorter than 10000 characters
  int i = packetLength % 1000;
  client.print(data.substring(0, i));
  while (i < packetLength) {
    client.print(data.substring(i, i + 1000));
    i = i + 1000;
  }

  Serial.println("Data send");

  // server response

  Serial.println("Wating for server response...");
  delay(20); //Dont know if is necesary
  String response = client.readString();
  Serial.println("Server response :");
  Serial.println(response);
```