Xinyuan-LilyGO/T-SIM7600X #67: GPS not Working Lilygosim7600E-H

sandeepkumar449, 2023-09-04:
hi 
i am using Lilygosim7600E-H to get GPS Location in NMEA But Not getting any Location 
I (0) cpu_start: App cpu up.
I (231) cpu_start: Pro cpu start user code
I (231) cpu_start: cpu freq: 160000000 Hz
I (231) cpu_start: Application information:
I (235) cpu_start: Project name:     nmea_parser
I (241) cpu_start: App version:      v5.2-dev-2383-g82cceabc6e-dirty
I (248) cpu_start: Compile time:     Sep  4 2023 14:54:34
I (254) cpu_start: ELF file SHA256:  cbc964530...
I (259) cpu_start: ESP-IDF:          v5.2-dev-2383-g82cceabc6e-dirty
I (266) cpu_start: Min chip rev:     v0.0
I (271) cpu_start: Max chip rev:     v3.99 
I (276) cpu_start: Chip rev:         v3.0
I (281) heap_init: Initializing. RAM available for dynamic allocation:
I (288) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (294) heap_init: At 3FFB29C0 len 0002D640 (181 KiB): DRAM
I (300) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (306) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (313) heap_init: At 4008D2D8 len 00012D28 (75 KiB): IRAM
I (320) spi_flash: detected chip: generic
I (324) spi_flash: flash io: dio
I (328) app_start: Starting scheduler on CPU0
I (333) app_start: Starting scheduler on CPU1
I (333) main_task: Started on CPU0
I (343) main_task: Calling app_main()
I (343) uart: queue free spaces: 16
I (343) nmea_parser: NMEA Parser init OK

droidblastnz, 2023-09-11:
Are you inside or outside?

OmerFarukOruc, 2023-10-11:
> Are you inside or outside?

Do I need to be outside in order to have a signal? If so how many minutes approx. it connects to GPS?

droidblastnz, 2023-10-15:
Suggest you try outside based on you comment _Lilygosim7600E-H to get GPS Location in NMEA But Not getting any Location_ and time will depend on where you are and what you can see sat wise for the GPS.

Check the manual;
Output of NMEA sentences is automatic; no control via AT commands is provided. If executing
AT+CGPS=1, the GPS session will choose cold or hot start automatically.
UE-based and UE-assisted mode depend on URL (AT+CGPSURL). When UE-based mode fails, it
will switch standalone mode.
UE-assisted mode is singly fixed. Standalone and UE-based mode is consecutively fixed.
After the GPS closed, it should to wait about 2s~30s for start again. Reason: If the signal
conditions are right (strong enough signals to allow ephemeris demodulation) or ephemeris
demodulation is on going, sometimes MGP will stay on longer in order to demodulate more
ephemeris. This will help the engine provide faster TTFF and possibly better yield later (up to 2
hours), because it has the benefit of more ephemeris available.
For SIM7600E-H-M2/SIM7600SA-H-M2/SIM7600A-H-M2 module, GPS started should be decided
by the physical switch of GPS flight mode in the module firstly. Close the switch, GPS will be
started automatically, then you can open or close gps by AT command, otherwize, GPS could not
be started in any way

Again reading the manual,
AT+CGPSFTM Start GPS test mode

```
AT+CGPSFTM=1
OK
$GLGSV,78,20.6,66,25.6,77,21.6,79,21.9,67,26.2,68,23.6
$GPGSV,10,36.3,12,33.5,14,26.5,15,27.0,18,30.6,20,29.4,21,14.9,
24,32.8,25,30.6,31,29.1,32,27.0
$BDGSV,201,28.7,204,29.0,206,27.3,207,25.9,209,25.0,210,18.5
```

If test mode starts, the URC will report on AT port, Modem port and UART port.
If testing on actual signal, <SV> should be ignored, and GPS must be started by AT+CGPS,
AT+CGPSCOLD or AT+CGPSHOT.
If testing on GPS signal simulate equipment, <SV> must be choiced, and GPS will start
automatically.
URC sentence will report every 1 second

`#include <iostream>
#include <fstream>
#include <string>
#include <thread>
#include <chrono>
#include <serial/serial.h> // You might need a serial library for communication

int main() {
    // Define the serial port settings
    serial::Serial serialPort;
    serialPort.setPort("/dev/ttyUSB0"); // Change this to your specific serial port
    serialPort.setBaudrate(9600);
    serialPort.setTimeout(serial::Timeout::simpleTimeout(1000));

    try {
        serialPort.open();
    } catch (std::exception &e) {
        std::cerr << "Error opening the serial port: " << e.what() << std::endl;
        return 1;
    }

    // Enable the GPS module
    serialPort.write("AT+CGPSPWR=1\r\n");

    // Wait for a moment before sending the next command
    std::this_thread::sleep_for(std::chrono::seconds(2));

    // Set NMEA output format
    serialPort.write("AT+CGPSOUT=32\r\n");

    // Wait for a moment before sending the next command
    std::this_thread::sleep_for(std::chrono::seconds(2));

    // Read and print GPS data
    while (true) {
        std::string data = serialPort.read(1024); // Adjust the buffer size as needed
        std::cout << data;

        // Check for a GPS fix
        if (data.find("$GNGGA") != std::string::npos || data.find("$GPGGA") != std::string::npos) {
            std::cout << "GPS Fix Acquired!" << std::endl;
            break;
        }

        std::this_thread::sleep_for(std::chrono::seconds(1)); // Adjust the interval as needed
    }

    // Continuously read and print GPS data
    while (true) {
        std::string data = serialPort.read(1024); // Adjust the buffer size as needed
        std::cout << data;
        std::this_thread::sleep_for(std::chrono::seconds(1)); // Adjust the interval as needed
    }

    // Close the serial port (optional)
    serialPort.close();

    return 0;
}
`