Xinyuan-LilyGO/T-SIM7600X #70: Battery and charging

Bubleelectronic, 2023-10-26:
Hi, I have a question about the use of battery. I've read from other issues using a USB-C charger with the battery connected will  power the borad and  charge the battery, but if the charger is disconnected, the board will tourn off ?. The switch sould be ON so it will still powered by the batterry ?. Is there any protection for overcharging?

droidblastnz, 2023-11-30:
readme 

pin VIN is a USB input, if a battery is being used to power the device there will be no voltage output from VIN meaning 3.3v is the only power output.

When using the built-in battery pack the on/off switch will function as normal, supplying/cutting off power to the board, However, if an external battery pack is used and attached to the VBAT pin the on/off switch will be bypassed meaning the only way to shut off will be to disconnect the batteries.

On/off switch is for battery use only, it will not function when plugged into USB.

Battery voltage can not be read when plugged into USB using the onboard BAT ADC(35)

suggest you try each scenario to further understand what occurs.

snowgato, 2025-06-30:
Hi, thus there's no way to measure the built-in battery pack voltage  with built in ADC when the board is connected to pc using  USB C  named TTL ?

droidblastnz, 2025-07-01:
Pin 35 (ADC1_CH7) is wired to a voltage divider that reads from the VBAT line, which is only powered when:

- A battery is connected to the battery JST connector, AND
- The board is being powered from the battery.

When you plug in USB-C, the system shifts power input from VBAT to VUSB/5V rail (which bypasses VBAT). The voltage divider on VBAT becomes disconnected or gets pulled low.

Therefore, ADC35 will either float or read 0V when USB is connected, even if the battery is present.

You cannot monitor battery voltage during USB-connected serial operation, such as when using Serial Monitor for logging/debugging via the USB-TTL port.

The on/off switch only works when battery-powered, not when USB is powering the board.

Workarounds.

Use external UART (Serial) connection:

-    Disconnect USB-C.
-    Power the board from the battery.
-    Connect external USB-TTL (like FTDI adapter) to GPIO1/3 (TX/RX) for debugging via serial.

Log battery voltage to external interface:

    E.g., use MQTT, BLE, or SD card logging for voltage if USB-C is needed for power or charging.