Xinyuan-LilyGO/T-SIM7600X #76: USB power and Battery power switchover

SmalltimeSphere, 2024-01-06:
I was hoping someone could help me.

I have the T-SIM7600 board and I have created a small prohram to text me when there is battery voltage >1 and when the voltage is =0

Basically a hot power messaging program. So when the USB is turned off (Either USB port) I get a message with the voltage reading of the battery. This tells me that the unit is being powered by the battery. When The USB power is present again, I get a message saying that power is restored (Not using the battery to power the device).

The issue I am having is that when the USB powre is switched off, the board resets. This is an unwanted behaviour as it then has to reload the whole program, log on to the network, and finally send me the "Device using battery power" message.

When the power is switched back on there is no such restart and the USB power takes over.

Does anyone have any idea how to solve this restarting issue, or is this typical behaviour of the unit? I would really appreciate the communities help on this.

lewisxhe, 2024-01-07:
The hardware circuit determines that the power supply cannot be switched seamlessly, and it is certain that the modem will restart during the switching process.

SmalltimeSphere, 2024-01-07:
Thank you for your reply. I have tested with other programs and the behaviour is the same.

I checked out the schematic (T54_SIM7600_20201012-1.pdf) and I'm a little lost, But I Believe it's down to the switching time when the USB power is lost. I keep the cable connected, I just switch it off at the wall.

Is there anything I can do to stop this from happening? I've seen some posts regrading different boards where a 2200uF cap can solve this. Any ideas are welcome.

I suppose my program still works, There is just a long delay of receiveing the message when the board restarts as it has to restart the program. At the moment it can be anywhere up to a minute before I get an SMS when USB power is lost and it switches to the batery.