Xinyuan-LilyGO/T-SIM7600X #78: SIM7670X doesn't connect to network due to wrong APN (TinyGSM fork issue?)

brettbeeson, 2024-01-12:
I had a problem with a SIM7670SA running via a [lilygo32 ](https://www.lilygo.cc/products/t-sim-a7670e) board.  I'm Australia using Telstra.) 

**My question is: The 'fork' of TinyGSM (referenced in the README) is maybe out of date with an old bug (see below). What's the best way to 'update' TinyGSM. I'm wary due to the warning " If you use the master branch, the compilation will not go smoothly"?**

The problem I see is that the modem doesn't connect (either status 3 (denied) or 11 (emergency only) or 0 (disconnected). 

Here is an annotated log of the problem and work around solution, using the [AT_DEBUG](https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/examples/ATdebug/ATdebug.ino)  to manually run AT commands.

Short version: set the APN 'manually' with `AT+CGDCONT=1,"IP","telstra.internet","",0,0,,,,` and try disabling automatic mode in favour of LTE only: `AT+CNMP=38` 

```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
Start Sketch
Trying baud rate 115200
Modem responded at rate:115200
***********************************************************
 You can now send AT commands
 Enter "AT" (without quotes), and you should see "OK"
 If it doesn't work, select "Both NL & CR" in Serial Monitor
 DISCLAIMER: Entering AT commands without knowing what they do
 can have undesired consiquinces...
***********************************************************

AT
OK
AT+CSQ
# Signal rssi=19
+CSQ: 19,99
OK

PB DONE
AT+CGREG?
# 0=no network?, 11=emergency only
+CGREG: 0,11

OK

+CGEV: MDE ETACH

# fails even after a long time
AT+CGREG?
+CGREG: 0,0

# correct
OK
AT+CFUN?
+CFUN: 1

# change to LTE only (not 2=auto) and reset
AT+CNMP=38
OK

# reset and try
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
AT+CSQ
+CSQ: 21,99

OK
AT+CGREG?
# With LTE (set via AT+CNMP=38), connection is now "forbidden" not 11: "emergency only"
+CGREG: 0,3

OK
AT+CSQ
+CSQ: 21,99

# current operator: 0=automatic, 2=numeric, 50501=telstra,7=EUTRAN (???)
# ok
OK
AT+COPS?
+COPS: 0,2,"50501",7

OK
AT+COPS=?
+COPS: (1,"Telstra Mobile","Telstra","50501",7),(3,"Optus AU","Optus","50502",7),(3,"vodafone AU","voda AU","50503",7),,(0,1,2,3,4),(0,1,2)

# Wait a few minutes
# Network changes from 3 (denied) to 11 (emergency)
AT+CGREG?
+CGREG: 0,11

OK

AT+CSQ
+CSQ: 21,99

# Check bands  <mode>[,<lte_mode>][,<lte_modeExt>]
+CNBP?
+CNBP: 0X0000000000280180,0X00000000080000DF,0X0002
# check bitwise operators on datasheet
# LTE BANDS ARE: 1,2,3,4,5,7,8,28 as per datasheet on lilygo and
# 0X0002 means band66 as per datasheet
# so: as expected

AT+CGDCONT?
+CGDCONT: 1,"IP","Internet","",0,0,,,,

OK
# **this is incorrect (should be telstra.internet!)**

# set to APN correctly
AT+CGDCONT=1,"IP","telstra.internet","",0,0,,,,

# Reset and it connects!
AT+CRESET
OK
*ATREADY: 1
+CPIN: READY
SMS DONE
+CGEV: EPS PDN ACT 1
+CGEV: ME PDN DEACT 8
PB DONE
#  "1" means registered
+CGREG: 0,1

lewisxhe, 2024-01-12:
You are using A7670X, you should ask questions here https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX

As the readme says, TinyGSM in the lib directory has been modified and is not an upstream branch.

Using LilyGO-T-A76XX should not be updated through the library manager, only use TinyGSM within the LilyGO-T-A76XX lib directory

lewisxhe, 2024-01-12:
Some countries need to configure APN to allow registration. This is a known issue. So is your problem solved now?

brettbeeson, 2024-01-15:
@lewisxhe : OK: I've posted to LilyGo. 

Yes, (in Australia) I need to configure the APN  (to "telstra.internet" in my case) and ensure the modem was updated (as per the above process). My problem is solved now :)

Thanks for your assistance and a great library!

closed 2024-01-15 by brettbeeson