Xinyuan-LilyGO/T-SIM7600X #87: Max voltage on solar panel

agilly, 2024-05-01:
In the specs, you write that the board supports 4.4V to 6V solar panels. It says that voltage above 6V may damage the board.

But 6V solar panels can have open circuit voltages up to 7V or more when they are in direct sunlight. So does it mean:

A. Is it safe to use a 6V solar panel, even if sometimes the voltage may go above 6V
B. The board will be damaged if the voltage is above 6V, so we should use solar panels rated for 4.5V or 5V instead, *not* 6V.

jh2000, 2024-05-15:
As you can see [here](https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/schematic/T54_SIM7600_20201012.pdf), there is a 6.8V over voltage protection diode. Since there are no specs on how much current it can withstand we can not know for sure, but at least there should be no problem with a 6V panel with open circuit voltage over 6V, unless it is oversized and can deliver too much current even on 7V to blow that diode up.

agilly, 2024-05-16:
Thank you! At least I know that if solar charging suddenly stops working, I should be replacing D3 :)

closed 2024-05-16 by agilly