Xinyuan-LilyGO/T-SIM7600X #88: Charge status and battery percents

KlimRider, 2024-05-04:
Dear community and LilyGo Team
I try to start AllFunctions example and on section about battery is all 0
Only voltage is show actual data

Here is example of output^
+CPIN: READY

OK
AT+CBC

+CBC: 3.843V

OK
[2646] Battery charge state: 0
[2646] Battery charge 'percent': 0
[2647] Battery voltage: 3.84
AT+CPMUTEMP

Can i get actual data about status and percentage of battery?
What i need to do?

bartdereu, 2024-12-18:
I went out from the fact that fully loaded is 4.2v . If you read 4.0 , you can calculate the percentage by 4/4.2*100

lewisxhe, 2024-12-19:
Well. SIM7600 does not provide percentages, you need to calculate it yourself, and this voltage is fixed, it is meaningless to get it, this is not the battery voltage