Xinyuan-LilyGO/T-SIM7600X #91: How to Perform Firmware Update on ESP32 Using T-SIM7600X

rajmehta28599, 2024-05-21:
I am looking for a detailed guide on how to update the firmware of an ESP32 microcontroller using a T-SIM7600X module. Specifically, I need information on:

1. Required hardware and software tools for the firmware update process.
2. Steps to configure the T-SIM7600X for use with the ESP32.
3. Methods to upload the firmware to the ESP32 via the T-SIM7600X module.
4. Troubleshooting common issues encountered during the firmware update.
5. Best practices for ensuring a smooth and secure firmware update process.

Any examples, code snippets, or links to comprehensive tutorials would be greatly appreciated.

AleReb, 2024-06-06:
i don't understand you whant the defautl code or a new code? firmware in a microcotroler its like a program, and usually you said firmware for back code

rajmehta28599, 2024-06-07:
I hope you're familiar with the ESP32 OTA update feature, which allows for OTA updates over Wi-Fi.

That same thing I want to perform using T-SIM7600X. without ESP32 Wi-Fi.

I appreciate your response.

Thank you

AleReb, 2024-06-10:
I understand the process, but initially, you must work with AT commands to download data from a server, then copy the information to EEPROM, followed by an OTA update. I’m uncertain if the available RAM will suffice for the data download. If time permits, this project seems intriguing.

AleReb, 2024-06-10:
totaly feasible i find some librarys but i can try https://github.com/mpagnoulle/GSMOTAUpdater righ now cheers