Xinyuan-LilyGO/T-SIM7600X #94: [Question] - Pin 0 of NAU8810 chip in T-SIM7600X schematic sheet

PsySc0rpi0n, 2024-07-15:
In the following file:
https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/schematic/T54_SIM7600_20201012.pdf

In page 2, there is a section with the audio chip NAU8810. According to the datasheet of this chip, there is no pin 0 (GND).
However in the said schematic, there is a pin 0. What is the purpose of this pin?
[image]

From the datasheet here:
https://www.nuvoton.com/resource-files/NAU8810_Datasheet_Rev_2.8.pdf
page 98, there is no such pin.
[image]

X-Ryl669, 2024-09-18:
It's likely the thermal pad pin ("metal paddle VSSA" in the datasheet) that's used for connecting the chip thermal contact to the ground plane on the PCB. Just guessing here...

lewisxhe, 2024-09-19:
@X-Ryl669 You are right, it is the heat sink pad at the bottom of the chip

closed 2024-09-23 by PsySc0rpi0n