Xinyuan-LilyGO/T-SIM7600X #97: Reading Battery Voltage While USB is Plugged In

CanselYalnkaya, 2024-08-06:
While my device is plugged into USB, I want to read the voltage to see if the battery is charging, but with the code below, I can only read 7.26V. Is there a way to read the battery voltage while it is plugged into USB?

`#include <driver/adc.h>
#include "esp_adc_cal.h"

void setup() {
Serial.begin(115200);
esp_adc_cal_characteristics_t adc_chars;
esp_adc_cal_value_t val_type = esp_adc_cal_characterize((adc_unit_t)ADC_UNIT_1, (adc_atten_t)ADC_ATTEN_DB_2_5, (adc_bits_width_t)ADC_WIDTH_BIT_12, 1100, &adc_chars);
pinMode(14, OUTPUT);
}

void loop() {
digitalWrite(14, HIGH);
delay(1);
float measurement = (float) analogRead(34);
float battery_voltage = (measurement / 4095.0) * 7.26;
digitalWrite(14, LOW);
Serial.println(battery_voltage);
}`

bartdereu, 2024-12-18:
No ,there is not. It's explained in the manual.

lewisxhe, 2024-12-19:
Due to hardware limitations, when the USB is plugged in, the power source switches to USB, and the voltage detected at this time is no longer the battery voltage.