Xinyuan-LilyGO/T-SIM7600X #98: T-SIM7600X FTP Upload from SD card or delete Files from ESF

DanielKuenzler, 2024-08-08:
hello togheter

I have tried your example FTPUpload.ino from the GIT page and I can send the files to the FTP. Does this also work directly from the SD card or does the file always have to be transferred to the ESP memory first? 
I have now spent hours trying to use parameter 2 for the PutFile, but I always get error 1. It can't be the FTP connection, as parameter 3 can be used to send over the ESP.

I then also tried to delete the file which I wrote to the ESP with your code "writeFileToEFS(const char *filename)" - unfortunately also in vain. I was not able to delete the files from the EFS again.

I hope you can help me, as I have already spent so many hours trying and have not found anything useful on the internet.

if you could send me some sample code, I would be grateful

thank you very much
Daniel

DanielKuenzler, 2024-08-09:
has anyone an example code on how to delete files from the EFS in the module so that the memory is free again?

lewisxhe, 2024-09-09:
[image]