Xinyuan-LilyGO/T-SIM7600X #99: Error occured during connecting to TLS secured MQTT broker ( 34 – open SSL session failed)

Pasan1021, 2024-08-14:
I have tried to connect to mqqt.datacke.co broker which provides TLS-enabled security. In this process, I have downloaded the CA signed server certificate to the SIM7600 module and configure the server root CA of specified SSL context. Both these commands are returns successfulll but SIM7600 module didn’t connect to the MQTT broker and it returns the error number 34 which identifies the error as open SSL session failed. In the below I have attached my code and screenshots of the results. 

**Firmware of the SIM7600 G-H module**
Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G-H
Revision: SIM7600M22_V2.0.1   (identification of firmware)
SVN: 01
+GCAP: +CGSM 

**Code** 

#include <Arduino.h>
#include <Wire.h>
#include <WiFi.h>  // Include the WiFi library for MAC address
#include <ArduinoJson.h>
//#include "Secret.h" // Include the file to get the username and pasasword of MQTT server
#include"datacake.h"

String gsm_send_serial(String command, int delay);

#define SerialMon Serial
#define SerialAT Serial1
#define GSM_PIN ""

#define D0 36
#define D1 35
#define D2 32
#define D3 33
#define R0 4

// Your GPRS credentials, if any
const char apn[] = "dialogbb";
const char gprsUser[] = "";
const char gprsPass[] = "";

// MQTT details
String broker = "mqtt.datacake.co";
String MQTTport = "8883";

String username = "";
String password = "";

#define uS_TO_S_FACTOR 1000000ULL // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP 60          // Time ESP32 will go to sleep (in seconds)

#define UART_BAUD 115200

#define MODEM_TX 23
#define MODEM_RX 18
#define GSM_RESET 34

#define MAC_ADDRESS_SIZE 18 // Assuming MAC address is in format "XX:XX:XX:XX:XX:XX"
byte mac[6]; 
String str_macAddress;

unsigned long prevMillis = 0;
const unsigned long interval = 60000; // Interval for sending messages

// Device-specific details
const char* deviceSerial = "34865D461830";  // Replace with your device serial

void Init(void);                  // Function prototype for network and GPRS initialization
void connectToGPRS(void);         // Function prototype for GPRS connection
void connectToMQTT(void);         // Function prototype for MQTT connection
bool isNetworkConnected();        // Function prototype to check if network is connected
bool isGPRSConnected();           // Function prototype to check if GPRS is connected

void mqttCallback(char* topic, byte* payload, unsigned int len) {
  SerialMon.print("Message arrived [");
  SerialMon.print(topic);
[... 243 lines cut ...]
    return (response.indexOf("+CREG: 0,1") != -1 || response.indexOf("+CREG: 0,5") != -1);
}

bool isGPRSConnected() {
    String response = gsm_send_serial("AT+CGATT?", 3000);
    return (response.indexOf("+CGATT: 1") != -1);
}

String gsm_send_serial(String command, int delay) {
    String buff_resp = "";
    Serial.println("Send ->: " + command);
    SerialAT.println(command);
    long wtimer = millis();
    while (wtimer + delay > millis()) {
        while (SerialAT.available()) {
            buff_resp += SerialAT.readString();
        }
    }
    Serial.println(buff_resp);
    return buff_resp;
}

**Results**
[image]
[image]

droidblastnz, 2024-09-08:
AT+CMQTTCONNECT: 0,34 open SSL session failed refer to [https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/49](url). In my case and your case its firmware and that it has deprecated Cyphers.  [](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/49#issuecomment-1614106275)

Pasan1021, 2024-09-09:
> AT+CMQTTCONNECT: 0,34 open SSL session failed refer to [https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/49](url). In my case and your case its firmware and that it has deprecated Cyphers.

Do you have a supported firmware?

droidblastnz, 2024-09-09:
> > AT+CMQTTCONNECT: 0,34 open SSL session failed refer to [https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/49](url). In my case and your case its firmware and that it has deprecated Cyphers.
> 
> Thank you for the reply. Do you have a supported firmware?

There is none available that I am aware of....