Xinyuan-LilyGO/T-SIM7600X #102: Meaning of leds present on the schematic of LilyGo T-SIM7600X

PsySc0rpi0n, 2024-09-23:
Hello.

I was trying to figure out what is the meaning of the leds of the leds in this device.
The schematic shows 5 leds.
1 - LED_G7 yellow (next to battery charger, between VCC and GND)
2 - LED_G6 green (connected to CN3065 STDBY pin)
3 - LED_G3 red (connected to CN3065 CHRG pin)
4 - LED_G5 red (connected to NETLIGHT pin of SIM7600X chip)
5 - LED_G8 red (connected to STATUS pin of SIM7600X)

However, I couldn't find a simple explanation of how the 3 first leds work and I also have a few questions I could not find an answer.

**1)** - It would be useful to have a simple explanation of at least LED_G7, LED_G6 and LED_G3. Like when do they turn on/off and what is the meaning when they are turned on/off.

**2)** - There are a few scenarios we can have regarding powering the module and charging the battery and I couldn't anser the following questions:
2.a) When Solar Panel is connected without any battery, does the Solar Panel powers the module itself? Which leds should be turned on in this scenario?

**2.b)** When Solar Panel is connected with a LiOn battery present, which one powers the module? Only the battery or the Solar Panel also "helps" powering the module, allowing slower battery discharge? Which leds should be turned on/off in this scenario?

**2.c)** When a Solar Panel is connected with a LiOn battery is present, does the Solar Panel charges the battery at the same time? Which leds should be turned on/off in this scenario?

Note: I never saw the yellow LED on. Is this normal?

lewisxhe, 2024-09-24:
1. LED_G7 yellow If you see it lights up, it means the battery is connected in reverse, it will not light up when the battery is connected normally
2. LED_G6 green Charging completion indicator, lights up after charging is completed
3. LED_G3 red Charging indicator, lights up when charging is normal
4. LED_G5 red Network status indicator
5. LED_G8 red Modem status indicator, it will light up when the device is turned on normally

Connecting a solar panel alone will not power the board, it will only charge, provided that the battery is present

Whether charging is done by USB or solar panel, charging is controlled by CN3065, and there are only two charging indicators

PsySc0rpi0n, 2024-09-24:
A few more questions, if I'm allowed, yet about the leds.
I did a few tests just to try to understand a bit more about the led's meaning.

1 - For instance, if I use a USB cable to power the device, the green led is always green. But if this led is to indicate full charge reached, how is it on without any LiOn battery? Does it have any meaning in this case?

2 - If I use only the solar panel, the green led is also on but the module is not working, meaning (at least) the SIM7600X does not work. Same question: green led on have any meaning here?

2.a - If I turn the battery switch ON (no battery present), using only the solar panel and I partially cover the solar panel, the red led turns ON and stays like that even if I uncover completely the solar panel afterwards. Knowing that this lead indicates charging is ongoing, does it have any meaning in this situation?

2.b - If the battery switch is OFF (no battery present), the green led is ON. If I partially cover the solar panel, green led turns OFF, red led turns ON. If I totally cover the solar panel, both leds turns OFF. If I uncover the solar panel, the green led becomes ON again. The description I just did was performed in the described order. If there is no battery present, all these behaviours have any specific meaning or we should simply ignore them?

3 - I also tried to plug in the LiOn battery with reverse polarity and the led on the same side of the battery holder turned on but it is not yellow. It was redish (dim though). What does this mean?

4 - What is the role of the LED connected to IO12 of the ESP32 chip, labelled as LED_G1 of blue colour? I never saw it ON nor I know where is it placed in the LilyGo.

Thanks

lewisxhe, 2024-09-25:
1. The light turns on without battery. This is determined by the battery management chip and cannot be changed from the existing hardware.
2, 2a, 2b are the same as the previous one.
3. It can only be said that it is actually red, not yellow.
4. The IO12 LED is controlled and determined by the user, and there is no predefined function.

PsySc0rpi0n, 2024-09-25:
> 1. The light turns on without battery. This is determined by the battery management chip and cannot be changed from the existing hardware.
>    2, 2a, 2b are the same as the previous one.
> 2. It can only be said that it is actually red, not yellow.
> 3. The IO12 LED is controlled and determined by the user, and there is no predefined function.

Ok, so from what I can tell, the schematic showing a yellow led is probably just a wrong label of for some reason the LED that was used was not yellow but red.

lewisxhe, 2024-09-25:
Yes !

PsySc0rpi0n, 2024-09-25:
Ok, I think I'm closing this issue for now!
Thank you for all explanations!

closed 2024-09-25 by PsySc0rpi0n