Xinyuan-LilyGO/T-SIM7600X #104: Sim7600G-H Voice Call feature is not working..

TexeCode07, 2024-10-01:
I have a LilyGO SIMCom 7600G-H module, which I tested with a Verizon or T-Mobile SpeedTalk SIM. I found that the voice call command was not working, with "NO CARRIER" as the reply from the modem. However, it was working fine two weeks ago. To check, I went to T-Mobile's IMEI checker and found that this device is no longer supported. Despite this, internet and SMS are working fine, so I think the issue might be related to the B71 band not supporting calls with this modem.

Could you suggest some good SIMCom LilyGO modules that support the maximum number of SIM modules in the USA? Thank you.
Reference Sites:
[https://www.frequencycheck.com/carriers?utf8=%E2%9C%93&q%5Bname_cont%5D=&q%5Bcountry_id_eq%5D=240&q%5Bfrequency_bands_id_eq%5D=&commit=Search](USA frequency support by sim)
[https://www.t-mobile.com/resources/bring-your-own-phone](T-Mobile IMEI Check)
[https://www.verizon.com/sales/prepaid/byod/enterDeviceIMEI.html](Verizon IMEI Check)

lewisxhe, 2024-10-04:
This has nothing to do with IMEI. Did you buy the SIM7600 version with audio decoding and solder the speaker and microphone to the motherboard? If any of these steps are missing, you will not be able to make a call.

TexeCode07, 2024-10-07:
No normal Lilygo Sim7600G-H Device..  i Just placing call as an Alert. This simcom behaves Very odd now have 2 sim with same data plan. One sim work with Call and sms good other one work sms and internet and no call. Dont know why 2 different cases will try to get more info on this..

lewisxhe, 2024-10-07:
Please confirm whether both boards have NUA8810 audio decoder chip.

If both have this chip, please provide all AT interaction logs