Xinyuan-LilyGO/T-SIM7600X #105: Cannot stop GPS session via AT commands

PsySc0rpi0n, 2024-10-01:
Hello.

I'm using the example code from [here](https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/examples/GPSDebug/GPSDebug.ino) and after the device is ready to take AT commands, it seems that I cannot stop the GPS session.

This is what I get:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c

# Startup #
# Sending "AT" to Modem. Waiting for Response
# .
# Response:
AT
OK
#

***********************************************************
 You can now send AT commands
 Enter "AT" (without quotes), and you should see "OK"
 If it doesn't work, select "Both NL & CR" in Serial Monitor
 DISCLAIMER: Entering AT commands without knowing what they do
 can have undesired consequences...
***********************************************************

AT+CGPS?
+CGPS: 1,1

OK
AT+CGPS=0,1
OK
AT+CGPS?
+CGPS: 1,1

OK
AT
OK
AT+CGPS?
+CGPS: 1,1

OK
AT+CGPS=0,1
OK
AT+CGPS?
+CGPS: 1,1

OK
AT+CGPS=0
OK
AT+CGPS?
+CGPS: 1,1

OK
AT+CGPS?
+CGPS: 1,1

OK
```

So, for some reason I cannot get the GPS session to stop. Why is this? Any common reason?

TexeCode07, 2024-10-01:
i see similar result in case i respond before its 2s result : 
AT+CGPS?
+CGPS: 0,1
OK
AT+CGPS=1
OK
AT+CGPS?
+CGPS: 1,1
--------if respond fast 
AT+CGPS?
+CGPS: 1,1
OK
AT+CGPS=0 --- This Put
OK
AT+CGPS?
+CGPS: 1,1  ---- wrong result as respond before 2s
OK
+CGPS: 0    ---- prev command response i think
AT+CGPS?
+CGPS: 0,1  ---- now it works..

in AT Command module note:
After the GPS closed, it should to wait about 2s~30s for start again. Reason: If the signal
conditions are right (strong enough signals to allow ephemeris demodulation) or ephemeris
demodulation is on going, sometimes MGP will stay on longer in order to demodulate more
ephemeris. This will help the engine provide faster TTFF and possibly better yield later (up to 2
hours), because it has the benefit of more ephemeris available.
[https://m2msupport.net/m2msupport/atcgps-start-stop-gps-session/](url)

PsySc0rpi0n, 2024-10-02:
Yes, I also read about that note. But I also tried to wait longer than that period of time and it was still not working. So, I issued a `AT+CFUN` command to restart the modem, and after that I was able to start/stop GPS sessions almost freely! Like almost immediately one after the other! So, I have no idea what was keeping the GPS session stuck.
I'l still leave this issue open so that we can still discuss about it in the next days, if needed!

PsySc0rpi0n, 2024-10-02:
As a side note, I did something like this to wait for the modem response before issuing new AT command:

```c
bool sendATCommand (const char* command, const char* expectedResponse,
                   unsigned long timeout){
    SerialAT.println (command);
    unsigned long startTime = millis();

    String response = "";
    while (millis() - startTime < timeout){
        while (SerialAT.available()){
            char c = SerialAT.read();
            response += c;
            if (response.indexOf(expectedResponse) != -1){
                Serial.print("Response: "), Serial.println(response);
                return true;
            }
        }
    }
    Serial.println("Timeout for response...");
    return false;
}
```

lewisxhe, 2024-10-04:
Give me the label on the modem and I will determine the hardware version and then test it. @PsySc0rpi0n

PsySc0rpi0n, 2024-10-09:
> Give me the label on the modem and I will determine the hardware version and then test it. @PsySc0rpi0n

[image]

lewisxhe, 2024-10-10:
@PsySc0rpi0n Send AT+SIMCOMATI to see what software version is returned.
Please power on again, use ATDebug, and tell me how to reproduce this problem

PsySc0rpi0n, 2024-10-10:
> @PsySc0rpi0n Send AT+SIMCOMATI to see what software version is returned. Please power on again, use ATDebug, and tell me how to reproduce this problem

```
AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G-H
Revision: LE20B04SIM7600G22
QCN: 
IMEI: 862636053700868
MEID: 
+GCAP: +CGSM
DeviceInfo: 173,170

OK
```

At this point I'm not being able to reproduce. Here is the current output, using ATDebug.ino from gituhb:
```
AT+GPS?
ERROR
AT+CGPS?
+CGPS: 1,1

OK
AT+CGPS=0
OK

+CGPS: 0
AT+CGPS?
+CGPS: 0,1

OK
AT+CGPS=1,1
OK
AT+CGPS?
+CGPS: 1,1

OK
AT+CGPS?
ERROR
AT+CGPS?
+CGPS: 1,1

OK
AT+CGPS?
+CGPS: 1,1

OK
AT+CGPS=0
OK

+CGPS: 0
```

In the meantime, I'll also keep an eye on this behaviour to see if I can catch it again, along the work I'm doing at this moment.

lewisxhe, 2024-10-10:
Please follow the upgrade guide to upgrade to the latest firmware before testing

* [LE20B05SIM7600G22](https://drive.google.com/file/d/1-Tazl3IdiIPqqLkxrEu5lQSlqpgZgBdE/view?usp=sharing)
* [UpdateGuide](https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/docs/UpdateGuide/UpdateGuide.md)

PsySc0rpi0n, 2024-10-10:
> Please follow the upgrade guide to upgrade to the latest firmware before testing
> 
> * [LE20B05SIM7600G22](https://drive.google.com/file/d/1-Tazl3IdiIPqqLkxrEu5lQSlqpgZgBdE/view?usp=sharing)
> * [UpdateGuide](https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/docs/UpdateGuide/UpdateGuide.md)

Is there a guide to upgrade the SIM7600G-H firmware on Linux (Debian Bookworm)?

lewisxhe, 2024-10-11:
Only Windows

PsySc0rpi0n, 2024-10-11:
I may have found a way to upload the firmware onto the device but I can't unrar the file you sent me.

I tried a bunch of possibilities:
```
$ 7z x ~/Documents/SmartBin/LE20B05SIM7600G22_240325.rar 

7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=en_GB.UTF-8,Utf16=on,HugeFiles=on,64 bits,8 CPUs Intel(R) Core(TM) i5-1035G4 CPU @ 1.10GHz (706E5),ASM,AES-NI)

Scanning the drive for archives:
1 file, 83436126 bytes (80 MiB)           

Extracting archive: /home/carlos/Documents/SmartBin/LE20B05SIM7600G22_240325.rar
--
Path = /home/carlos/Documents/SmartBin/LE20B05SIM7600G22_240325.rar
Type = Rar5
Physical Size = 83436126
Solid = -
Blocks = 15
Encrypted = -
Multivolume = -
Volumes = 1

ERROR: Unsupported Method : LE20B05SIM7600G22_240325/appsboot.mbn
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/boot.img
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/contents.xml
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/ENPRG9x07.mbn
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/modem.img
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/NPRG9x07.mbn
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/partition.mbn
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/partition_nand.xml
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/recovery.img
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/recoveryfs.img
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/rpm.mbn
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/sbl1.mbn
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/system.img
ERROR: Unsupported Method : LE20B05SIM7600G22_240325/tz.mbn
                                      
Sub items Errors: 14

Archives with Errors: 1

Sub items Errors: 14
```

```
$ unrar x ~/Documents/SmartBin/LE20B05SIM7600G22_240325.rar 

unrar-free 0.1.3  Copyright (C) 2004  Ben Asselstine, Jeroen Dekkers

Extracting from /home/carlos/Documents/SmartBin/LE20B05SIM7600G22_240325.rar

Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  [�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
Extracting  c�p`                                                     Failed    
14 Failed
```

```
$ unrar-free x ~/Documents/SmartBin/LE20B05SIM7600G22_240325.rar 

unrar-free 0.1.3  Copyright (C) 2004  Ben Asselstine, Jeroen Dekkers

Extracting from /home/carlos/Documents/SmartBin/LE20B05SIM7600G22_240325.rar

Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
Extracting  �Q�V                                                     Failed    
14 Failed
```

Maybe it's my laptop. I'll try in a Windows laptop....