Xinyuan-LilyGO/T-SIM7600X #106: Fails or disconnects after a month

droidblastnz, 2024-10-14:
Deployed a T-SIM7600 board which has been running for close to the entire month. I have code that reconnects the SIM if it shows its disconnected. The T-SIM7600 is currently 4hrs drive away so not a easy fix.

Prior to deploying it to the remote site hence why I needed this type of 4G connectivity I disabled the debugging serial as this is now the second time I had experienced this issue.

Setup code that if it receives a call from cell A it will kick off a reboot or a call from cell B place it into deep sleep that only the ESP not the modem as I couldn't get the code to work for the modem as well.

If I ring the remote T-SIM7600 board it goes straight to answer phone so I am guessing this means its off line or off 4G connectivity?

I also setup code that if it receives a text with a exact code number and task it could reboot, deep sleep, arm or disarm.

There is also code running to check SIM credit balance every 23hrs and this stopped a few days ago as I can see the history logs in the SIM carry portal.

Note the way I have coded is T-SIM7600 board runs a process with timers to check x at different intervals. I do remember seeing some where issues with running timers over a long periods?

So my question is as follows:

- Is the T-SIM7600 board off line as in disconnected from the carrier?
- How do I now fault find if it takes literally a month to enter the fault state.

Any advice on how to move forward please?

lewisxhe, 2024-10-14:
If the system needs to run continuously for a month before a failure occurs, logging to SD may be the only solution.

droidblastnz, 2024-10-14:
> If the system needs to run continuously for a month before a failure occurs, logging to SD may be the only solution.

Thanks @lewisxhe  -  I ring the remote T-SIM7600 board it goes straight to answer phone so I am guessing this means its off line or off 4G connectivity? what is your view on this? Normally you could ring wait for x times until it meets the threshold and answer. The fact it answers straight away?

```
if (SerialAT.available()) {
  String response = SerialAT.readString();
  if (response.indexOf("RING") != -1) {                                            // Incoming call detected   
    String phoneNumber = response.substring(response.indexOf("+CLIP: ") + 7, response.indexOf("+CLIP: ") + 18);  // Retrieve the phone number of the incoming call
    phoneNumber.trim();                                                            // Trim leading/trailing spaces
    SerialMon.print("Incoming call from: ");
    SerialMon.println(phoneNumber.substring(1, phoneNumber.length()));
    phoneNumber = (phoneNumber.substring(1, phoneNumber.length()));
```