Xinyuan-LilyGO/T-SIM7600X #110: AT+CASSISTLOC returns no data

mayoseh, 2024-11-30:
```
AT+CASSISTLOC=1
+CASSISTLOC:,,,
```

I have checked 
```
AT+CLBSCFG=0,3
+CLBSCFG: 0,3,"lbs-simcom.com:3002"
```

and internet connection works:
```
AT+CDNSGIP="google.com"
+CDNSGIP: 1,"google.com","142.250.203.206"
OK

AT+CPING="google.com",1,1,64,1000,10000,255
OK
+CPING: 1,142.250.203.206,64,19,255
+CPING: 3,1,1,0,19,19,19
```

Is there anything I'm missing? according to documentation also `AT+CLSB` should work, but it always returns ERROR

Firmware:
```
AT+CGMR
+CGMR: LE11B13SIM7600M22
```

module SIM7600E-H

lewisxhe, 2024-12-02:
Can I get the location using AT+CLBS=1?

mayoseh, 2024-12-02:
It returns `ERROR` without any details (or maybe I don't know how to read errors).
```
AT+CLBS=1
ERROR
```

as far as I understood from documentation, I should provide CID also:
[image]

but it also returns error.
```
AT+CLBS=1,1
ERROR
```

Once again, first tested network connection and it worked.

That's my loop:
```
void loop() {
  if (SerialAT.available()) {
    Serial.write(SerialAT.read());
  }
  if (Serial.available()) {
    SerialAT.write(Serial.read());
  }
  delay(1);
}
```

I also tested `AT+CPSI?` command:

```
AT+CPSI?
+CPSI: LTE,Online,260-03,0xE30A,71738389,435,EUTRAN-BAND3,1474,4,4,-115,-783,-481,20
```

lewisxhe, 2024-12-02:
Send AT+SIMCOMATI to check the version and the PN number on the modem label, and send it to me

mayoseh, 2024-12-02:
```
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600E-H
Revision: LE11B13SIM7600M22
SIM7600M22_B13V01_191122
QCN: 
IMEI: 862798060930951
MEID: 
+GCAP: +CGSM
DeviceInfo: 173,170

OK
```

lewisxhe, 2024-12-02:
Follow these steps
# step1
```
AT+CNETSTART
````
# step2
```
AT+CLBS=1

```

mayoseh, 2024-12-02:
Ok, so now `AT+CLBS=1` returned a location `0,52.xxxxxx,20.yyyyyy,550` which looks like location of cell tower it's connected to. 

However AT+CASSISTLOC still doesn't work. Should I enable something before as well?

lewisxhe, 2024-12-02:
The command is used to base station location. This command is only applicable to CDMA only or
CDMA and LTE hybrid network or CDMA and EVDO hybrid network. Are you sure your base station supports this?

mayoseh, 2024-12-02:
ok, that makes sense and solves all my questions :) Do you know if this module gives possibility to scan neighbourhood cell towers (eg. LTE/GSM/UMTS ones) for manual triangulation purposes?

lewisxhe, 2024-12-02:
No, it does not support

closed 2024-12-02 by mayoseh