Xinyuan-LilyGO/T-SIM7600X #112: power the board trough the pins. No clear explanation in the manuals.

bartdereu, 2024-12-18:
Hi,

i have been reading trough all of the documentation, but have not found anywhere how i can power the
board trough the pins. There is no VIN pin , only a VBAT , and according to the manual that should be limited 
to 4.2v . Is that correct? 

This need to make allot more clear on what the external power requirements are. 

thanks

lewisxhe, 2024-12-19:
# Notes for new Users
* pin VIN is a USB input, if a battery is being used to power the device there will be no voltage 
output from VIN meaning 3.3v is the only power output.

The homepage has a description. When USB is not used, 5V2A can be connected to the VIN Pin.

bartdereu, 2024-12-19:
Thanks , but for some reason the board does not boot when i power on that bus. , switch is in the ON position. 
It only boots when i power it with a battery installed (and powers from that battery) or by adding 5v to the solar charging
port.

lewisxhe, 2024-12-20:
What about plugging in a USB? VBUS and USB are connected, they are one line.

bartdereu, 2024-12-20:
That works too, plugging 5v to VBUS directly doesnt. Doesnt make any sense?

lewisxhe, 2024-12-21:
That's unlikely, because USBC and VBUS are the same line
[image]

bartdereu, 2024-12-21:
I know it should be. Checked my wires, my soldering connections. Nothing.
When i plug in my battery, it boots (but runs obviously from the battery since there is no current measured)
When i remove the battery it does not boot at all.

[image]

lewisxhe, 2024-12-23:
Unbelievable, VBUS Pin and USB-C are on the same line. You can measure whether the 3V3 pin has power.

And when you plug in the USB, the charging indicator should light up. You can test the difference between plugging in the USB and using the VBUS Pin for power.

bartdereu, 2024-12-23:
When i plug in the USB , no charging indicator lights up. It only lights up when i add power to the solar charging port.

I can't think of anything else then that this is a faulty board.

lewisxhe, 2024-12-24:
There are two components that may be damaged, the first is Q7 or R49 , the second is D4.
They are both responsible for USB access, and I suspect it is Q7 or R49 .
Check here
[image]
[image]
[image]
[image]