Xinyuan-LilyGO/T-SIM7600X #114: unable to open ftdi device with description '*', serial '*' at bus location '*'

norielcahilig, 2025-01-12:
good day, we purchased T-SIM7600G-H for our project. Unfortunately, we are always encountering this error and can't proceed.

[image]
[image]

lewisxhe, 2025-01-13:
SIM7600G This board does not have any FTDI device. It only has a USB TO TTL and USB bridge

norielcahilig, 2025-01-13:
good day, what steps shall we do to solve this error? thank you

lewisxhe, 2025-01-13:
This board does not have an onboard debugger, and can only upload programs via UART

norielcahilig, 2025-01-13:
Thank you for your response, we are able to buy ftdi device, but another
error occur, it says "Invalid Head of packet: Possible noise or serial
corruption" can you help us with this pls.

lewisxhe, 2025-01-13:
You don't need to buy an FTDI device, just plug the USB-C to the TTL port, select the serial port in the Arduino IDE and click upload.
If you encounter a problem where you can't upload, take a wire, connect GPIO0 to GND, then power on, and click upload. After the upload is complete, remove the wire, exit download mode, and press reset. It should work now.

norielcahilig, 2025-01-13:
Thanks, it does work. I appreciate how responsive you all are. This time,
we are encountering this error

Initializing GSM module...
Failed to restart modem. Check power or connection.
Initializing GSM module...
Failed to restart modem. Check power or connection.

What could be the possible issue this time?

Thank you so much.

lewisxhe, 2025-01-13:
Paste your test code

norielcahilig, 2025-01-13:
#include <Arduino.h>
#define TINY_GSM_MODEM_SIM7600 // Define the GSM modem model (SIM7600)
#include <TinyGsmClient.h>

// Define the SIM7600 communication pins
#define RX_PIN 16  // Connect to SIM7600 TX
#define TX_PIN 17  // Connect to SIM7600 RX
#define SerialMon Serial

HardwareSerial gsmSerial(1); // Use UART1 for GSM module
TinyGsm modem(gsmSerial);

String phoneNumber = "09764037833";
String triggerMessage = "Hello";

void setup() {
  // Initialize Serial for debugging
  SerialMon.begin(9600);
  delay(10);

  // Initialize GSM module serial
  gsmSerial.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN);
  delay(3000); // Allow GSM module to power on

  SerialMon.println("Initializing GSM module...");

  // Restart the modem
  if (!modem.restart()) {
    SerialMon.println("Failed to restart modem. Check power or
connection.");
    while (true); // Halt execution
  }
  SerialMon.println("Modem initialized successfully.");

  // Set SMS text mode
  modem.sendAT("AT+CMGF=1"); // Send the AT command to set SMS mode
  if (modem.waitResponse(10000L) != 1) { // Wait for OK response
    SerialMon.println("Failed to set SMS mode.");
    while (true); // Halt execution
  }

  // Configure modem to display incoming SMS
  modem.sendAT("AT+CNMI=2,2,0,0,0");
  if (modem.waitResponse(10000L) != 1) {
    SerialMon.println("Failed to configure SMS notifications.");
    while (true); // Halt execution
  }
  SerialMon.println("GSM module ready.");
}

void loop() {
  // Check for incoming data from the GSM module
  if (gsmSerial.available()) {
    String incomingData = gsmSerial.readString();
    SerialMon.println("Incoming Data: " + incomingData);

    // Look for the trigger phone number in the incoming message
    if (incomingData.indexOf(phoneNumber) != -1) {
      SerialMon.println("Trigger phone number detected!");
      sendSMS(phoneNumber, triggerMessage);
    }
  }
}

void sendSMS(String number, String message) {
  SerialMon.println("Sending SMS...");

  modem.sendAT("AT+CMGF=1"); // Ensure SMS text mode
  if (modem.waitResponse(10000L) != 1) {
    SerialMon.println("Failed to set SMS mode.");
    return;
  }

  modem.sendAT("AT+CMGS=\"" + number + "\""); // Start sending SMS
  if (modem.waitResponse(10000L) != 1) {
    SerialMon.println("Failed to send SMS command.");
    return;
  }

  gsmSerial.print(message); // Send SMS body
  gsmSerial.write(26);      // CTRL+Z to send the message

  if (modem.waitResponse(10000L) == 1) {
    SerialMon.println("Message sent successfully.");
  } else {
    SerialMon.println("Failed to send message.");
  }
}

lewisxhe, 2025-01-13:
When I asked you this, I guessed that you didn't do anything to turn on the modem. Please check the examples we provided. The modem must be turned on before it can be used.
https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/350b1c5c0c37bbf26b7ddd26cc5b4b9314be7843/examples/ATdebug/ATdebug.ino#L46

norielcahilig, 2025-01-13:
we'll check it out, thank you so much sir !!

norielcahilig, 2025-01-14:
Our group had a lot of progress on our project, thank you for responding so
quickly. This time we are encountering an error where the location is not
sent to the registered number, is there another thing that needs to be
activated just like the modem?

lewisxhe, 2025-01-14:
What does it mean? I don't understand?

The modem must be manually activated every time it is turned on

norielcahilig, 2025-01-14:
our project supposed to sent sms containing device's location to a
registered number, but unfortunately, it is not working, the board don't
send sms to a registered number, is there something that can be done? thank
u always

lewisxhe, 2025-01-15:
How did you do it? Post your minimal test code

norielcahilig, 2025-01-15:
I always appreciate all the response, here's the void loop section of our
codes

void loop() {
  char* bufPtr = fonaNotificationBuffer;    // Handy buffer pointer

  if (fona.available()) {      // Any data available from the FONA?
    int slot = 0;              // This will be the slot number of the SMS
    int charCount = 0;

    // Read the notification into fonaInBuffer
    do {
      *bufPtr = fona.read();
      Serial.write(*bufPtr);
      delay(1);
    } while ((*bufPtr++ != '\n') && (fona.available()) && (++charCount <
(sizeof(fonaNotificationBuffer)-1)));

    // Add a terminal NULL to the notification string
    *bufPtr = 0;

    // Scan the notification string for an SMS received notification.
    // If it's an SMS message, we'll get the slot number in 'slot'
    if (1 == sscanf(fonaNotificationBuffer, "+CMTI: " FONA_PREF_SMS_STORAGE
",%d", &slot)) {
      Serial.print("slot: ");
      Serial.println(slot);

      char callerIDbuffer[32];  // We'll store the SMS sender number in here

      // Retrieve SMS sender address/phone number
      if (!fona.getSMSSender(slot, callerIDbuffer, 31)) {
        Serial.println("Didn't find SMS message in slot!");
      }
      Serial.print(F("FROM: "));
      Serial.println(callerIDbuffer);

      // Retrieve SMS value
      uint16_t smslen;
      if (fona.readSMS(slot, smsBuffer, 250, &smslen)) { // Pass in buffer
and max length!
        smsString = String(smsBuffer);
        Serial.println(smsString);
      }

      if (smsString == "location") {
        gpslocation();
      }
      if (fona.deleteSMS(slot)) {
        Serial.println(F("OK!"));
      } else {
        Serial.print(F("Couldn't delete SMS in slot "));
        Serial.println(slot);
        fona.print(F("AT+CMGD=?\r\n"));
      }
    }
  }
}

void gpslocation() {
  float lat = 0;
  float lon = 0;
  float speed = 0;
  float alt = 0;
  int vsat = 0;
  int usat = 0;
  float accuracy = 0;
  int year = 0;
  int month = 0;
  int day = 0;
  int hour = 0;
  int min = 0;
  int sec = 0;

  for (int8_t i = 15; i; i--) {
    SerialMon.println("Requesting current GPS/GNSS/GLONASS location");
    if (modem.getGPS(&lat, &lon, &speed, &alt, &vsat, &usat, &accuracy,
&year, &month, &day, &hour, &min, &sec)) {
      SerialMon.println("Latitude: " + String(lat, 8) + "\tLongitude: " +
String(lon, 8));
      SerialMon.println("Speed: " + String(speed) + "\tAltitude: " +
String(alt));
      SerialMon.println("Visible Satellites: " + String(vsat) + "\tUsed
Satellites: " + String(usat));
      SerialMon.println("Accuracy: " + String(accuracy));
      SerialMon.println("Year: " + String(year) + "\tMonth: " +
String(month) + "\tDay: " + String(day));
      SerialMon.println("Hour: " + String(hour) + "\tMinute: " +
String(min) + "\tSecond: " + String(sec));
      break;
    } else {
      SerialMon.println("Couldn't get GPS/GNSS/GLONASS location, retrying
in 5s.");
      delay(5000L);
    }
  }

  SerialMon.println("Retrieving GPS/GNSS/GLONASS location again as a
string");
  String gps_raw = modem.getGPSraw();
  SerialMon.println("GPS/GNSS Based Location String: " + gps_raw);

  mylati = dtostrf(lat, 3, 6, buff);
  mylong = dtostrf(lon, 3, 6, buff);
  textForSMS = textForSMS + "http://www.google.com/maps/place/" + mylati +
"," + mylong;
  delay(5000);
  modem.sendSMS(SMS_TARGET, textForSMS);  // Send SMS with location link
  Serial.println("SMS sent with location");
  textForSMS = "";
}

lewisxhe, 2025-01-15:
You should use a minimal example to test whether you can send the location via SMS without adding any business logic. Here is a simple demonstration.
https://github.com/Xinyuan-LilyGO/T-SIM7600X/tree/master/examples/GPS_SMS

norielcahilig, 2025-01-15:
we'll try it out thankss

norielcahilig, 2025-01-15:
i'm sorry for asking u this, but it says RegStatus and status is not
declared, i don't know how to fix it sorry

lewisxhe, 2025-01-15:
RegStatus   change SIM7600RegStatus

norielcahilig, 2025-01-15:
IT WORKS. THANK YOU SO MUCH. How could we integrate this to our system?
What parts shall we remove and replace? Thank you very much sir !!

lewisxhe, 2025-01-15:
This is related to your business. I can only provide an example. You need to solve the others by yourself.

norielcahilig, 2025-01-15:
No problem, thank you very much again sir

norielcahilig, 2025-01-15:
we're too shy to ask but sorry, we are desperately really in need of your
help, we are already receiving sms containing the location, but we need it
to be activated first by sending sms to the esp32 before it will send the
sms with the location. I hope you can help us please.

lewisxhe, 2025-01-15:
I can only help you up to this point. The rest is up to you.
https://github.com/Xinyuan-LilyGO/T-SIM7600X/tree/master/examples/Trigger_GPS_form_SMS

norielcahilig, 2025-01-15:
Thank you so much, we really appreciate all your help.