Xinyuan-LilyGO/T-SIM7600X #115: https request

DGnice06, 2025-01-17:
Hello,
I am trying to use https on T-SIM7600G-H with LTE .
I don't need and don't use Wifi.
Everything is ok on Http , but impossible to have a request with https (I try AT commands and many libraries )
Originally, I wanted to use ArduinoWebsockets.
I can't even get an answer with:
xTimerStart(shutdown_signal_timer, portMAX_DELAY); assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

I can't even get an answer with:
I use TinyGsmClient. 
    // HTTPS
  client.println("AT+HTTPINIT");
  delay(1000);
  client.println("AT+HTTPPARA=\"CID\",1");
  delay(1000);
  client.println("AT+HTTPSSL=1");
  delay(1000);
  client.println("AT+HTTPPARA=\"URL\",\"https://497bis.gidev.net/test.txt");
  delay(1000);

Do you have any idea?

lewisxhe, 2025-01-20:
Refer to this manual
https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/datasheet/SIM7500_SIM7600_Series_HTTP(S)_Application%20Note_V3.00.pdf

droidblastnz, 2025-01-26:
Example

```
#include <TinyGsmClient.h>
#include <SoftwareSerial.h>

// SIM7600 Serial setup
#define MODEM_RX 10
#define MODEM_TX 11
#define MODEM_PWRKEY 9
#define MODEM_RST 8
#define MODEM_POWER_ON 12

// APN settings
const char apn[] = "your_apn_here"; // Replace with your APN
const char user[] = "";             // Leave empty if not required
const char pass[] = "";             // Leave empty if not required

SoftwareSerial modemSerial(MODEM_RX, MODEM_TX);
TinyGsm modem(modemSerial);

void setup() {
  Serial.begin(115200);
  delay(10);

  modemSerial.begin(9600);
  delay(3000); // Wait for modem to power on

  pinMode(MODEM_PWRKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);

  // Power on the modem
  digitalWrite(MODEM_POWER_ON, HIGH);
  delay(1000);
  digitalWrite(MODEM_PWRKEY, HIGH);
  delay(1000);
  digitalWrite(MODEM_PWRKEY, LOW);

  // Initialize modem
  Serial.println("Initializing modem...");
  if (!modem.restart()) {
    Serial.println("Failed to restart modem, attempting manual configuration");
  }

  // Set APN
  Serial.println("Setting APN...");
  if (!modem.gprsConnect(apn, user, pass)) {
    Serial.println("Failed to connect to GPRS");
    while (true);
  }
  Serial.println("GPRS connected");

  // HTTPS Request
  sendHttpsRequest();
}

void sendHttpsRequest() {
  Serial.println("Starting HTTPS request...");

  // Initialize HTTP service
  modem.sendAT(GF("AT+HTTPINIT"));
  waitForResponse();

  // Set parameters
  modem.sendAT(GF("AT+HTTPPARA=\"CID\",1"));
  waitForResponse();

  modem.sendAT(GF("AT+HTTPSSL=1"));
  waitForResponse();

  // Set URL
  modem.sendAT(GF("AT+HTTPPARA=\"URL\",\"https://www.google.fr/\""));
  waitForResponse();

  // Start GET request
  modem.sendAT(GF("AT+HTTPACTION=0"));
  waitForResponse();

  // Read response
  modem.sendAT(GF("AT+HTTPREAD"));
  waitForResponse();

  // Terminate HTTP service
  modem.sendAT(GF("AT+HTTPTERM"));
  waitForResponse();
}

void waitForResponse() {
  if (modem.waitResponse(10000) == 1) {
    Serial.println("OK");
  } else {
    Serial.println("Failed");
  }
}

void loop() {
  // Empty loop
}

```