Xinyuan-LilyGO/T-SIM7600X #117: SIM7670G AGPS time to fix issue

david-biro, 2025-01-27:
Hi 

I faced a problem when playing with SIM7670G-MNGV, not really sure if I am using the proper AT commands as there is no significant difference between the time to position fix (after cold start) when sending AT+CGNSSAGPS and when not. I would expect shorter time (now it is up to 10 minutes) to position fix when AGPS enabled indoors. I am using the following commands:

AT+CGPSWARM
OK

AT+CGNSSIPR=115200
OK

AT+CGNSSMODE=15
OK

AT+CGNSSNMEA=1,1,1,1,1,1,0,0,0,0
OK

AT+CGNSSNMEARATE=1
OK

AT+CGNSSAGPS
OK

+CGNSSAGPS: SUCCESS
AT+CGNSSPORTSWITCH=1,1
OK

If I am using the AT commands incorrectly, how is the way to activate AGPS?

lewisxhe, 2025-02-06:
This warehouse is for 7600G, not SIM7670G, and SIM7670G does not have acceleration function

david-biro, 2025-02-06:
Hi Lewis!

From **SIM767XX**Series_AT_Command_Manual_V1.04.pdf, Chapter 21.2.23, _"AT+CGNSSAGPS Get AGPS data from the AGNSS server for assisted positioning"_.

Response
1)If successfully:
OK
**+CGNSSAGPS: success.** 
2)If failed:
ERROR
3)If failed:
OK

I got AT+OK and +CGNSSAGPS: SUCCESS. What does those indicate then?

lewisxhe, 2025-02-06:
V1.04? The latest version I have is only V1.01. Can you upload the manual for me to check?

david-biro, 2025-02-06:
[SIM767XX Series_AT_Command_Manual_V1.04.pdf](https://github.com/user-attachments/files/18683903/SIM767XX.Series_AT_Command_Manual_V1.04.pdf)

david-biro, 2025-02-06:
Yes, it is a LilyGO board, the SIMCOM module is SIM7670G-MNGV, the general processor is ESP32-S3-WROOM-1 on it. What do you mean under it does not support? Do you get error when executing AT+CGNSSAGPS?

[https://www.banggood.com/LILYGO®-T-SIM7670G-...](https://www.banggood.com/LILYGO%C2%AE-T-SIM7670G-S3-LTE-4G-ESP32-S3-Development-Board-Cellular-Module-IOT-Devices-Low-power-Long-range-Communication-With-GPS-p-2020494.html?gmcCountry=SE&currency=SEK&cur_warehouse=CN&createTmp=1&ID=6328895&utm_source=googleshopping&utm_medium=cpc_pt&utm_campaign=aceng-pmax-se-brad-type-240223-meruem&utm_content=meruem&ad_id=&gad_source=1&gclid=CjwKCAiAtYy9BhBcEiwANWQQLyPjr3Gy0X8LikyNSWVciz2Cxzti2eaeLz5bYaGAwrCJFiuiYQOCFRoCJW8QAvD_BwE)

lewisxhe, 2025-02-06:
I need to check, the latest version of the B05 firmware does not support the AT+CGNSSAGPS command.

lewisxhe, 2025-02-06:
You send AT+SIMCOMATI to check the version

david-biro, 2025-02-06:
AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2360B01SIM767XM5A_M
SIM767XM5A_B01V03_231207
IMEI: XXXXXXXXXXXXXX

lewisxhe, 2025-02-06:
Then your manual should be A76XX, not SIM7670G. I will contact SIMCOM to confirm and I am waiting for their reply.

david-biro, 2025-02-06:
That I don't understand. Where does the A767XX type come from?

lewisxhe, 2025-02-06:
SIMCOM has replied to me that AGPS is indeed available, but the manual you provided is incorrect. The correct version is V1.0.6. Below is my test process.
 示例将同步更新到https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/c6885adde546b0dcd4586b59d891e81df9f49e0c/examples/GPS_Acceleration/GPS_Acceleration.ino

```bash
« AT+IPADDR

AT+IPADDR

+IPADDR: 10.19.53.204

OK
Network IP:AT+IPADDR
+IPADDR: 10.19.53.204
Enabling GPS/GNSS/GLONASS
AT+CGDRT=4,1
AT+CGDRT=4,1

OK
AT+CGSETV=4,1
AT+CGSETV=4,1

OK
AT+CGNSSPWR=1
AT+CGNSSPWR=1

OK

GPS Enabled
AT+CGNSSIPR=115200
AT+CGNSSIPR=115200

OK
GPS acceleration is enabled
AT+CGNSSPWR?
AT+CGNSSPWR?

+CGNSSPWR: 1
AT+CAGPS

OK
AT+CAGPS

OK

+AGPS: success success!!!
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
FixMode:3
Latitude:22.655754	Longitude:114.060234
Speed:1.73	Altitude:76.50
Visible Satellites:8	Used Satellites:0
Accuracy:2.39
Year:2025	Month:2	Day:6
Hour:9	Minute:46	Second:49
Retrieving GPS/GNSS/GLONASS location again as a string
AT+CGNSSINFO
AT+CGNSSINFO

+CGNSSINFO: 3,08,,,,22xxxxxxxxxxxxxxx,8

OK
GPS/GNSS Based Location String:3,08,,,,2xxxxxxxx11,N,11xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Disabling GPS
AT+CGNSSPWR=0
AT+CGNSSPWR=0

OK

```