Xinyuan-LilyGO/T-SIM7600X #118: LilyGo Sim-T7600G-H

PsySc0rpi0n, 2025-01-28:
Hello.
I have a 18650 Li-Ion battery which is being (or should be) charged by a 5V 2.5W solar panel. However, the battery is not charging over 2.99V. The solar panel is reading 4.33V, so what could be the reasons why the battery is not charging? The device swtich is OFF (so no load/discharge at the same time)  and there is a red led ON, which I think indicates charging, but seems it's not charging according to the multimeter on the battery. It's cloudy and rainy today but the solar panel is still capturing light, so I can't see a reason for the battery not getting any charge!

PsySc0rpi0n, 2025-02-05:
Support of LilyGo, at least the official one sucks, ffs...

The device apparently have been updated in terms of hardware, because I have a board revision t54_V1.3 22-6-27 and I think in a previous conversation with someone from support, I was told the battery charger IC was changed from a CN3065 to a CN3068.

The schematics were not updated, nothing is said anywhere about this change and I can't even find a proper datasheet for this CN3068 to compare it with the CN3065 and see the differences!

I'm using a 5V, 2.5W (500mA) solar panel and somehow it is not charging the battery. I'm not sure if it's due to insufficient voltage or what. I needed to know which IC is being used to be able to try to figure out if this is a low solar panel voltage issue or if it is something else!

lewisxhe, 2025-02-06:
The SIM7600G solar charging chip is updated to CN3165. The schematic diagram remains unchanged, but it is changed from CN3065 to CN3165. The input range is also 4.4V ~ 6V input voltage.

PsySc0rpi0n, 2025-02-06:
> The SIM7600G solar charging chip is updated to CN3165. The schematic diagram remains unchanged, but it is changed from CN3065 to CN3165. The input range is also 4.4V ~ 6V input voltage.

Ohh really? CN3165? I thought it was CN3068. I think I was told by someone in the support chat that it was replaced by a CN3068. And I wanted to make sure!

lewisxhe, 2025-02-06:
Yes, I have already asked

PsySc0rpi0n, 2025-02-06:
Ok, @lewisxhe if you know, please tell me something about the charging circuit of this device.

If I measure the voltage of the solar panel I have, in OL (not attached to anything), I can measure 5.31V. However, when I connect it to the device (the device has a 18650 cell connected but the slide switch is off), the voltage reading drops to 4.33V which makes me think that the CN3165 is not being "activated" to charge the cell. Why does this happen? The voltage drop?

lewisxhe, 2025-02-07:
What is the OL interface? You can connect the solar interface to a voltage regulator of about 5V, and check the current indication to determine whether it is charging.

PsySc0rpi0n, 2025-02-07:
> What is the OL interface? You can connect the solar interface to a voltage regulator of about 5V, and check the current indication to determine whether it is charging.

The OL interface? I meant the solar panel! The solar panel in OL (open loop voltage).

lewisxhe, 2025-02-08:
You mean the open circuit voltage of the solar panel is 5.31V, but after connecting the board it becomes 4.33V right?

The open circuit voltage is not the actual load voltage of the solar panel. The actual voltage should be measured after connecting the load.

PsySc0rpi0n, 2025-02-10:
> You mean the open circuit voltage of the solar panel is 5.31V, but after connecting the board it becomes 4.33V right?
> 
> The open circuit voltage is not the actual load voltage of the solar panel. The actual voltage should be measured after connecting the load.

Got it... But is it normal that the voltage drops to this value that seems to be kind of fixed 4.33V? This looks like something is forcing it to be this value. I just wondered if anything in the circuitry of my device could be forcing to this situation!

lewisxhe, 2025-02-11:
As I described before, no-load voltage is the voltage without load, and it has no practical meaning.

The real voltage is measured after connecting the load. It is impossible for the component to force the voltage to drop. Because the current increases, the voltage of your solar panel also decreases.

PsySc0rpi0n, 2025-02-11:
> As I described before, no-load voltage is the voltage without load, and it has no practical meaning.
>
> The real voltage is measured after connecting the load. It is impossible for the component to [...]

Ok, so this means that the solar panel is not provide enough current to the device and that is the reason the voltage is dropping, yes? Becasue if the panel was able to provide the necessary current, the voltage wouldn't drop, right?

lewisxhe, 2025-02-12:
Yes